Klipper3d/klipper #3016: Error loading kinematics 'delta'

hinteor, 2020-06-23:
[tmc2208 stepper_a]
uart_pin: P1.17
microsteps: 16
interpolate: True
run_current: 0.800
hold_current: 0.700
stealthchop_threshold: 80

[tmc2208 stepper_b]
uart_pin: P1.15
microsteps: 16
interpolate: True
run_current: 0.800
hold_current: 0.700
stealthchop_threshold: 80

[tmc2208 stepper_c]
uart_pin: P1.10
microsteps: 16
interpolate: True
run_current: 0.800
hold_current: 0.700
stealthchop_threshold: 80

[stepper_a]
step_pin: P2.2
dir_pin: P2.6
enable_pin: !P2.1
step_distance: .01
endstop_pin: P1.28  # P1.29 for X-min
homing_speed: 100
position_endstop: 262.967772 
angle: 329.193445 
arm_length : 212.053746

[stepper_b]
step_pin: P0.19
dir_pin: P0.20
enable_pin: !P2.8
step_distance: .01
endstop_pin: P1.26  # P1.27 for Y-min
position_endstop: 262.967772 
angle: 329.193445 
arm_length : 212.053746

[stepper_c]
step_pin: P0.22
dir_pin: P2.11
enable_pin: !P0.21
step_distance: .01
endstop_pin: P1.24  # P1.25 for Z-min
position_endstop: 268.101798 
angle: 90.000000 
arm_length : 212.053746

[extruder]
step_pin: P2.13
dir_pin: P0.11
enable_pin: !P2.12
step_distance: 0.006271
nozzle_diameter: 0.500
filament_diameter: 1.750
heater_pin: P2.7
sensor_type: EPCOS 100K B57560G104F
sensor_pin: P0.24
control: pid
pid_Kp : 25.714
pid_Ki : 1.617
pid_Kd : 102.215
min_temp: 0
max_temp: 255
max_extrude_only_distance: 100.0

[heater_bed]
heater_pin: P2.5
sensor_type: EPCOS 100K B57560G104F
sensor_pin: P0.23
control: watermark
max_delta: 2.0
min_temp: 0
max_temp: 130

[fan]
pin: P2.3

[mcu]
serial: /dev/serial/by-id/usb-Klipper_lpc1768_09400116011C3DAFD9E3875CC52000F5-if00

#[probe]
#pin: P1.25
#z_offset: 25.600

[printer]
kinematics: delta
max_velocity: 300
max_accel: 3000
max_z_velocity: 150
delta_radius: 96.896549

[delta_calibrate]
radius: 90

########################################
# EXP1 / EXP2 (display) pins
########################################

[board_pins]
aliases:
    # EXP1 header
    EXP1_1=P1.30, EXP1_3=P1.18, EXP1_5=P1.20, EXP1_7=P1.22, EXP1_9=<GND>,
    EXP1_2=P0.28, EXP1_4=P1.19, EXP1_6=P1.21, EXP1_8=P1.23, EXP1_10=<5V>,
    # EXP2 header
    EXP2_1=P0.17, EXP2_3=P3.26, EXP2_5=P3.25, EXP2_7=P1.31, EXP2_9=<GND>,
    EXP2_2=P0.15, EXP2_4=P0.16, EXP2_6=P0.18, EXP2_8=<RST>, EXP2_10=<NC>
    # Pins EXP2_1, EXP2_6, EXP2_2 are also MISO, MOSI, SCK of bus "ssp0"

[display]
lcd_type: hd44780
rs_pin: P1.19
e_pin: P1.18
d4_pin: P1.20
d5_pin: P1.21
d6_pin: P1.22
d7_pin: P1.23
encoder_pins: ^P3.26, ^P3.25
click_pin: ^!P0.28
#kill_pin: ^!EXP2_8

[output_pin beeper]
pin: P1.30

[verify_heater heater_bed]
check_gain_time:240

-------------------

error : 
Klipper state: Disconnect
Error loading kinematics 'delta'
Once the underlying issue is corrected, use the "RESTART"
command to reload the config and restart the host software.
Printer is halted
Klipper state: Not ready

hinteor, 2020-06-23:
[klippy.log](https://github.com/KevinOConnor/klipper/files/4820190/klippy.log)

hinteor, 2020-06-23:
Starting Klippy...
Args: ['/home/pi/klipper/klippy/klippy.py', '/home/pi/printer.cfg', '-l', '/tmp/klippy.log']
Git version: 'v0.8.0-606-ga4c31baf'
CPU: 4 core ARMv7 Processor rev 3 (v7l)
Python: '2.7.16 (default, Apr  6 2019, 01:42:57) \n[GCC 8.2.0]'
Building C code module c_helper.so
Start printer at Tue Jun 23 13:12:32 2020 (1592914352.5 2156.1)
Unable to open config file /home/pi/printer.cfg
Traceback (most recent call last):
  File "/home/pi/klipper/klippy/configfile.py", line 104, in _read_config_file
    f = open(filename, 'rb')
IOError: [Errno 2] No such file or directory: '/home/pi/printer.cfg'
Config error
Traceback (most recent call last):
  File "/home/pi/klipper/klippy/klippy.py", line 136, in _connect
    self._read_config()
  File "/home/pi/klipper/klippy/klippy.py", line 122, in _read_config
    config = pconfig.read_main_config()
  File "/home/pi/klipper/klippy/configfile.py", line 221, in read_main_config
    data = self._read_config_file(filename)
  File "/home/pi/klipper/klippy/configfile.py", line 110, in _read_config_file
    raise error(msg)
Error: Unable to open config file /home/pi/printer.cfg
Unable to open config file /home/pi/printer.cfg
Once the underlying issue is corrected, use the "RESTART"
command to reload the config and restart the host software.
Printer is halted

Restarting printer
Start printer at Tue Jun 23 13:47:19 2020 (1592916439.1 4242.7)
===== Config file =====
[tmc2208 stepper_a]
uart_pin = P1.17
microsteps = 16
interpolate = True
run_current = 0.800
hold_current = 0.700
stealthchop_threshold = 80

[tmc2208 stepper_b]
uart_pin = P1.15
microsteps = 16
interpolate = True
run_current = 0.800
hold_current = 0.700
stealthchop_threshold = 80

[tmc2208 stepper_c]
uart_pin = P1.10
microsteps = 16
interpolate = True
run_current = 0.800
hold_current = 0.700
stealthchop_threshold = 80

[stepper_a]
step_pin = P2.2
dir_pin = P2.6
enable_pin = !P2.1
step_distance = .01
endstop_pin = P1.28
homing_speed = 100
position_endstop = 262.967772
angle = 329.193445
arm_length = 212.053746

[stepper_b]
step_pin = P0.19
dir_pin = P0.20
enable_pin = !P2.8
[... 853 lines cut ...]
Config error
Traceback (most recent call last):
  File "/home/pi/klipper/klippy/klippy.py", line 136, in _connect
    self._read_config()
  File "/home/pi/klipper/klippy/klippy.py", line 131, in _read_config
    m.add_printer_objects(config)
  File "/home/pi/klipper/klippy/toolhead.py", line 577, in add_printer_objects
    config.get_printer().add_object('toolhead', ToolHead(config))
  File "/home/pi/klipper/klippy/toolhead.py", line 262, in __init__
    raise config.error(msg)
Error: Error loading kinematics 'delta'
Error loading kinematics 'delta'
Once the underlying issue is corrected, use the "RESTART"
command to reload the config and restart the host software.
Printer is halted
Klipper state: Not ready
Error loading kinematics 'delta'
Once the underlying issue is corrected, use the "RESTART"
command to reload the config and restart the host software.
Printer is halted

Error loading kinematics 'delta'
Once the underlying issue is corrected, use the "RESTART"
command to reload the config and restart the host software.
Printer is halted

hinteor, 2020-06-23:
the problem disappears if removed

"position_endstop:
angle:
arm_length: "

on all motors except a

jakep82, 2020-06-23:
You used the same angle for stepper_a and stepper_b.

KevinOConnor, 2020-07-10:
I'm going to close this issue as it looks like the original question was answered.

-Kevin

closed 2020-07-10 by KevinOConnor