Klipper3d/klipper #3017: SKR Pro baud @115200bps, 250000 possible?

DroneMang, 2020-06-23:
Hi,

I have an SKR Pro and it is working fine, but when it connects it says its baud is at @115200bps.  From what I have read, it is using USB serial so this baud isn't the actual connection speed?

My request is that I could specify the fastest connection speed, or that this notification would change, or that a note would be placed in the SKR Pro example configuration saying to not worry about the baud and that it is as fast as possible already.  Thank you!
[klippy.zip](https://github.com/KevinOConnor/klipper/files/4820054/klippy.zip)

jakep82, 2020-06-23:
https://github.com/KevinOConnor/klipper/blob/master/docs/FAQ.md#how-do-i-change-the-serial-baud-rate

DroneMang, 2020-06-23:
If you want to change the baud rate anyway, then the new rate will need to
be configured in the micro-controller (during make menuconfig) and that
updated code will need to be compiled and flashed to the micro-controller.

These instructions do not work for the SKR pro. During the make commands
there are no options for baud rate in the options. If I put in the config
file setting the baud rate to 250,000 it seems to throw errors and
eventually just connect at 115,200 anyways. Thanks!

jakep82, 2020-06-23:
I don't follow.  Your log indicates no connection speed was specified.  By default Klipper uses 250000 ([see here](https://github.com/KevinOConnor/klipper/blob/master/config/example.cfg#L284)) and the log doesn't show any issues with the serial connection.

DroneMang, 2020-06-23:
Here is from the octoprint terminal:

Connecting to: /tmp/printer
Changing monitoring state from "Offline" to "Opening serial port"
Connected to: Serial<id=0x65335530, open=True>(port='/tmp/printer', baudrate=115200, bytesize=8, parity='N', stopbits=1, timeout=10.0, xonxoff=False, rtscts=False, dsrdtr=False), starting monitor
Changing monitoring state from "Opening serial port" to "Connecting"

jakep82, 2020-06-23:
From the FAQ I linked earlier.

> The baud rate shown on the OctoPrint web page has no impact on the internal Klipper micro-controller baud rate. Always set the OctoPrint baud rate to 250000 when using Klipper.

Even if Octoprint won't connect at 250000, that has no effect on the internal connection which is 250000 by default.

DroneMang, 2020-06-23:
Okay that makes sense. So I am trying to max out the step rate on my TMC
drivers. The fastest I can go is 64 micro stepping and I max out at 360 mm
per second, faster then the MCU goes into shutdown. I just want to make
sure that there are no baud rate bottlenecks as far as that goes. Will the
printing commands be slowed down because of the slower baud rate connecting
to octoprint? or is that sort of pre-compiled because of how clipper runs?
Thanks.

jakep82, 2020-06-23:
If you want more speed use 32 microsteps.  You don't gain any perceptible precision by using higher microstep settings, and you shouldn't notice a noise difference if you use interpolation.  The connection speed between Octoprint and Klipper is not what is limiting you.

DroneMang, 2020-06-24:
I got this figured out, just changed the baud in octoprint settings to default to 250000.  Perhaps this could be added to the initial setup document, to also specify the baud to 250000 in addition to setting the serial port to /tmp/printer.  I'll close this. Thanks.

closed 2020-06-24 by DroneMang