Klipper3d/klipper #3018: 3/4 Pin Fan Control

W-Floyd, 2020-06-23:
I am going to build a new hotend controller for my printer, and would like to use nice 12V fans. However, these only come in 3 or 4 pin configurations. While I understand I might get away with ignoring the 2 exra pins, and drive the power directly, but I'd much prefer to control it correctly. Is this something that might be implemented?

theophile, 2020-06-24:
For a 3-pin fan, ignoring the tach wire is "controlling it correctly," you just won't have the RPM feedback, which i don't think you'd be able to use in this application anyway. For a 4-pin fan, I think you need extra hardware. Read up on how to control a 4-pin fan with the RPI as I think that'll be the procedure.

KevinOConnor, 2020-07-10:
I'm going to close this issue as it looks like the original question was answered (as far as we know, you can control the fan, but you wont have the tach feedback).

-Kevin

closed 2020-07-10 by KevinOConnor

W-Floyd, 2021-05-21:
See #3897.