Klipper3d/klipper #3020: Tool not heating when getting loaded in octoprint

Diana6969, 2020-06-24:
It's not that serious of an issue, but I really don't want to be switching the tool temps all the time. When I load and select print through octoprint it automatically selects the target temperature for the bed (from cura profile settings), but it doesn't do that for the tool.
I would be happy to hear any solutions!
PS: Kevin, what a marvellous job, I mean the Klipper is just amazing.
[klippy.log](https://github.com/KevinOConnor/klipper/files/4826731/klippy.log)

jakep82, 2020-06-24:
Not sure I follow what you're reporting.  I see no problems in the log regarding heating.  Have you looked at the gcode file in question to verify it's requesting the correct temperatures?  The log shows an initial hotend temp of 100, followed by a change to 215.

jakep82, 2020-06-25:
Temperature is controlled by gcode.  If your start gcode includes `M109 S215`, then klipper will heat the hotend to 215 and wait until it reaches that temperature before running the next line of gcode.

KevinOConnor, 2020-07-10:
I'm going to close this issue as it looks like the original question was answered (specify the temperature in the slicer).

-Kevin

closed 2020-07-10 by KevinOConnor