Klipper3d/klipper #3022: tmc config example

robthide37, 2020-06-26:
Might not be big deal but the hysteresis start and end values are switched for the 2208 and 2209 drivers.

KevinOConnor, 2020-07-02:
FYI, I do not understand what is being reported here.  I recommend explicitly stating "what I observed" and "what I expected to observe".

-Kevin

robthide37, 2020-07-02:
I will link the tmc white papers pertaining to the chopper timings. It is
just a typo i believe when the defaults were set in the 2208 and 2209 .py
files and also the config extra tmc section it has hend and hstrt
backwards. Hstrt should never exceed 3 . They are additive values and
should not exceed 15 combined taking into account that hstrt has +1 and
hend -3 offsets applied.  The only time it matters where you out the extra
time is if using hysteresis decrement in which case it should be a ratio
around 20 percent after offsets.  So with that in mind the default config
is breaking the less than 3 rule on hstrt which does seem to affect the
sine wave at low speed when i switched them for testing . It def isn't
drastic and its not dangerous for hardware just seems like simple fix to
fall inline with the manufacturer guidelines.  Hstrt 0 hend 5 vs hstrt 5
and hend 0 now

KevinOConnor, 2020-07-02:
I continue to not understand what is being reported.  Please take the time to _explicitly_ state what you observed and what you expected to observe.

For example: "In file a/b/c.py I saw 'x= 27', and I expected to see 'x = 32', because at htttp://xyz/some_site.pdf on page 99 it says xyz."

-Kevin

robthide37, 2020-07-02:
Got it ok in the config extra file under the tmc 2209 and tmc 2209 it has
hyesteris start 5 and hysterisis end 0 ots supposes to be the other way
around if you look at marlin 24v defaults and Also here
https://www.google.com/url?sa=t&source=web&rct=j&url=https://www.trinamic.com/fileadmin/assets/Support/Appnotes/AN001-spreadCycle.pdf&ved=2ahUKEwjCm5a35q7qAhUQhXIEHe7xAGsQFjAAegQIAhAC&usg=AOvVaw3u9cJK7FwvqpSoSSL1Q3RW

I'm on my phone but their was also a 2 python files that have those default
values as well that I will link ,I think this is one of them iirc
https://github.com/KevinOConnor/klipper/blob/master/klippy/extras/tmc2209.py.
Thanks and sorry for not being clearer at l first.

KevinOConnor, 2020-07-03:
Okay, I understand.

The code takes the HSTRT and HEND values directly from the spec (it says CHOPCONF has "Reset default=0x10000053".  The spec is confusing here as it does not make a recommendation for a default value.  So, it's unclear if the "reset default" is meaningful.  However, I'm reluctant to "invent a default" in Klipper.

-Kevin

robthide37, 2020-07-03:
But don't you have a default in tmc2208 and tmc 2209.py and also in the
example config under the tmc header . I am just recommending to change it
to the marlin default and the smoothie default which also would bring it in
line with the tmc guidelines. Right now it's in violation with the allowed
timing which they state won't damage hardware but will cause more vibration
at mid speed and less linear motion at low speed. I personally tested it
and if you switch them it makes a difference. Also off topic but with your
whole s curve and motion as far as decel if you lower pwm grad to say 5 or
6 it will give much better control on decel as it will not drop off as fast
especially on heavy carriage or bed. I highly recommend following that
trinamic tuning guide it made my printer smoother ,quieter and also am able
run higher acceleration.  Sorry for the tangent.  I know your busy but if
you take min to read the link you will see switching the values is def
called for and i truly believe whoever wrote the tmc section in klipper
just made a simple copy paste error.

KevinOConnor, 2020-07-03:
The tmc2208 and tmc2209 specifications state that on a reset the driver defaults to HSTRT=5, HEND=0.  So, Klipper does the same.

The tmc specifications can be very confusing.  Be careful not to confuse the high-level "hysteresis start" and "hysteresis end" descriptions with the low-level HSTRT and HEND register fields.  I'm not aware of any issue setting these two fields to the given values (as Trinamic has stated that they do on a reset).

If you are aware of a published recommendation from Trinamic for these registers, please provide a link to the document, the page number, and the specific paragraph with that recommendation.

-Kevin

n8bot, 2020-07-03:
It appears Mr. OConnor  is correct, based on this:

>Default: StealthChop (GCONF.en_SpreadCycle=0)
OTP 1.0 to 1.7 and 2.0 used for StealthChop
SpreadCycle settings: **HEND=0; HSTART=5;** TOFF=3

From: [TMC2208/2 & TMC2224 family Datasheet](https://www.trinamic.com/fileadmin/assets/Products/ICs_Documents/TMC220x_TMC2224_datasheet_Rev1.09.pdf)

But as he says, the documentation isn't quite clear.

KevinOConnor, 2020-07-25:
Okay, I'm going to close this as it looks like the original question was answered.

-Kevin

closed 2020-07-25 by KevinOConnor