Klipper3d/klipper #3023: TMC5160 not going into stealthChop

changedsoul, 2020-06-26:
I am finding these drivers are randomly going in and out of stealthChop mode. I am hoping someone can lend a hand. I dont know if its an issue with klipper, the drivers, how they are configured, or what.

Attached is the log with the TMC_DUMP output of both X and Y.

[log.log](https://github.com/KevinOConnor/klipper/files/4838613/log.log)

[printer.txt](https://github.com/KevinOConnor/klipper/files/4838633/printer.txt)

KevinOConnor, 2020-06-27:
Unfortunately, the attached log is not complete, so there is little we can do to assist.  Also, note that DWC2 alters the internal state of Klipper, so it's not something we can support here - you'll either need to reproduce the issue using the pristine code from this repo or report it to the dwc2 developers.

-Kevin

changedsoul, 2020-06-27:
I was meaning to revisit this post and update it. I was looking around the web and came across something about needing to disconnect the CLK pin. I gave that a try and it seemed to have worked. I dont understand why, but so far it seems its powering up in stealthChop, and the StealthChop threshold now seems to be working too.

So I dont know if this is a valid fix or not, or if this is only pertinent to my version TMC drivers. Maybe this post can help someone down the road having similar issues.

If it seems this was not a fix, perhaps I can return with a more complete log file. My apologies, I cleared the log before I uploaded it because it was gargantuan in size.

KevinOConnor, 2020-07-10:
I'm going to close this issue as it looks like a solution was found for the original issue.

-Kevin

closed 2020-07-10 by KevinOConnor