Klipper3d/klipper #3027: [FR] Add accelerometers support to Klipper for input shapers tuning

dmbutyugin, 2020-06-27:
Accelerometers can be used to find resonances of the printer more accurately. More importantly, they can be used to find all resonances of the printer per axis, instead of only the strongest one using a test print. This can make it easier to tune input shapers.

TBH, I'm not sure what the demand for this feature is. Accelerometers (e.g. ADXL345) are not very expensive these days, and may be worth it given the time savings for tuning. However, they require some soldering and wiring - and not everyone is comfortable doing that. @KevinOConnor what's your take one this? Do you think this is something that makes sense to have support for in Klipper?

@BlackStump, @Sineos what is your feedback on the current implementation of this feature in [adxl345-spi](https://github.com/dmbutyugin/klipper/tree/adxl345-spi)? It is admittedly relatively slow, because it has to run a test for multiple frequencies in a row. I'll see if that process can be improved using the suggestions from @nagimov. Alas, my initial experiments with FFT were not particularly successful - it sort of works, but lacks some robustness (and I'm sure the test process is not good, and I've been also likely making mistakes in how I put a non-periodic signal into FFT). It also appears that the results of the measurements with FFT do not exactly agree with the direct measurements of vibration response with accelerometer - the difference is not large, only a few Hz, but it is nonetheless not expected and quite suspicious. And direct measurements previously showed good agreement with the measurements using simple ringing test. Still, it is not clear which one is more accurate.

This is a dedicated ticket to discuss accelerometers support instead of #2030, which got too large.

BlackStump, 2020-06-27:
I am using it on a dedicated Pi4 2Gb and have had it freeze a few times,  I suspect that is a pigpiod issue as they mention pi4 support is experimental.
It does make tuning a lot easier and faster after the adxl345-spi is installed and as you mention they are cheap to buy.
As a tuning tool well worth it imho.

Sineos, 2020-06-28:
Well, I'm a big fan of measuring things, so I think it is a fantastic addition:

- Tuning S-Curve parameters
- Assessing modification, like resonance pre and post modification
- Scientific interest
- Future: Quality guidance for printers - maybe a database could be setup to compare popular models in terms of their mechanical properties

Installation-wise, it is quite easy and can be done below 10 €/USD. There are even 50cm jumper cable available, which make it even easier. Hardest part is hitting the right pins on the RPi - good eyes and the ability to count surely help.

On my RPi3 I didn't experience any stability issues, running alongside Klipper and OctoPrint.

dmbutyugin, 2020-06-28:
@BlackStump To be fair, there could be some issues with the code stability. It was more like a proof of concept, and does require some rework and fixes. If you run into issues with it - feel free to report them here (and please attach the klippy.log). What I meant was that the code is quite promising and appears to be robust to measurement errors and has good repeatability of measurements - so it appears that we're not just measuring noise.

@Sineos As a follow-up from another thread: my feedback on the measurements after re-doing X axis is you do appear to have resonances. It is just that the test is not designed for very low frequency measurements, and gives bogus results at a few Hz, so the true resonances are lost if those values are not discarded. BTW, I have added now a low limit to the code to forbid measurements below 10 Hz, but realisticaly the good precision can be obtained starting from ~15 Hz I would say.

So, my conclusion is:

- X axis has a resonance at 71.5 Hz and a smaller one at ~49 Hz (and perhaps at 63 Hz).
- Y axis has a resonance at ~58 Hz.

If you end up trying input shapers tuned for these frequencies - please share how it goes. I guess for Y axis you could use 'ei' shaper tuned for 60 Hz and for X axis - 'ei' shaper tuned for 60 Hz (so just EI shaper at 60 Hz for both axes - this should be sufficient in your case), or '2hump_ei' for X axis at 65 Hz (the shapers can be configured per axis with `shaper_type_x` and `shaper_type_y` parameters).

KevinOConnor, 2020-06-28:
FYI, I moved the accelerometers on my m2 printer to locations closer to the carriages and reran some tests.  The graphs below are the combination of two separate runs each (I graphed the two runs together to better show the overlap).

X axis: [image]
Y axis: 
[image]

The results look more clean.  I should run some more tests, but I suspect the X axis belt resonates at ~55hz while the X axis frame resonates at <10 hz .  I suspect the Y axis belt also resonates at ~55hz.  I suspect the linear Z rails holding the cantilevered bed have a tendency to resonate at ~35hz.

Some questions:
1. If the X axis is being tested, does the code ensure that the y axis moves at constant speed?  I suspect the slight 35hz Z axis vibration picked up by the bed accelerometer during the X axis test is due to slight changes in speed of the y.  (And I have a similar suspicion on the 55Hz X axis toolhead vibration during the Y test.)
2. Can you elaborate on what the Y axis is for these graphs?  What units is it in and how is the measurement obtained?

-Kevin

KevinOConnor, 2020-06-28:
> Do you think this is something that makes sense to have support for in Klipper?

I think the functionality is very useful and great to have.  However, given the requirement to solder, crimp, etc., and given that it requires wiring to an rpi, I'd say that it is currently only applicable to experts.  As such, I'm "not in a rush" to merge it.  (As I'd expect experts to also be able to pull down the separate branch with the test code.)  Lets see what feedback we get and go from there.

Thanks.
-Kevin

nagimov, 2020-06-28:
@dmbutyugin @KevinOConnor 
It seems like lower end frequencies on FFT plots are a bit too noisy. This could be caused by too short recording times for FFT to catch lower-end frequencies reliably. E.g. if something shakes with resonance frequency of 1Hz, then for FFT to recognize these periods it needs to analyze multiple oscillations (say 10), which means that recordings must be at least 10 seconds long (the longer the better).

dmbutyugin, 2020-06-28:
@nagimov 

Actually, all the plots posted here and in other ticket did not use FFT (see down below). I separately tried to make a different test - to just swing the toolhead back and forth with a constant speed, and making it to instantly change the direction and speed at the ends - to create a very low frequency periodic step velocity function - but had only limited success processing the results with FFT so far. Your comment to take more oscillations into account does make sense. I will get back to it later and post some results here from that approach - maybe you will have some ideas what I did wrong.

@KevinOConnor 

> The results look more clean. I should run some more tests, but I suspect the X axis belt resonates at ~55hz while the X axis frame resonates at <10 hz . I suspect the Y axis belt also resonates at [...]

It [tries](https://github.com/dmbutyugin/klipper/blob/adxl345-spi/klippy/extras/resonance_tester.py#L309-L313) that, yes. I did not attempt to dump the serial output though, and that would be the best way to confirm the problems or to check there aren't any.  Anyway, the code has to segment the movement to approximate sinusoidal oscillations, but it uses segmentation of 0.0005 sec by [default](https://github.com/dmbutyugin/klipper/blob/adxl345-spi/klippy/extras/resonance_tester.py#L40), which is 2000 Hz.

Also, TBH the movement of a different axis is only needed to make sure that iterative solver will not 'skip moves' of a tested axis at higher frequencies. Otherwise, because the movement is periodic, it can detect | x(t+dt) - x(t) | < step / 2 when dt is still too large and skip the movement that's in between t and t + dt. It might be possible to adjust some constants in iterative solver to prevent that.

>     2. Can you elaborate on what the Y axis is for these graphs?  What units is it in and how is the measurement obtained?

The way the processing works is as follows. The sensor readings are recorded at approx 3200 Hz, which is the highest sampling rate for this sensor - all while the printer is moving. The movement currently is run for 2 seconds, and data is recorded for 2.5 seconds. Then the code finds the beginning of the test in sensor readings, skips 0.4 seconds and takes T = 1.5 seconds (discarding the remaining 0.1 seconds of the actual movement). Then the code calculated the integral `I_k` of `(a_k - avg(a_k))^2` over T = 1.5 seconds, where k = x, y, z. `avg(a_k)` is subtracted to remove biases from the sensor measurements due to earth gravity (all axes can pick up part of the gravity force if the sensor was not mounted precisely).

If the printer followed the sinusoidal movement precisely, the integral `I = a^2 * T / 2`. For oscillator, `I` increases or decreases depending on the frequency. So the code actually reports the value `I_k * 2 / (a^2 * T)` - the dimensionless value, which is a squared module of vibrations excitation, with the 'ideal' value being 1.0.

> EDIT: FYI, this was on code revision [2c9cd98](https://github.com/KevinOConnor/klipper/commit/2c9cd98ed1ce701757af9d60e771e4a176fa6826).

FYI there wasn't much added afterwards. I only changed the code to take not exactly T = 1.5 seconds during integration, but the closest full periods of oscillations below 1.5 seconds. It can slightly affect the calculations at lower frequencies.

dmbutyugin, 2020-06-28:
> I think the functionality is very useful and great to have. However, given the requirement to solder, crimp, etc., and given that it requires wiring to an rpi, I'd say that it is currently only [...]

I was asking more like - right now the code has some rough edges and cut corners, so it may require some babysitting (e.g. restart measurements or even Klipper if something went very wrong). If there isn't much interest in this feature, it's probably not worth investing too much efforts into polishing the code, as its current quality seems to be quite decent for experimentation by experts.

nagimov, 2020-06-29:
@dmbutyugin 

> Actually, all the plots posted here and in other ticket did not use FFT (see down below). I separately tried to make a different test - to just swing the toolhead back and forth with a constant [...]

I've added a little example of using ADXL for vibration monitoring [here](https://github.com/nagimov/adxl345spi#measuring-vibration-spectrums-using-fft), in case if that's of any use for testing/troubleshooting.

Resonant frequencies obtained via FFT are usually much more accurate and robust compared to analyzing the waveform directly. FFT results mostly depend on the accuracy of timekeeping and much less so on the accuracy of absolute values of measured acceleration.

I've tested adafruit's ADXL345 boards with very slow-speed machinery (~1.5Hz), asynchronous AC motors (~60Hz) and even turbopumps (~1000Hz) and it always showed good enough timing accuracy to catch the resonant peaks and sometimes secondary peaks (e.g. vibrations of worn bearing bodies) via FFT.

KevinOConnor, 2020-06-29:
> I did not attempt to dump the serial output though, and that would be the best way to confirm the problems or to check there aren't any.

Okay, I took a look at the batch output (from `g28 ; g1 x100 y100 ; TEST_FREQ AXIS=X FREQ=35`) and the Y axis is effectively moving at constant velocity.  (The batch output process is described at https://www.klipper3d.org/Debugging.html .)  So, I guess my previous theory on bed Z vibrations during the X axis test is wrong.  It seems even the X axis motor on the m2 can excite vibrations in the cantilevered bed.

> Also, TBH the movement of a different axis is only needed to make sure that iterative solver will not 'skip moves' of a tested axis at higher frequencies.

That's odd.  Did you actually run into this problem?  The iterative solver only looks at one axis at a time, so introducing another axis shouldn't change its behaviour.  Also, the iterative solver's search is reset between every move, so it should never skip a step on a cartesian or corexy.  It could skip a step on a delta if a linear move results in a tower changing direction within that move - but that should be rare and even then I think it should only occur when the stepper changes direction before making it fully to the next step position.

> The way the processing works is as follows. ...

Okay, thanks.  So if I understand correctly, super simplified, it's the variance of the measured acceleration data divided by the expected variance.

-Kevin

dmbutyugin, 2020-06-29:
> Okay, I took a look at the batch output (from `g28 ; g1 x100 y100 ; TEST_FREQ AXIS=X FREQ=35`) and the Y axis is effectively moving at constant velocity. (The batch output process is described at [...]

Yep, I meant that. Sorry that you had to do it yourself.
 
> > Also, TBH the movement of a different axis is only needed to make sure that iterative solver will not 'skip moves' of a tested axis at higher frequencies.
> 
> That's odd. Did you actually run into this problem? The iterative solver only looks at one stepper at a time, so introducing another axis shouldn't change its behaviour.

Hmm, now that you mention it... I may be mistaken and may have thought incorrectly about the root cause of the problem. However, I believe adding the movement did improve ringing response. And one other thing: I only personally tested on a delta printer so far, and on that kinematics introducing another axis changes movement of all 3 axes. So, I will try to re-test this theory - but it could be that I got confused.

In the meanwhile, it should be possible to disable the movement on the additional axis completely via MOVE_SPEED parameter, i.e.
```
TEST_VIBRATIONS MOVE_SPEED=0 AXIS=...
```

> It seems even the X axis motor on the m2 can excite vibrations in the cantilevered bed.

It may be possible, if the frame can transmit toolhead vibrations to the bed (e.g. if it insufficiently rigid). Then at a certain frequency (~35-37 Hz) the bed can resonate. Note that at ~36-37 Hz there's also an increased response on Z axis of the toolhead sensor, which probably means that the whole printer shakes in Z direction at that frequency.

> Okay, thanks. So if I understand correctly, super simplified, it's the variance of the measured acceleration data divided by the expected variance.

Yes, that's correct.

KevinOConnor, 2020-06-30:
From a separate thread..

> I can only say that I tried measuring vibration response of a printer with input shapers enabled, and I do run into some performance issues even with the current implementation on shapers with many pulses (e.g. 2hump and 3 hump shapers) - but this may be a not very realistic scenario in practice because vibration testing uses very fine segmentation now with moves having 0.0005 seconds duration (so an input shaper can easily span 50-80 moves).

I wasn't aware it would be valid to run `[resonance_tester]` in combination with `[input_shaper]`.  I'll have to try that on my m2.

-Kevin

dmbutyugin, 2020-06-30:
@KevinOConnor 
> I wasn't aware it would be valid to run `[resonance_tester]` in combination with `[input_shaper]`. I'll have to try that on my m2.

Well, yes, sort of. After my experiments I came to a conclusion that the test process likely needs to be adjusted. As a simple illustration, let's imagine a perfectly tuned ZV shaper, which is 2 pulses 0.5 half a period apart (assuming no damping). Then if we take a sinusoidal vibrations at exactly the resonance frequency, ZV will simply cancel them to 0 (so the toolhead won't move on that axis at all). That's technically correct (reducing the vibrations), but does not give us much to work with :) The only useful information it will give us, in my opinion - whether we missed any additional resonances when we configured the input shaper or not.

A better, more real-life test for input shapers, perhaps, could be moving the toolhead along the side of a small square - there we'll acceleration/deceleration and velocity jumps. I didn't get too far in designing this test however, so therefore I was not advertising this option of running [resonance_tester] with [input_shaper] together too much (yet).

Sineos, 2020-07-02:
FWIW, I tried the "move speed 0" command:

`TEST_VIBRATIONS X=100 Y=100 Z=20 MOVE_SPEED=0 AXIS=X FREQ_START=15 FREQ_END=125 FREQ_STEP=.5 OUTPUT=/tmp/vib-xh-ms0.csv RAW_OUTPUT_FMT=/tmp/accel-xh-freq-ms0-%.2f.csv`

It failed with `Error measuring vibrations: Invalid speed in 'G1'` after the 3rd iteration.

[move_speed0.zip](https://github.com/KevinOConnor/klipper/files/4864637/move_speed0.zip)

dmbutyugin, 2020-07-02:
@Sineos Fair enough, I'll need to fix that one way or another.

ehtnevets, 2020-07-07:
I'm wondering, does the resonance change when the speed changes. My musician side is hearing different tones at different speeds. Sometimes when I get tired playing song, I mimic the sound of my printer on my guitar. Also at different speeds the mechanical experiences different belt tensions and stresses. 

I think Pressure Advance needs different numbers for different speeds. I print small parts at much slower speed and starts seeing a little more oozing than my normal speed. Unfortunately, I don't have the formula on changing PA numbers. Just trial and error.

dmbutyugin, 2020-07-07:
@ehtnevets 
> I'm wondering, does the resonance change when the speed changes.

For linear model of oscillations - no, it does not. However, that model is always an approximation.

> My musician side is hearing different tones at different speeds.

I would think that what you can here is not a sound of resonances. Resonances are typically within 30 - 100 Hz - strictly speaking, audible, but you likely wouldn't here them during normal printer operation (as resonances are excited not all the time, but only near sharp corners, etc., otherwise they quickly die out due to friction). You are more likely to hear mechanical noises from the printer from stepper motors, drivers, belts on pulleys, even wheels or linear guides.

ehtnevets, 2020-07-07:
Thanks D for the explanation.

Sineos, 2020-07-17:
@KevinOConnor 
May I kindly ask you to rethink about integrating the accelerometer support into Klipper:

- It is incredibly useful. Even reworking the printer-head / part cooling mount had a positive influence on resonance. It is so easy to check and optimize with this feature
- Native support would surely increase the acceptance and spread of this feature. People not knowing about @dmbutyugin's work might easily miss it (of course including documentation in Klipper) 
- It would be absolutely unique to Klipper

Many thanks for considering.

dmbutyugin, 2020-07-19:
@Sineos, I think we'd want to do some more experimenting before doing that. In particular, maybe try some slightly different approach at the testing and using FFT for post-processing.

KevinOConnor, 2020-07-20:
FWIW, I looked briefly at the adxl345 spec, and I think it would be possible to query it from the Klipper micro-controller (if C code was written for it).  In particular, it has a 32 entry buffer that can hold past measurements just in case a query is slightly delayed.  Using the regular Klipper mcu system might be a little easier (it would require mcu C code, but shouldn't require any host C code, and it might be easier to wire the device directly to a 3.3V mcu).

Just an observation - not a requirement one way or another.

Cheers,
-Kevin

dmbutyugin, 2020-07-20:
BTW, I've put the setup and measurement instructions for the ADXL345 branch in one place [here](https://github.com/dmbutyugin/klipper/blob/adxl345-spi/docs/Measuring_Resonances.md).

@KevinOConnor,
> it has a 32 entry buffer that can hold past measurements just in case a query is slightly delayed

Yes, that is actually a very promising feature that could improve the robustness of the measurements and reduce the load on the Klipper host and Raspberry Pi in general. It may even enable more complicated testing scenarious.

> I think it would be possible to query it from a Klipper micro-controller

Are you referring to [this](https://github.com/KevinOConnor/klipper/blob/master/docs/RPi_microcontroller.md) feature (RPi as a secondary MCU)? Or are you suggesting to connect ADXL345 to the printer board directly? If the latter, I'm a bit worried that it may still have performance issues even with the buffer, plus it might be a bit less convenient to connect and disconnect the accelerometer to printer board. The former is more promising. But it might be the case that the controller-side C code can be written such that it will work in both setups, with the help of some host-side configuration facilities (python-based).

KevinOConnor, 2020-07-21:
> Are you referring to this feature (RPi as a secondary MCU)? Or are you suggesting to connect ADXL345 to the printer board directly?

Both.  If low-level query code is added to the micro-controller (specifically, if it uses the `spidev_transfer()` helper code) then it should run on the rpi, the beaglebone, and all the other micro-controllers.  (Well, the AVR is unlikely to work without a level-shifter because of 5V vs 3.3V.)

> I'm a bit worried that it may still have performance issues even with the buffer

Well, it's only 3000 queries a second, which isn't really that much.  Getting the data back to the host may be a pain though as the IO code wasn't really designed for bulk data.

-Kevin

The-Monkey-King, 2020-07-21:
> BTW, I've put the setup and measurement instructions for the ADXL345 branch in one place [here](https://github.com/dmbutyugin/klipper/blob/adxl345-spi/docs/Measuring_Resonances.md).
> 
@dmbutyugin 
For bed slingers like the Ender 3/CR series, would it be best to have two accelerameters, one for hotend (basically the X axis) and the other for the bed (just the Y axis)? 
 
> Are you referring to [this](https://github.com/KevinOConnor/klipper/blob/master/docs/RPi_microcontroller.md) feature (RPi as a secondary MCU)? Or are you suggesting to connect ADXL345 to the printer [...]

My thoughts are that you could wire directly to the board, especially boards that allow for enabling/disabling the microPlyer feature in TMC2130, TMC2208/9. As for programming, the printer side would only need to record and store acceleration data of current and (n) previous deltas. The controller or host side would do the heavy lifting of query acceleration data, parse sub-nominal data according to tuning presets, and then pass updated commands back down to printer board. Yes?

dmbutyugin, 2020-07-21:
@KevinOConnor 
> Both. If low-level query code is added to the micro-controller (specifically, if it uses the `spidev_transfer()` helper code) then it should run on the rpi, the beaglebone, and all the other micro-controllers.

I think it makes sense. For experimentational code it's probably fine as-is, but it would be a nice to use the existing, more generic facilities, and get more broad support as a result.

> (Well, the AVR is unlikely to work without a level-shifter because of 5V vs 3.3V.)

FWIW, it seems that at least Adafruit ADXL345 boards can be connected to a 5V source (via VIN):
https://learn.adafruit.com/adxl345-digital-accelerometer/assembly-and-wiring#i2c-wiring-326570-6
I didn't test this setup though.

> Well, it's only 3000 queries a second, which isn't really that much.

Well, maybe, it's roughly 3200 * 7 * 8 = 179200 Hz, so reads at 1.6 MHz SPI frequency are roughly ~11% of the wall time (e.g. AVR SPI implementation [seems](https://github.com/KevinOConnor/klipper/blob/master/src/avr/spi.c#L89) to be blocking).

> Getting the data back to the host may be a pain though as the IO code wasn't really designed for bulk data.

So we'd need to read about 180 KB/sec, not sure if that's a lot.

@The-Monkey-King 

> @dmbutyugin
> For bed slingers like the Ender 3/CR series, would it be best to have two accelerameters, one for hotend (basically the X axis) and the other for the bed (just the Y axis)?

As I mentioned in the docs, it is indeed necessary to run 2 tests: one with the accelerometer mounted to the toolhead, and one - mounted to the bed. Whether you buy 2 accelerometers, or just move 1 around is up to you.

> My thoughts are that you could wire directly to the board, especially boards that allow for enabling/disabling the microPlyer feature in TMC2130, TMC2208/9. As for programming, the printer side [...]

If we use the existing Klipper MCU-side facilities, it should work in such a setup too. Well, modulo potential performance issues; I'm not sure if I will be able to test this setup myself. I do have RUMBA board which has SPI, but I already have TMCs5160 connected to it.

jdlongenecker, 2020-07-27:
this is really cool--while I'm not able to help out with code I'll definitely try to test it out. At least on initial results, I've been blown away by the improvements from [input_shaper] and this looks like a really cool next test! Thanks for the great work.

KevinOConnor, 2020-07-30:
@dmbutyugin - FYI, I took a look at querying the ADXL345 from the Klipper mcu code.  It's available in the work-adxl345-20200730 branch ( https://github.com/KevinOConnor/klipper/tree/work-adxl345-20200730 ).

I was able to query out the sensor using the Linux mcu (as described at https://github.com/KevinOConnor/klipper/blob/work-adxl345-20200730/docs/RPi_microcontroller.md ).  The config snippet is just:
```
[mcu rpi]
serial: /tmp/klipper_host_mcu

[adxl345 my_accel]
cs_pin: rpi:None
```

With this, one can grab the raw data using:
```
ACCELEROMETER_MEASURE CHIP=my_accel RATE=3200
... do something interesting ...
ACCELEROMETER_MEASURE CHIP=my_accel RATE=0
```

The raw data is just written to the klippy.log file on the `RATE=0` request.

This support is very raw (and not particularly useful right now).  However, it does seem the code is capable of querying out 3200 samples per second when using a raspberry pi mcu host.  I was seeing about 10% cpu usage for both klippy and the linux mcu.  I'm pretty sure one would not be able to do 3200sps on an mcu using a UART, but I suspect it would work on an ARM chip that has native USB.  (It's also possible an AVR with UART would work okay at a slower (eg, RATE=400) mode.)

One nice feature of performing the samples in the mcu code is that we can get very accurate start/stop times for the measurements.

Thoughts?
-Kevin

jdlongenecker, 2020-07-30:
this would be fantastic @KevinOConnor. 

I used the adxl345 guide linked in @dmbutyugin's measuring resonance document, installed it on a separate Pi and connected the accelerometer to that.

On the main Pi driving the printer, I set up a macro to run the toolhead in a square tool path, 50mm/s, at 100mm/s. I started that macro, and then immediately started a 2 second 3200Hz scan of the accelerometer on the second Pi. This allowed me to measure the accelerometer output at the same time the toolhead was moving.

I did 3 runs, one without input shaping, one with input shaping at x=37Hz and y=36Hz, and one at the accelerometer tuned x=31Hz and y=20Hz. Here are plots of X, Y, and Z. x, x`, and x`` are these three tests respectively. 

[image]

[image]

[image]

And finally, a plot of X and Y resonance together:
[image]

If we could tie tool path/target positioning with accelerometer data, could we create a displacement model to overlay target/actual tool paths and show ringing reduction that way?

dmbutyugin, 2020-07-30:
@KevinOConnor Thanks, that's really promising as a proof-of-concept! I think the resonance testing code can use some generic interface to ADXL345 class.

>  I was seeing about 10% cpu usage for both klippy and the linux mcu.

Interesting. I've tried to implement FIFO (1d53115978a2347154e6bd48d55a28769c6a0b00). I think it works OK (but I need to do more tests to be sure). However, I did not observe a significant reduction in cpu taken by pigpiod (~35%) and Klipper (~25%). Maybe I did something wrong though.

For your code I only had a minor comment that, according to the specs, it is necessary to have 5 usec delay between FIFO read and the next read or querying FIFO_STATUS register. At 5 MHz SPI speed there is insufficient delay between the read of the last Z register and reading FIFO_STATUS register now (reading FIFO_CTL register only: 8 bits / 5000000 = 1.6 usec).

dmbutyugin, 2020-07-30:
@jdlongenecker My thinking is that we can make the toolhead go over a segment (line) back and forth and have it make a velocity jump at the ends of the segment. Then we can check the vibrations with and without the input shaper. But mapping measured (and noisy) acceleration data to a toolpath might be a bit more problematic.

jdlongenecker, 2020-07-30:
makes sense, I was looking at some info that essentially said even a small error in the accelerometer values stacks up significantly once it turns into velocity and displacement--probably a bit too much to ask of an inexpensive sensor and imperfect mounting 👍 

It would be cool if theoretically you could mount the accelerometer, and then run a script/macro that essentially iteratively auto tunes the settings to optimize the response.

dmbutyugin, 2020-07-31:
@jdlongenecker
> It would be cool if theoretically you could mount the accelerometer, and then run a script/macro that essentially iteratively auto tunes the settings to optimize the response.

 That would be great, of course. But it may take a while before we get there :) Though we may be able to automate the resonance detection and tuning, with less human intervention required.

KevinOConnor, 2020-07-31:
> For your code I only had a minor comment that, according to the specs, it is necessary to have 5 usec delay between FIFO read and the next read or querying FIFO_STATUS register.

Thanks for reviewing.  I did see that in the spec, but the wording is actually, `To ensure that the FIFO has completely popped (that
is, that new data has completely moved into the DATAX, DATAY,
and DATAZ registers), there must be at least 5 µs between the
end of reading the data registers and the start of a new read of
the FIFO or a read of the FIFO_STATUS register (Address 0x39).`  I take that to mean it is okay to read FIFO_STATUS in a burst, but it may reflect the past state or (depending on timing) the new state.  That's why the code checks for "fifo_entries > 1" before querying again, as a 1 may be the data just read.  But anything more than 1 should indicate there's data buffered.

That said, I still need to verify the data.

> However, I did not observe a significant reduction in cpu taken by pigpiod (~35%) and Klipper (~25%).

That might be overhead in pigpiod.

@jdlongenecker 
> If we could tie tool path/target positioning with accelerometer data, could we create a displacement model to overlay target/actual tool paths and show ringing reduction that way?

It should be possible to align the accelerometer responses with the internal "velocity trapezoid queue", but that would be a lot of work.  So, it's possible, but probably not a short term project.

-Kevin

KevinOConnor, 2020-07-31:
FYI, I was able to make some further enhancements and run further tests on the work-adxl345-2020730 branch.

The config and commands work as before ( https://github.com/KevinOConnor/klipper/issues/3027#issuecomment-666742653 ).  However, now the data is written to a file named like `/tmp/adxl345-YYYYMMDD_HHMMSS.csv`.  I also put together a graphing script (requires matplotlib): `./scripts/graph_adxl345.py adxl345-YYYYMMDD_HHMMSS.csv`.

It looks like the query code is stable - I added more diagnostics to try to catch lost or corrupted data and I haven't seen any indication of a problem.  I'm getting a very reliable sample rate of .000319s per sample (which isn't quite 3200sps, but I'm guessing the difference is just due to clock drift).

Here's a graph of a homing operation on the M2:
[image]

This is with the sensor on the bed, so the X movement isn't very noticeable.  However, the Y and Z homing can be clearly seen.  I'm not sure if this sensor is just noisy, or if the bed really does vibrate that much even during constant velocity.

If anyone else wants to try:
`cd ~/klipper ; git fetch ; git checkout origin/work-adxl345-20200730 ; sudo service klipper stop ; make flash ; sudo service klipper start`
The acceleration data is currently stored in memory, so don't run measurements for more than a minute or so at a time.

-Kevin

KevinOConnor, 2020-07-31:
I took a look at the "noise" during the constant speed Y homing moves.  It appears to be a ~275hz signal - which matches the full step rate for the Y axis during homing.  Similarly, during the Z second home phase there is a ~500hz signal - which matches the Z leadscrew micro-step homing rate.

So, interestingly, it appears the accelerometer is able to pick up the "noise" from the stepper motors steps.

-Kevin

pengsloth, 2020-07-31:
Hello everyone,

I've been watching this conversation for a while and was able to hook up an ADXL345 to my modded (direct drive) Ender 3 Pro. The resonances I measured seem pretty clear to me (about the same in both axis, 38.6Hz in Y and 39.2Hz in X at peak). What's lost on me is understanding which Shaping algorithm I should use (MZV vs EI etc)

[image]

Any advice or clarification would be helpful.

dmbutyugin, 2020-07-31:
@pengsloth Well, the accelerometer only helps you find the resonances. Sine you only have 1 strong resonance per axis, you can just use these results in lieu of [this print](https://github.com/KevinOConnor/klipper/blob/master/docs/Resonance_Compensation.md#ringing-frequency). To choose the shaper, refer to the corresponding [section](https://github.com/KevinOConnor/klipper/blob/master/docs/Resonance_Compensation.md#choosing-input-shaper) of the documentation - you will still need to print the test model a couple of times. In general, a quote from there: "EI shaper may be more suited for bed slinger printers (if the resonance frequency and resulting smoothing allows)"

dmbutyugin, 2020-07-31:
@KevinOConnor 

> That's why the code checks for "fifo_entries > 1" before querying again, as a 1 may be the data just read. 

Ah OK, then it was simply my confusion.

> That might be overhead in pigpiod.

Yes, very possible.

> I'm getting a very reliable sample rate of .000319s per sample (which isn't quite 3200sps, but I'm guessing the difference is just due to clock drift).

So the difference is within +/- 2%. It is probably due to small inaccuracy of the internal clocks (most probably the one on ADXL345). I do not think it will affect the quality of the resonance detection (within that accuracy range, naturally). It might be easier just to assume the stable rate of exactly 3200 Hz.

FWIW, I tried your branch and got the basic measurements successfully. Though in my case I am getting consistently time_per_sample=0.000324. Which is almost 4% error.

> I took a look at the "noise" during the constant speed Y homing moves. It appears to be a ~275hz signal - which matches the full step rate for the Y axis during homing. Similarly, during the Z [...]

Yes, indeed! I noticed that kind of signal during my tests too - though I was not sure if this a noise from the steppers (most likely), pulleys, or even linear guide rails (balls inside the carriage). And on a delta the signal has much less stable frequency. BTW, it does not appear to affect the ability of the code to measure the response (and find resonances), which is good.

In general, I would propose to test the reliability more and then integrate your ADXL345 code into the mainline. I will change resonance_tester.py code to use adxl345.py (with some nicer API) then. Afterwards, we can try adjusting the testing procedure, since we'll have more freedom of what kind of tests we can schedule and measure. I think we anyways agreed that the code to handle accelerometer reads should be something like what you have implemented. Or I can try doing that on top of your branch - but for the next couple of weeks I'll be away from my printer, so I won't be able to run the tests 'in hardware' myself.

KevinOConnor, 2020-07-31:
> It might be easier just to assume the stable rate of exactly 3200 Hz.

It's easy to measure the actual rate (`total_time / total_samples`), so doesn't hurt to use more precise timing.  I agree it's unlikely to impact the resonance testing though.

FYI, I made some additional tweaks on the work-adxl345-20200730 branch to further improve the calculation of "total_time".  It should now be able to calculate that to within about 100us.

> Though in my case I am getting consistently time_per_sample=0.000324. Which is almost 4% error.

Interesting.  Did the final csv file show `limit_count=0` and (at the end of the file) `drops=0`?  If you didn't get any drops and there was no indication of an overrun then I'd say it's most likely the adxl345 clock running a little slow.  If it consistently runs slow (that is, if several tests all show .000324) then it shouldn't be a problem.

> In general, I would propose to test the reliability more and then integrate your ADXL345 code into the mainline. I will change resonance_tester.py code to use adxl345.py (with some nicer API) then. 

Okay, thanks.  I'll clean up the code and open a PR for work-adxl345-20200730.  Key to this support will be the resonance testing feature.  We can merge that when you're ready though.

-Kevin

dmbutyugin, 2020-07-31:
> It's easy to measure the actual rate (`total_time / total_samples`), so doesn't hurt to use more precise timing. I agree it's unlikely to impact the resonance testing though.

TBH, 4% error may already impact ZV and even MZV shaper tuning (though less for MZV). So, as long as we are confident about the measurements (I can imagine that the total_time may be computed with an error of ~2 / 3200 sec, so cumulatively the error should be small), then it's probably better to get as accurate measurement rate as possible.

> Interesting. Did the final csv file show `limit_count=0` and (at the end of the file) `drops=0`? If you didn't get any drops and there was no indication of an overrun then I'd say it's most likely [...]

Yes, it was really consistent. The latest attempt:
```
##start=35.848551/35.848652,end=59.308862/59.308891
##limit_count=0,end_seq=9065,time_per_sample=0.000323527
#time,x,y,z
35.848652,0.627626,-0.156906,9.885103
35.848976,0.078453,0.706079,9.414384
...
59.308244,0.313813,0.627626,9.257478
59.308568,0.235360,0.470719,9.728197
##count=72514/72514,drops=0
```

> Okay, thanks. I'll clean up the code and open a PR for work-adxl345-20200730. Key to this support will be the resonance testing feature. We can merge that when you're ready though.

OK, sounds good. Agreed that it's only useful with the resonance testing.

jdlongenecker, 2020-08-10:
@KevinOConnor it would be helpful if you could define your accelerometer orientation relative to your printer axes, like:
```axis_alignment: zyx ```
if your z and x axes are swapped, or
```axis_alignment: xyz```
 if they match the accelerometer. Then the data output would be correct without post processing.

I'll share my before/after data tomorrow here--I took a baseline before swapping from Dmitri's branch and am redoing the same data points to make sure everything looks good!

Thanks

jdlongenecker, 2020-08-11:
@KevinOConnor here's before (Dmitri's fork) and after (this pull request) data for X, Y, A, and B on a CoreXY printer. I believe this is well within the standard run to run variance!

[image]

[image]

[image]

[image]

jdlongenecker, 2020-08-11:
I've created two basic macros for testing data on my printer, including one that measures vibrations when the toolhead has stopped. Raw data, basic instructions, and macros can be found here:

https://github.com/jdlongenecker/documentation/tree/master/vibration_data

Results with/without input shaper:
[image]

[image]

dmbutyugin, 2020-08-30:
@KevinOConnor I ran some tests with your branch `work-adxl345-20200730` and it seems to work fine. I think it makes sense to target that ADXL345 code, and do some cleanups of the `resonance_tester.py`. One big advantage of adxl345 support from your branch is that now the host code can handle long sequences of moves, not just a few seconds. This gives an opportunity to run the test for a few minutes and still get all readings without the gaps between the movements.

Separately, I started some experiments with slightly different resonance tests. For one, I updated `TEST_RESONANCES` to work properly. Note that it requires some post-processing with FFT now, but the measurement process is, perhaps, more robust. If some of you folks could give it a try, you can check out [this branch](https://github.com/dmbutyugin/klipper/tree/work-adxl345-20200730), [configure](https://github.com/dmbutyugin/klipper/blob/work-adxl345-20200730/docs/Measuring_Resonances.md#software-installation) it as appropriate, and run the test
```
TEST_RESONANCES X=100 Y=100 Z=20 AXIS=X RAW_OUTPUT=/tmp/accel-data-x.csv
```
(X, Y and Z values are just examples here). Then download the file `/tmp/accel-data-x.csv`, and run the [following script](https://github.com/dmbutyugin/klipper/blob/work-adxl345-20200730/scripts/specgram.py), e.g.
```
$ python scripts/specgram.py accel-data-x.csv
```
It can plot a chart like this
[image]

The results of the measurements are very comparable to the old method, but I think it works better for low frequencies:
[image]

If this new test works well for others, I want to see if further advancements are possible.

jdlongenecker, 2020-08-30:
Hello @dmbutyugin!

I was able to test this tonight! 

Basic feedback for someone not super savvy:
-Was able to switch to your branch without changing anything other than checking it out. Nice and easy (had been on the respective klipper branch before)
-Test ran much faster than the previous version which is a welcome change!
-After it reached 100Hz, it said the following and I wasn't sure if it was still working or not (it was, just took a bit to process I guess--I didn't realize the file would be so large!
`
Recv: // Testing frequency 97 Hz
Recv: // Testing frequency 98 Hz
Recv: // Testing frequency 99 Hz
Recv: // Testing frequency 100 Hz
Communication timeout while idle, trying to trigger response from printer. Configure long running commands or increase communication timeout if that happens regularly on specific commands or long moves.
Send: M105
Recv: // ADXL345 stats: drops=0,overflows=0,time_per_sample=0.000304702,start_range=0.000041,end_range=0.000048
Recv: ok
Send: M105
Recv: ok B:0.0 /0.0 C:22.9 /0.0 T0:0.0 /0.0
`

When trying to run the python script as you mentioned (after changing accel-data-x.csv tot /tmp/accel-data-x.csv), I get the following error:
`pi@miniV:~/klipper $ python scripts/specgram.py accel-data-x.csv
Traceback (most recent call last):
  File "scripts/specgram.py", line 98, in <module>
    main()
  File "scripts/specgram.py", line 88, in main
    fig = plot_accel(data, args[0])
  File "scripts/specgram.py", line 56, in plot_accel
    fig, ax = plt.subplots()
  File "/usr/lib/python2.7/dist-packages/matplotlib/pyplot.py", line 1203, in subplots
    fig = figure(**fig_kw)
  File "/usr/lib/python2.7/dist-packages/matplotlib/pyplot.py", line 535, in figure
    **kwargs)
  File "/usr/lib/python2.7/dist-packages/matplotlib/backends/backend_tkagg.py", line 81, in new_figure_manager
    return new_figure_manager_given_figure(num, figure)
  File "/usr/lib/python2.7/dist-packages/matplotlib/backends/backend_tkagg.py", line 89, in new_figure_manager_given_figure
    window = Tk.Tk()
  File "/usr/lib/python2.7/lib-tk/Tkinter.py", line 1823, in __init__
    self.tk = _tkinter.create(screenName, baseName, className, interactive, wantobjects, useTk, sync, use)
_tkinter.TclError: no display name and no $DISPLAY environment variable
`

Both of these may be trivial to fix and I may be missing something obvious :) but I figured I'd give my experience regardless.

I've attached 4 tests in a Zip folder - your branch for all 4. X and Y with TEST_RESONANCES, and X and Y with TEST_VIBRATIONS.
[Archive.zip](https://github.com/KevinOConnor/klipper/files/5146337/Archive.zip)

Sineos, 2020-08-30:
I thought I give it a run as well.

@jdlongenecker 
The error you are getting is due to running the script (via SSH?) on the PI. The script tries to open an interactive Xwindows, which will fail on this device. Try running on Windows or Linux with a desktop environment.

**Results:**

**"Traditional" Way X** 
[image]

**FFT X**
[image]

**"Traditional" Way Y**
 [image]

**FFT Y**
[image]

**Data Package:**
https://drive.google.com/file/d/1N8_UODsMVXwOLGF0KUM51u8ySJBq4B5J/view?usp=sharing

dmbutyugin, 2020-08-30:
Hi, thanks! FWIW, the results do seem very comparable between both tests, but the new test does not emphasize the vibrations at the low end of the spectrum. That, I think, is happening because low-frequency vibrations can excite higher-frequency vibrations due to small non-linear effects, but the old test just captures the increased amplitude, and the new test attributes these vibrations to the correct higher frequency oscillations.

Indeed, the new test runs much faster. But the raw output is much larger now than before (a few tens of MB), because it actually contains all the accelerometer readings during the test. As a side effect, it takes a bit at the end to write the file down to SD card, but no need to worry - it only takes a few dozens of seconds.

@jdlongenecker 
> When trying to run the python script as you mentioned (after changing accel-data-x.csv tot /tmp/accel-data-x.csv), I get the following error:

Yes, that script should be executed on your desktop/laptop. Alternatively, you could run it as
```
$ python ~/klipper/scripts/specgram.py /tmp/accel-data-x.csv -o plot-x.png
```
which will write `plot-x.png` that can be downloaded from Raspberry Pi.

@Sineos I see that your results on X axis abruptly end at ~100 Hz, which is the end of the test as well. Could you run the test also with the higher end frequency (also updating to the latest code from the branch)?
```
TEST_RESONANCES X=100 Y=100 Z=20 AXIS=X FREQ_END=150 RAW_OUTPUT=/tmp/accel-data-x.csv
```
I actually do not expect your results to change, but just want to double-check.

Separately, your X test shows more cross-axes excitation near the resonance in FFT test than in the 'traditional' test. I wonder why is that.

Sineos, 2020-08-30:
@dmbutyugin 
As requested - for my eyes they look quite different, especially on the "untested" axes

**X up to 150 "traditional":**
[image]

**X up to 150 "FFT":**
[image]

**Data Package**
https://drive.google.com/file/d/1wRSfALPGsy2j6IF91-72MNPE-n57W2BN/view?usp=sharing

It would be helpful when the FFT script also outputs a csv with the processed values since analysis with a spreadsheet tool like Excel or similar is easier than only the graphical approach of the python engine.

KevinOConnor, 2020-08-31:
@dmbutyugin - FYI, I ran a test with TEST_RESONANCES (just for Y and just with accelerometer on the corner of the bed):
```
set_input_shaper shaper_freq_x=0 shaper_freq_y=0
set_velocity_limit accel=2000
g28
g1 x100 y100 z20 f600
set_velocity_limit accel=7000
TEST_VIBRATIONS X=100 Y=100 Z=20 AXIS=Y FREQ_START=10 FREQ_END=100 FREQ_STEP=0.5 OUTPUT=/tmp/vib-y.csv
TEST_RESONANCES X=100 Y=100 Z=20 AXIS=Y RAW_OUTPUT=/tmp/accel-data-y.csv
```

I'm not sure this printer is a good test case, but old and new do seem to show a lot of overlap.

Old:
[image]
New:
[image]
New test, raw data graph:
[image]

-Kevin

dmbutyugin, 2020-09-01:
@KevinOConnor Thanks! Would you mind archiving and sharing the raw file with accel data? I am testing some autocalibration code now, and I think your case is quite interesting.

KevinOConnor, 2020-09-01:
@dmbutyugin - I'm not sure this printer is a good test case, but feel free to take a look:

[accel-data-y.csv.gz](https://github.com/KevinOConnor/klipper/files/5153393/accel-data-y.csv.gz)
[vib-y.csv.gz](https://github.com/KevinOConnor/klipper/files/5153394/vib-y.csv.gz)
[klippy.log.2020-08-31.gz](https://github.com/KevinOConnor/klipper/files/5153400/klippy.log.2020-08-31.gz)

-Kevin

dmbutyugin, 2020-09-01:
Well, in this case the autocalibration code recommends 3hump_ei shaper at 58.3 Hz for Y axis:
[image]

dmbutyugin, 2020-09-01:
FWIW, this is how other fitted shapers look like in this case:
[image]
Fitted shaper 'zv' frequency = 65.2 Hz
Fitted shaper 'mzv' frequency = 41.4 Hz
Fitted shaper 'ei' frequency = 59.8 Hz
Fitted shaper '2hump_ei' frequency = 55.2 Hz
Fitted shaper '3hump_ei' frequency = 58.3 Hz

KevinOConnor, 2020-09-01:
That's really useful!  Any suggestions on a good way to test those values?  Visually, it does look like 2hump_ei is pretty good as well.

FYI, I moved the wires to the accelerometer on the toolhead and tested X with that.

Old (axes are correct):
[image]

New (y and z are swapped):
[image]

New (raw; y and z are swapped):
[image]

Interesting that this does differ a bit from the old method.

[klippy.log.gz](https://github.com/KevinOConnor/klipper/files/5153606/klippy.log.gz)
[vib-x-th.csv.gz](https://github.com/KevinOConnor/klipper/files/5153608/vib-x-th.csv.gz)
[accel-data-x.csv.gz](https://github.com/KevinOConnor/klipper/files/5153610/accel-data-x.csv.gz)

-Kevin

dmbutyugin, 2020-09-01:
OK, so I pushed the automatic shaper calibration code to my [test branch](https://github.com/dmbutyugin/klipper/tree/work-adxl345-20200730). @KevinOConnor I slightly reworked the logic to compare different input shaper, and now for your Y axis data it picks 2hump_ei shaper. For X axis the results are as follows:
[image]
Fitted shaper 'zv' frequency = 68.2 Hz (vibrations = 30.6%)
Fitted shaper 'mzv' frequency = 49.9 Hz (vibrations = 13.8%)
Fitted shaper 'ei' frequency = 68.2 Hz (vibrations = 12.2%)
Fitted shaper '2hump_ei' frequency = 63.4 Hz (vibrations = 6.2%)
Fitted shaper '3hump_ei' frequency = 61.8 Hz (vibrations = 3.9%)
Recommended shaper is 2hump_ei @ 63.4 Hz

BTW, I updated the chart format, let me know what you think. It has more technical details now, but maybe it is too 'loaded' now and harder to read.

There are 2 ways to try the new shaper tuning algorithm. First, if the raw data is generated using the command like
```
TEST_RESONANCES X=100 Y=100 Z=20 AXIS=X RAW_OUTPUT=/tmp/accel-data-x.csv
```
then one can run the following command on their Pi:
```
$ ~/klippy-env/bin/python ~/klipper/klippy/extras/shaper_calibrate.py /tmp/accel-data-x.csv -o /tmp/shaper_calibration_x.png
```
and it will print the output like above for the shapers, as well as will generate the png with the graphical results.

There's one even better way now. One can run the command in Octoprint
```
SHAPER_CALIBRATE PROBE_POINTS=100,100,20 AXES=XY RAW_OUTPUT=/tmp/accel-data.csv FIG_BASENAME=/tmp/shaper_calibration.png
```
Parameters `RAW_OUTPUT` and `FIG_BASENAME` are, in principle, optional, but if you try this, please generate these outputs for now too (they are useful for debugging and such). `AXES=XY` is a default and can be omitted too (so it makes sense to set `AXES=X` or `AXES=Y` in case of a separate testing). `PROBE_POINTS` can accept multiple points, e.g. if you want to test a few points on the buildplate, you can set, for instance, `PROBE_POINTS=100,100,20:50,100,20:-50,100,20:100,50,20:100,-50,20`, and the `SHAPER_CALIBRATE` will measure the vibration responses at all these points and use them in the calculations. After completion, it will print the tested and suggested parameters for the shaper, and you can execute `SAVE_CONFIG` to save the parameters if you are satisfied with the automatic choice. In case of a bed slinger printer the sensor must be mounted in different locations for X and Y axes tests, so for now one has to run the test twice for X and Y axes separately, re-mounting or reconnecting the adxl345 sensor in the meanwhile.

Note that shaper autocalibration requires additional dependencies not installed by default. You will have to run on your Raspberry Pi
```
$ ~/klippy-env/bin/pip install matplotlib
```
first time prior to running either of the methods.

Some other interesting calibration results:

- @jdlongenecker from your Y axis data, suggested is 3hump_ei @ 51.2 Hz (and 2hump_ei @ 38.4 Hz for X axis):
[image]

- @Sineos from your X axis data, suggested is 2hump_ei @ 95.4 Hz (and 2hump_ei @ 78.7 Hz for Y axis):
[image]

Now, how to test the tuned parameters? I don't know :) But I suppose it would be better experimentally, to simply try and print the resonance test and see if there is still some ringing left.

As a side note, I do not claim that the calibration code does physically sound calculations for each of the input shapers, but at least it appears to be picking good frequencies for each shaper. The algorithm can be rough around the edges for now, and also may not produce exactly repeatable results run-to-run: for different runs the measured data can be somewhat different, and this may make the tuning algorithm pick slightly different parameters (and even different shapers). Hopefully the ringing suppression should not change too much from calibration to calibration. But this is one of the reasons I wouldn't recommend one to make this tuning test an everyday procedure.

KevinOConnor, 2020-09-02:
> Now, how to test the tuned parameters? I don't know :) But I suppose it would be better experimentally, to simply try and print the resonance test and see if there is still some ringing left.

Just for kicks, I reran TEST_RESONANCES after setting an input shaper:
```
  set_input_shaper shaper_freq_x=63.4
  set_velocity_limit accel=2000
  g28
  g1 x100 y100 z20 f600
  set_velocity_limit accel=7000
  TEST_RESONANCES X=100 Y=100 Z=20 AXIS=X RAW_OUTPUT=/tmp/accel-data-x2.csv
```

What I thought was interesting is that the predictions for remaining vibrations was actually quite good.

Predicted with 2hump_ei/63.4:
[image]
Measured with 2hump_ei/63.4:
[image]

The actual magnitude for the remaining frequencies between 50-100hz is about 4x the prediction, but the peaks do line up.  (And, it's still a ~10x total reduction in magnitude.)

-Kevin

agsolino, 2020-09-02:
Hey @dmbutyugin:

Running `TEST_RESONANCES` standalone works fine and I get the automatic calibration data. However, when running it from OctoPrint by running `SHAPER_CALIBRATE PROBE_POINTS=100,100,20 AXES=XY RAW_OUTPUT=/tmp/accel-data.csv FIG_BASENAME=/tmp/shaper_calibration.png` I get an error:
```
stepcompress o=6 i=0 c=74 a=0: Invalid sequence
stepcompress o=6 i=0 c=74 a=0: Invalid sequence
Internal error on command:"SHAPER_CALIBRATE"
Traceback (most recent call last):
  File "/home/pi/src/klipper/klippy/gcode.py", line 177, in _process_commands
    handler(gcmd)
  File "/home/pi/src/klipper/klippy/gcode.py", line 115, in <lambda>
    func = lambda params: origfunc(self._get_extended_params(params))
  File "/home/pi/src/klipper/klippy/extras/resonance_tester.py", line 240, in cmd_SHAPER_CALIBRATE
    freq_end, hz_per_sec, gcmd)
  File "/home/pi/src/klipper/klippy/extras/resonance_tester.py", line 370, in _run_test2
    toolhead.move([nX, nY, Z, E], V)
  File "/home/pi/src/klipper/klippy/toolhead.py", line 415, in move
    self.move_queue.add_move(move)
  File "/home/pi/src/klipper/klippy/toolhead.py", line 182, in add_move
    self.flush(lazy=True)
  File "/home/pi/src/klipper/klippy/toolhead.py", line 171, in flush
    self.toolhead._process_moves(queue[:flush_count])
  File "/home/pi/src/klipper/klippy/toolhead.py", line 331, in _process_moves
    self._update_move_time(next_move_time)
  File "/home/pi/src/klipper/klippy/toolhead.py", line 284, in _update_move_time
    sg(sg_flush_time)
  File "/home/pi/src/klipper/klippy/stepper.py", line 171, in generate_steps
    raise error("Internal error in stepcompress")
error: Internal error in stepcompress
```
Attaching  `klippy.log` for your reference. 
[klippy.log](https://github.com/KevinOConnor/klipper/files/5163132/klippy.log)

Sineos, 2020-09-02:
Same issue here
[klippy.log](https://github.com/KevinOConnor/klipper/files/5164121/klippy.log)

dmbutyugin, 2020-09-02:
That is strange, what I can see is that the error (first) is actually "Timer too close". I also understand that it first tests vibrations on one axis, X, and then fails, right?

@KevinOConnor I also see somewhat strange timings in the logs (e.g. from Sineos):
```
Receive: 92 300409.044036 300409.042549 11: seq: 19, clock clock=416644386
Receive: 93 300409.141670 300409.042549 15: seq: 19, analog_in_state oid=10 next_clock=463621376 value=31481
.....................
Receive: 98 300409.862298 300409.042549 15: seq: 19, analog_in_state oid=22 next_clock=550021376 value=31443
Receive: 99 300409.946182 300409.944839 12: seq: 1a, shutdown clock=524921845 static_string_id=Timer too close
Transition to shutdown state: MCU 'mcu' shutdown: Timer too close
.....................
Dumping send queue 100 messages
Sent 0 300234.342101 300234.342101 6: seq: 12, get_clock
........................
Sent 12 300246.337860 300246.337860 6: seq: 1e, get_clock
Sent 13 300246.814835 300246.814835 9: seq: 1f, query_adxl345 oid=1 clock=0 rest_ticks=0
Sent 14 300332.226261 300332.226261 10: seq: 10, spi_transfer oid=0 data='\x80\x00'
Sent 15 300332.226910 300332.226910 6: seq: 11, get_clock
Sent 16 300332.234350 300332.234350 10: seq: 12, spi_send oid=0 data='1\x0b'
Sent 17 300332.234454 300332.234454 10: seq: 13, spi_send oid=0 data='8\x80'
Sent 18 300332.234507 300332.234507 10: seq: 14, spi_send oid=0 data=',\x0f'
Sent 19 300332.728335 300332.728335 15: seq: 15, query_adxl345 oid=1 clock=371073417 rest_ticks=62500
Sent 20 300333.211251 300333.211251 6: seq: 16, get_clock
...............................
Sent 98 300410.031605 300410.031605 6: seq: 14, get_clock
Sent 99 300410.034252 300410.034252 6: seq: 15, emergency_stop
```
(I removed some repetitions from the logs)

I must say that I may misunderstand the timers in Klipper. But what seemed strange to me was this discrepancy between the clocks: `query_adxl345 oid=1 clock=371073417` and `shutdown clock=524921845`, as if that query_adxl345 command was scheduled in the past for some reason.

KevinOConnor, 2020-09-02:
The main klippy thread isn't expecting code to run for more than a few milliseconds.  But, at the end of these tests, the code is performing processing for multiple seconds.  That's going to cause timing mishaps like the above.  A command was buffered up while TEST_RESONANCES was running and things went haywire as a result.  

Long term, we'll need to rework the code so that it doesn't block, run it via the multiprocesing module, or something else.  As a short-term work around, it's probably possible to avoid by always issuing an `M400` after a `TEST_RESONANCES` command.

-Kevin

[133 more replies]

closed 2021-04-15 by github-actions[bot]