Klipper3d/klipper #3030: Scale issue

ereish64, 2020-06-29:
The G-CODE I send to Klipper seems to get scaled up on all axis by about 15%. I tried the enhanced delta calibration, but that only accounts for the X and Y axis. It worked fine before, but after a reboot, now everything is scaled up 15%. What parameter could have changed that would result in this issue?

Thanks for your help!

ereish64, 2020-06-29:
I will post a log later today, but I don't think it is necessary.

ereish64, 2020-06-29:
[klippy.log](https://github.com/KevinOConnor/klipper/files/4847696/klippy.log)

ld3300, 2020-07-07:
hi @ereish64 
Is your step size actually exactly 0.1?
Can you please send the g-code file you are having trouble with?

David

ereish64, 2020-07-07:
I managed to fix the problem by adjusting the step size to 0.125
thanks for your help

closed 2020-07-07 by ereish64