Klipper3d/klipper #3035: [FR] Allow Negative "unretract_extra_length"

DroneMang, 2020-06-30:
Hi,

I am wondering if unretract_extra_length could be allowed to be negative.  When printing, I usually get a blob after a retract, in Marlin I used this being negative to help that.

I suppose the wipe feature could help with this as well, but it doesn't look that nice.

Also, I've discovered you can use very fast retract and unretract speeds (say 600-800) to help blobs, and keep the nozzle moving.  This is especially useful when using a Volcano nozzle.

Thanks!

[klippy2.log](https://github.com/KevinOConnor/klipper/files/4853819/klippy2.log)

ld3300, 2020-07-07:
Hi @DroneMang 
in your rapsberry pi go to the folder Klipper > Klippy > extras
In firmware_retraction.py try changing the following:
Line 13, make the minval a negative number that meets your needs
Line 41 make the same change

save the file and restart the pi.  Test it out and see if it works effectively without causing errors or other print problems.

David

robthide37, 2020-07-13:
is their any benefit at this point for firmware retraction vs just doing it in the slicer?

BlackStump, 2020-07-13:
firmware retraction you can configure on the fly

robthide37, 2020-07-13:
Oh so no need reslice is their any downside ?

BlackStump, 2020-07-14:
no downside the slicer needs to support firmware retraction (most do) and klipper firmware retraction does not support Z hop

KevinOConnor, 2020-07-25:
Changes like this would require an interested developer to implement and test.  I'm going to close this as it looks like the conversation has concluded.  If an interested developer does implement it, feel free to submit a PR.

-Kevin

closed 2020-07-25 by KevinOConnor

DroneMang, 2020-07-27:
Just wanted to add that I did test this, and it worked as intended, but I basically just left it at 0, because some start points would not stick to the rest of the print so I would have a gap.  But a very small negative value could be useful.  It may be good to just let the default minimum to be less than zero, say maybe -1, so the user has the option?