Klipper3d/klipper #3036: Move exceeds maximum extrusion (10.180mm^2 vs 0.640mm^2)

taw-pi, 2020-07-01:
Hi,

I keep getting this error message whilst attempting a print. It only seems to occur when using Cura and/or Simplify 3D Slicers.

I have read older posts on this issue around 2018 and have tried the dummy move 'G1 F1500' after purging my extruder as recommended but the error still occurs. I have my reasons for using Simplify 3D for this print so feel a little stuck.

Any suggestions gratefully received.

Thanks

[klippy.log.zip](https://github.com/KevinOConnor/klipper/files/4859877/klippy.log.zip)

KevinOConnor, 2020-07-02:
There are a few typical reasons for this to occur:
1. Simplify3D is configured to use some "advanced extruder pressure" setting that is not compatible with Klipper - see: https://www.klipper3d.org/Slicers.html
2. A custom start g-code script is defined in the slicer that implements a purge line that is not compatible with Klipper.  If this is the case, easiest solution is to change the purge line to use more reasonable values.
3. There is an error in a custom script causing g90/g91 mode to be confused.  If so, the underlying script should be updated.
4. The slicer is buggy and is issuing commands that may cause nozzle jams and damage to the printer.  (For example, this was known to occur on some older versions of sli3er.)

Unfortunately, it's not easy for us to determine which of the above is the root cause.

-Kevin

taw-pi, 2020-07-04:
I believe I have found the problem...

It appears that some slicers do not like to be left in extruder relative mode (M83). Once I removed that from my START_PRINT macro the error did not reoccur. 

The clue was that the G90/G91 became confused because M83 was overriding G90 for the extruder.

Thanks for the feedback.

closed 2020-07-04 by taw-pi