Klipper3d/klipper #3037: [FR] Add support for redundant thermistor

DroneMang, 2020-07-01:
Hi,

I am wondering if you could add support for having 2 thermistors for 1 hotend, similar to "TEMP_SENSOR_1_AS_REDUNDANT" in Marlin 2.0, and also incorporate the error checking with runaway temps or thermistors that fall out etc.  Thanks!

DroneMang

The log is irrelevant...
[klippy.zip](https://github.com/KevinOConnor/klipper/files/4860278/klippy.zip)

ld3300, 2020-07-07:
Hi @DroneMang 
 Can you describe more about the expected behavior?  Would the temperature state always be considered as the average of the 2?  Would it switch sensors if runaway is detected?  It may not be difficult to implement.  Just need to have a clear understanding of the goal and probably some additional community input on implementation.

David

DroneMang, 2020-07-07:
I would say the main goal is to prevent a fire.  I realize there are already safety features built in to help with this, but its an easy addition to most printers for a little extra safety.

Other goals could be to detect problems with either thermistor, or continue a long print even if one thermistor is faulty (so there could be a setting that allows the user to specify if they should be averaged, or the second only used to detect an error, and what to do if there is an error-pause with bed heater on, pause with hotend and bed off) as far as the error detection goes, if both thermistors are tracking in parallel with neither rising but the difference is growing over time, then perhaps one is wearing out so detect the one that is changing and ignore the other so the print can continue, if one is having a runaway or cooling off fast and the other not, then something is definately wrong.  Idk if this captures the failure modes totally though.  But there could be some basic logic to detect what kind of failure is happening.  When I tried this in marlin, I had to set the MAX_REDUNDANT_TEMP_SENSOR_DIFF = 15, not because they were that much different, but because it would throw an error when heating up.  I have a 50W heater, so it heats up fast, and I am assuming part of the block heats differently.  So there could be a "diff" setting that compares the two during heatup, and another during steady state, or just don't compare until steady state is reached, or at least don't throw an error.

Also, it would be nice to see both temperatures in Octoprint so they can be compared live, even though it would be specified a single nozzle and heater is being used, this feature could be turned on or off in the cfg.

I have a 2 in 1 hotend, so I am not sure if the filament switch from T0 to T1 would confuse Octoprint though.

Thanks for taking a look at this!
DroneMang

ld3300, 2020-07-07:
hi @DroneMang 

Nearly all of that functionality can be implemented without changing Klipper.  Try defining a [custom sensor](https://github.com/KevinOConnor/klipper/blob/cd8f250e84b8f1ae56eb8fc87d2c6fa623e14c98/config/example-extras.cfg#L1064) for the second sensor and you can use a delayed gcode to do comparisons like differences to catch inconsistent values and determine whether to send an emergency stop or pause, or some other function.

The ability to have a second sensor to allow a print to continue in case one should fail would require some changes to the heater section of the klipper code.  It would probably be easiest to set up a case where as long as one of the sensors is  at the expected value than keep going, rather than treating it as a main and backup type scenario.  Somewhere in my fuzzy head there is a feeling that there is an additional factor to consider to make sure it doesn't break the thermal runaway safety mechanism, but I can't quite catch it.

If you have some time see how much of the functionality you can implement with custom gcodes.  

David

DroneMang, 2020-07-07:
I can code a little but it would be easier if I had something to start
with. Do you have some code you could post that would have some of the same
types of variables and code blocks? Thanks.

ld3300, 2020-07-10:
Here's a custom g-code command where I take action based on a temperature.  I'm not much of a coder, but I help out where I can:
```
[delayed_gcode PRINTER_SHUTDOWN]                # Check cooldown then shutoff machine
gcode:
    RESPOND TYPE=echo MSG="COOLING"
    {% if printer.extruder.temperature < 48 %}
        M117 Shutdown reached
        RESPOND TYPE=echo MSG="SHUTDOWN REACHED"
        M106 S0
        M84                                            # Kill Motors
        RESPOND TYPE=command MSG=action:M81            # Tell octoprint to shutdown power
    {% else %}
        RESPOND TYPE=echo MSG="SHUTDOWN SEQUENCE ACTIVE"
        UPDATE_DELAYED_GCODE ID=PRINTER_SHUTDOWN DURATION=3
        M117 Cooling Hotend
    {% endif %}
```

The firmware changes would likely be in klippy/heater.py

KevinOConnor, 2020-07-25:
I'm going to close this as it looks like the original question was answered.  (It's already possible to implement this via existing Klipper support.)  If a developer is interested in making this more streamlined then feel free to submit a PR with it.

-Kevin

closed 2020-07-25 by KevinOConnor