Klipper3d/klipper #3038: Pressure advance not applied to extruder_stepper

theophile, 2020-07-02:
So I noticed that pressure advance settings get applied to the target extruder but not to a defined extruder_stepper that is synced with the active extruder. In my case, where the extruder_stepper is between the active extruder and the hotend, this essentially means no pressure advance.

I know you can't pass the extruder_stepper config name to SET_PRESSURE_ADVANCE with the EXTRUDER parameter. Is there some other way to do it?

theophile, 2020-07-04:
[klippy.log](https://github.com/KevinOConnor/klipper/files/4873763/klippy.log)

I poked around in extruder.py enough to figure out (I think) that pressure advance is something that can only be applied to an "extruder" object, not a "stepper" object, and that since an extruder_stepper is not an "extruder," it doesn't seems like it's currently possible for pressure advance to be applied to an extruder_stepper. I made an initial attempt to apply the pressure advance setting of the current extruder to the extruder_stepper, similar to how the last commanded position of the current extruder is applied to the extruder_stepper, but it became clear that it's not that simple, because the extruder_stepper object doesn't know what to do with pressure advance settings. Conceptually, I would think that the extruder_stepper should always have the same pressure advance setting as the extruder it's synced to, as I can't imagine a situation where a user would want two steppers in the same filament path to have different pressure advance settings.

dmbutyugin, 2020-07-07:
@theophile, FWIW, it seems that most of the support is actually already there, and PrinterExtruder class takes care of most of it. The only thing that's currently missing is [extruder_set_smooth_time](https://github.com/KevinOConnor/klipper/blob/master/klippy/kinematics/extruder.py#L90) on the additional extruder stepper. The value is 0 by default in C code, which disables PA.

theophile, 2020-07-07:
Thanks @dmbutyugin . I clearly don't know what I'm doing. ;-) From your description it sounds like it wouldn't be difficult for someone who does know what they're doing!

KevinOConnor, 2020-07-25:
It is indeed possible to add support for pressure advance to "extruder_stepper" steppers.  An interested developer would need to take that on though.

-Kevin

theophile, 2020-08-08:
Okay I'm looking at this again. @dmbutyugin, if I understand your comment correctly, this would _not_ require changes to any C code, right? Just modifying the appropriate python script(s) to apply extruder_set_smooth_time to the additional extruder stepper?

Part of my confusion is that the extruder stepper has its own ExtruderStepper class that seems to be different from the PrinterExtruder class. So to get pressure advance to work on the ExtruderStepper, I would have to take all the relevant bits out of the PrinterExtruder class and put them into the ExtruderStepper class, right? Or am I approaching this the wrong way?

theophile, 2020-09-02:
I would respectfully submit that this issue should stay open. In my view, this constitutes a bug rather than a feature request because at present, I think the expected behavior of an extruder_stepper is that all of its movements are synchronized with the movements of the extruder it is linked to, but in fact the movements are not synchronized if the extruder is using pressure advance.

I'd really like to get this working but I need some help as I don't think I can do it alone. @dmbutyugin mentioned above that the PrinterExtruder class takes care of most of the pressure advance support, but "the only thing that's currently missing is extruder_set_smooth_time on the additional extruder stepper." Does this mean that the existing code already applies the pressure advance parameters to the extruder_stepper except for "extruder_set_smooth_time," and that the extruder_stepper will sync with the pressure advance movements of the main extruder if "extruder_set_smooth_time" is set with respect to the extruder_stepper? I'm trying to take stock of what exactly would have to be done to get this to work. Thanks all!

dmbutyugin, 2020-09-03:
@theophile Yes, no C code changes are necessary, I think. 

> Does this mean that the existing code already applies the pressure advance parameters to the extruder_stepper

It's a bit complicated, but the pressure advance itself is passed to the code that calculates the stepper steps. But if the smooth time is not set, the stepper code will ignore this pressure advance parameter.

Now passing the smooth time parameter will take some coding. The problem at hand is that [this](https://github.com/KevinOConnor/klipper/blob/master/klippy/kinematics/extruder.py#L90) line of code is executed only for the main stepper of the extruder. It needs to be called for the stepper kinematics object of the other synced extruders as well. The new synced stepper is passed [here](https://github.com/KevinOConnor/klipper/blob/master/klippy/kinematics/extruder.py#L101). So, what I _think_ needs to be done:

- `PrinterExtruder` must track (e.g. store a list) of all steppers synced to it, updated by `sync_stepper` method. Note that if a new stepper is added to a `PrinterExtruder`, it must be de-registered from the previous `PrinterExtruder`, if it was previously synced to another extruder (so a new method needs to be added to it, like `unsync_stepper` or something). So,
- `ExtruderStepper` must track which extruder it is synced to (to be able to de-register).
- `MCU_stepper` class needs to provide a getter for its `_stepper_kinematics` field (this is necessary so that `PrinterExtruder` can call `extruder_set_smooth_time` on that object.
- `_set_pressure_advance` method must go over all synced steppers and call `self.extruder_set_smooth_time` on each of their `_stepper_kinematics`.
- And finally, `sync_stepper` method should also call `self.extruder_set_smooth_time` for the newly synced stepper (perhaps it would be better to just add a helper method `_set_stepper_pressure_advance(self, stepper, smooth_time):` to `PrinterExtruder` class.

theophile, 2020-09-03:
Thanks @dmbutyugin, this is really helpful. @KevinOConnor actually added the sync_stepper method to facilitate the ability to change the synced extruder_stepper on the fly: https://github.com/KevinOConnor/klipper/commit/739e6e73963effdd64677872af87a303cd68c3de
Could it be as simple as adding extruder_set_smooth_time to the sync_stepper method, similar to how it's doing set_trapq?

dmbutyugin, 2020-09-03:
@theophile
> Could it be as simple as adding extruder_set_smooth_time to the sync_stepper method, similar to how it's doing set_trapq?

You can try doing that to test if that even works or not, before doing any heavy work. However, if a user calls `SET_PRESSURE_ADVANCE` with the smooth time parameter, it will not be updated in the synced extruders in this case. This is why a full solution requires more changes, so that `SET_PRESSURE_ADVANCE` can work correctly with synced extruders too.

theophile, 2020-09-03:
Oh I see. That makes sense. If the `extruder_set_smooth_time` parameter is applied when `sync_stepper` is called, then if the `extruder_set_smooth_time` parameter is changed after that there's no mechanism for applying that change to the synced extruder stepper without calling `sync_stepper` again. I will give it a try and see if I can make it work that way. If it does, then maybe the "easy" solution would be to just re-sync the extruder_stepper again immediately after changing `extruder_set_smooth_time`. I would think that would be sufficient for most use cases, as I would assume that during normal printing, a user likely wouldn't change `extruder_set_smooth_time` except at a filament change, in which case you're already injecting toolchange code anyway. The user would just have to make sure the toolchange script executes the change in the `extruder_set_smooth_time` parameter prior to executing `SYNC_STEPPER_TO_EXTRUDER STEPPER`.

I'll play with it and report back!

theophile, 2020-09-05:
Okay, so starting with @dmbutyugin 's list above, I figured that for proof-of-concept purposes, I could skip over trying to get `PrinterExtruder` to track synced steppers. So I started by trying to get `MCU_stepper` to provide a getter for its `_stepper_kinematics` field, by adding the following lines to klippy/stepper.py, around line 131:
```
def get_stepper_kinematics(self):
       return self._stepper_kinematics
```
Then in klippy/kinematics/extruder.py, I added the following lines at the end of the `sync_stepper` method, starting around line 105:

```
sk = self.stepper.get_stepper_kinematics
self.extruder_set_smooth_time(sk, self.pressure_advance_smooth_time)
```

My thinking is that when `sync_stepper` is called, it would get the stepper kinematics object for the synced extruder_stepper, and then call `extruder_set_smooth_time` on that object using whatever the current `pressure_advance_smooth_time` is.

When I restart klippy, I get this error in the log:

```
Unhandled exception during connect
Traceback (most recent call last):
  File "/home/pi/klipper/klippy/klippy.py", line 151, in _connect
    cb()
  File "/home/pi/klipper/klippy/extras/extruder_stepper.py", line 26, in handle_connect
    extruder.sync_stepper(self.stepper)
  File "/home/pi/klipper/klippy/kinematics/extruder.py", line 106, in sync_stepper
    self.extruder_set_smooth_time(sk, self.pressure_advance_smooth_time)
TypeError: initializer for ctype 'struct stepper_kinematics *' must be a cdata pointer, not instancemethod
```

I _think_ this is trying to tell me that my `get_stepper_kinematics` method isn't returning what I need it to return, so the wrong kind of data is being passed to `self.extruder_set_smooth_time`, but I'm not entirely sure. Any ideas for clearing this hurdle? TIA.

theophile, 2020-09-29:
My apologies, GitIssueBot, but please don't close this issue. I think this is a legit bug that should be fixed, although I recognize that it probably doesn't affect the majority of users. I have made some attempts to fix it myself but I think I am unlikely to succeed without some additional assistance/guidance from others. @dmbutyugin or @KevinOConnor, I'd be grateful for any further attention you might be willing to devote to this issue. I'd really love to be able to help get this included in the upcoming v0.9.0 release if at all possible. Thanks all!

dmbutyugin, 2020-10-01:
Hi @theophile,

> Then in klippy/kinematics/extruder.py, I added the following lines at the end of the `sync_stepper` method, starting around line 105:
> 
> ```
> sk = self.stepper.get_stepper_kinematics
> self.extruder_set_smooth_time(sk, self.pressure_advance_smooth_time)
> ```

That's indeed how you can test it. Just a couple of things: you should use `stepper` argument here, and not the member `self.stepper`, and you have forgotten to call a method (i.e. put parenthesis after the method name, so it should be `sk = stepper.get_stepper_kinematics()`), and that's why you got an error message

> ```
> Unhandled exception during connect
> Traceback (most recent call last):
>   File "/home/pi/klipper/klippy/klippy.py", line 151, in _connect
>     cb()
>   File [...]

theophile, 2020-10-05:
Thank you @dmbutyugin ! I fixed those errors and set up a fork for testing purposes. The commit is here: [https://github.com/theophile/klipper/commit/78b8b258c9a7e690af692ae4f8190b1416ec0e52](url). Unfortunately, I think this is causing a problem with the kinematic position of the toolhead.

My printer has a "purge bucket" on the left-hand side of the X gantry. I have a "PRINT_START" macro that heats up the necessary heaters and moves the tool carriage to the purge bucket, then the "upstream" extruder feeds filament until it reaches the "downstream" extruder (which is a BMG clone in direct-drive configuration). Then the downstream extruder is synced to the upstream extruder, and they then jointly feed more filament to prime the hotend, and then there is an additional extrusion to purge the nozzle. Then there is a G1 move to lower the Z by 10mm (it had been raised as part of the move to the purge bucket). Then there's an additional purge of 10mm of filament, followed by a back-and-forth X move of 15mm to wipe the nozzle on a strip of silicone to break off the purged filament. Then the printing moves begin.

With the above-linked commit applied, things go as expected until the `G1 Z-10 F420` move after the hotend has been primed. Instead of moving, the MCU goes into shutdown with this error:

```
stepcompress o=0 i=0 c=268 a=0: Invalid sequence
Internal error on command:"G1"
Traceback (most recent call last):
  File "/home/pi/klipper/klippy/gcode.py", line 177, in _process_commands
    handler(gcmd)
  File "/home/pi/klipper/klippy/extras/gcode_move.py", line 135, in cmd_G1
    self.move_with_transform(self.last_position, self.speed)
  File "/home/pi/klipper/klippy/extras/bed_mesh.py", line 181, in move
    self.toolhead.move(split_move, speed)
  File "/home/pi/klipper/klippy/toolhead.py", line 419, in move
    self.move_queue.add_move(move)
  File "/home/pi/klipper/klippy/toolhead.py", line 186, in add_move
    self.flush(lazy=True)
  File "/home/pi/klipper/klippy/toolhead.py", line 175, in flush
    self.toolhead._process_moves(queue[:flush_count])
  File "/home/pi/klipper/klippy/toolhead.py", line 335, in _process_moves
    self._update_move_time(next_move_time)
  File "/home/pi/klipper/klippy/toolhead.py", line 288, in _update_move_time
    sg(sg_flush_time)
  File "/home/pi/klipper/klippy/stepper.py", line 173, in generate_steps
    raise error("Internal error in stepcompress")
error: Internal error in stepcompress
Transition to shutdown state: Internal error on command:"G1"
```
The full klippy.log is here: [klippy.log](https://github.com/KevinOConnor/klipper/files/5324694/klippy.log)

In my earlier efforts to implement changing the synced extruder_stepper on the fly, I ran into similar `stepcompress` "Invalid sequence" errors, which @KevinOConnor pointed out was an indication that the stepper had gotten out of sync with its position: https://github.com/KevinOConnor/klipper/issues/2733#issuecomment-623033785

In that situation, the key was to have the sync sequence read the extruder's position then set the kinematic position of the newly synced extruder_stepper to match.

What's interesting here though is that the `stepcompress` error is being thrown by a non-extrusion move--the `G1 Z-10 F420`. So I suspect that trying to apply `extruder_set_smooth_time` to the extruder_stepper in this manner is disrupting something in the non-extrusion kinematics. Any ideas?

dmbutyugin, 2020-10-05:
@theophile A couple of clarifying questions:

- I see in the logs that the pressure advance is set to 0. Is this intentional? You have been trying to get pressure advance working for the synced extruder. Note there is [special handling](https://github.com/KevinOConnor/klipper/blob/master/klippy/kinematics/extruder.py#L85-L86) of 0 PA in `_set_pressure_advance`, which you don't have here.
- Do you get this error if you don't apply your patch, but run the rest in exactly the same way?

theophile, 2020-10-05:
The PA=0 line was a test to see if "manually" setting the PA value after syncing the extruder_stepper made a difference. It doesn't though. I get the same error without that line in the macro. That said, I actually have not tried it with a positive PA value yet. 

Without the patch everything works fine.

dmbutyugin, 2020-10-05:
Yes, please try PA>0 value. PA=0 forces 0 smooth time on the main extruder stepper (and you would need to replicate for additional steppers too, but for now you can just try PA>0), so it isn't testing much.

theophile, 2020-10-05:
@dmbutyugin , I think we're on to something! I changed the macro so that it issues a `SET_PRESSURE_ADVANCE ADVANCE=0.200` shortly before the purge extrusions and this time the print started successfully. Better yet, both extruders appeared to be correctly doing the PA moves in sync as expected. While it was printing, I issued a `SET_PRESSURE_ADVANCE ADVANCE=0.400` and it appeared to take effect seamlessly as expected, with the PA moves becoming more exaggerated on both steppers. Then I issued a `SET_PRESSURE_ADVANCE ADVANCE=0.200` and it immediately stopped with the same `Internal error on command:"G1"` error as before.

Now why would PA=0 cause this error? I see that when PA=0, the smooth_time is set to 0. But if that's the case, then the patched version of klippy/kinematics/extruder.py would just be setting the extruder_stepper kinematics to what they already are, right?

dmbutyugin, 2020-10-05:
> Now why would PA=0 cause this error?

Because, as I linked earlier, there is a special code in `_pressure_advance` that sets smooth time = 0 in this case, but only for the main stepper (but the value in the class is still the 'normal' value). So unless you replicate that logic in `sync_stepper`, you are settings incorrect (inconsistent) smooth time > 0 for the additional stepper when PA=0, steppers go out of sync and this results in an error (more like the toolhead lets them go out of sync, but that's details).

Generally, I think the extruder class needs to track all synced steppers (including its own), have sync_stepper / unsync_stepper methods, and go over all registered steppers in _pressure_advance and apply the same settings to them (and call _pressure_advance after syncing the new stepper).

theophile, 2020-10-05:
Thanks @dmbutyugin . I didn't realize that the method of getting the extruder's kinematics and applying it to the extruder_stepper would apply a non-zero `smooth_time` to the extruder_stepper when PA=0. I'll experiment with using a conditional of some sort to prevent the `self.extruder_set_smooth_time(sk, self.pressure_advance_smooth_time)` line in `sync_stepper` from running if PA=0. It seems like that would avoid the `stepcompress` error in that situation.

I think you're probably right about the ideal means of implementing this functionality, but I don't think I have the knowledge or the skill to accomplish it. Unless you don't mind continuing to coach me through it of course! ;-) I really do appreciate all your help thus far.

dmbutyugin, 2020-10-05:
Well, just copy-past that code about `new_smooth_time` here as well, and pass it to `extruder_set_smooth_time` for now.

theophile, 2020-10-06:
All right, so this patch appears to get me over the next hurdle: https://github.com/theophile/klipper/commit/4ea27c498937dff0ee2ad6e1606d9f9b2d683662

Essentially, if PA=0, this additional code does nothing at all. But if PA is greater than 0, then it processes the `self.extruder_set_smooth_time(sk, self.pressure_advance_smooth_time)` line when `sync_stepper` is called. I plan to run a PA tuning tower this evening per the calibration docs to confirm that it's working as expected.

I already know that another tweak is needed though, because if PA is set _back_ to 0 after being a positive value, then that will cause the `smooth_time` values to become inconsistent again resulting in the `stepcompress` error. Also, although I haven't tried it yet, I'm confident that the same thing would happen if `SMOOTH_TIME` is changed via gcode, since that change wouldn't be applied to the extruder_stepper. The workaround might be to create a macro to change `SMOOTH_TIME` and ensure that a `SYNC_STEPPER_TO_EXTRUDER` command is run immediately afterwards (before any G1 moves) to "resync" the extruder_stepper and apply the new `smooth_time` value.

Assuming the tuning tower completes successfully, I'll see if I can figure out a way to have `sync_stepper` called whenever `_set_pressure_advance` runs. I think that would make sure that the synced extruder_stepper stays synced whenever PA is changed.

theophile, 2020-10-07:
Okay, more progress with these new changes: https://github.com/theophile/klipper/commit/ef8fc929cf11d2f30fa474a8902747d2f4d49761

I'm confident there's a more elegant way to do this but it works. Every time the `sync_stepper` method is called, it defines the global variable `synced_stepper` as whatever extruder_stepper has just been synced. Then, whenever `_set_pressure_advance` is called, it calls `sync_stepper` again on the `synced_stepper` to apply the new `smooth_time` to the synced extruder_stepper.

Because the new `synced_stepper` variable is only defined when `sync_stepper` is called, I used a `try` block to keep Python from complaining about the undefined variable, and to skip the extra call to `sync_stepper` when that variable is undefined.

Finally, I added another couple of lines to `sync_stepper` to manually set `smooth_time` back to 0 when PA=0.

I just successfully printed a PA tuning tower without user intervention, and also tried a test print changing from PA=0 to various positive values and back to 0 again, all while printing, and all with the intended behavior.

For my use case (only one extruder_stepper) this is now fully working. But I realize that this will not work as intended in a setup that has more than one extruder_stepper. In that situation, when PA is changed, the change will only be applied to the most recently synced extruder_stepper, rather than to all synced extruder_steppers as the user likely would expect. This is why @dmbutyugin said that the best way to implement this would be to track (register and deregister) all synced steppers and apply PA changes to all synced steppers.

I don't know if I can accomplish that, but I think this is already an improvement as-is. It appears to work for the single extruder_stepper use case, and while it doesn't solve the problem for the multi-extruder_stepper use case, at least it doesn't make it worse (as far as I can tell).

dmbutyugin, 2020-10-10:
> I don't know if I can accomplish that, but I think this is already an improvement as-is.

@theophile, in my opinion, it might be a bit not as simple as that:

- the current code has a graceful degradation mode (it does not support pressure advance on additional steppers, but that's about it; this limitation is easy to explain/document);
- the suggested fix, if I am not mistaken, has a potential to crash Klipper if more than one extra stepper is configured on some sequences of commands (e.g. if smooth_time is changed, or if PA is changed from 0 to non-zero or vice versa) with that 'stepcompress ...: Invalid sequence' error.

Therefore, it is, at least, not entirely obvious that the proposed PR #3411 is better than the current code. But please keep in mind that I am not a maintainer of Klipper and have really no say in deciding which fixes to integrate or not, so this is simply my opinion, and Kevin may think differently.

Separately, I think you don't give yourself enough credit, I think you can make the remaining changes. You'd pretty much need to create `self.steppers = []` list in `PrinterExtruder` and make sure to update it appropriately, and iterate over it in some of the methods (e.g. `_set_pressure_advance`) instead of just using `self.stepper`.

theophile, 2020-10-13:
Thanks @dmbutyugin I appreciate the encouragement. I think you're right about the potential for crashing on multi-extruder_stepper systems. The docs would have to explicitly warn such users to leave PA turned off (since it doesn't work anyway) to avoid the potential for crashes. Obviously, the better approach is to make it work properly. 

I had figured out that a list is probably the way to go (Google informed me that a "list" in Python is like an array in other contexts). But then I started to worry about how to keep up with more complex sequences/scenarios, such as when there are multiple extruder_steppers that could be synced with any combination of multiple extruders. It would seem that each configured extruder should have its own list of synced extruder_steppers that must be maintained independently. This would be especially relevant in printers with independent carriages, multiple hotends, or mixing hotends.

I might pick this back up at some point down the road.

dmbutyugin, 2020-10-13:
> I had figured out that a list is probably the way to go (Google informed me that a "list" in Python is like an array in other contexts). ... It would seem that each configured extruder should have its own list of synced extruder_steppers that must be maintained independently.

FWIW, yes, that's exactly the way to go. In principle, if you create a list as a member of a PrinterExtruder object - it will be independent for each extruder automatically.

closed 2020-11-12 by github-actions[bot]