Klipper3d/klipper #3039: Firmware for additional MCUs

Hairyloon, 2020-07-02:


theophile, 2020-07-02:
You would use the Klipper firmware on them. Though there are plenty of other possible use cases, I've got three MCUs so I can use 7-8 independent extruder stepper motors. All the extra mosfets are useful too for configuring all sorts of different fans for different purposes, without having to use breakout boards.

Hairyloon, 2020-07-02:
Ah, does that mean that only the processors listed in the menuconfig can be used?
Although having had a look, there are more of them than I remember, although none that I have in my box of random things...

One thought was to use an ESP32-Cam to watch the printer, and wondering if it could do something else as well while it's there...

yschroeder, 2020-07-05:
You can use any of the suported MCUs that are available in the menuconfig. It does not need to be a printer board. Any other board (e.g. the cheap BluePill boards) works.

Having multiple MCUs allows you to split up the "brain" of your printer. You could have one MCU moving along with the hotend on the carriage that only manages the extruder motor, and hotend temperature for example.

You need to add additional `mcu` sections to your config as shown here:
https://github.com/KevinOConnor/klipper/blob/master/config/example-extras.cfg#L664

Hairyloon, 2020-07-06:
> 
> You can use any of the suported MCUs that are available in the menuconfig. It does not need to be a printer board. Any other board (e.g. the cheap BluePill boards) works.
> 
Is there a handy list?

> Having multiple MCUs allows you to split up the "brain" of your printer. You could have one MCU moving along with the hotend on the carriage that only manages the extruder motor, and hotend temperature for example.
> 

Yes, I had twigged to that. Perhaps use a wireless MCU at the hotend so that only power & filament need to be fed in?

jakep82, 2020-07-06:
> Is there a handy list?

https://github.com/KevinOConnor/klipper/blob/master/docs/Features.md#step-benchmarks

Hairyloon, 2020-07-06:
Tidy, thank you.

yschroeder, 2020-07-07:
I also thought about wireless MCU a few years ago. You need very reliable communication, which might be hard. Using bidirectional RS485 is easier. This would result in 2 thick wires for power and 4 thin ones for communication. I am running klipper via 50 meters of cat5 cable, even the power (the 4 wires that are not used for RS485). But it only drives one Nema17, not a heater.

Hairyloon, 2020-07-07:
> 
> 
> I also thought about wireless MCU a few years ago. You need very reliable communication, which might be hard. 

Have you come across ESP3d?  [https://github.com/luc-github/ESP3D](url) Not what we're talking about here, but worth a look.

yschroeder, 2020-07-08:
I'm not sure how klipper would react to delays in communication with the MCU. The serial line is handled in soft realtime. Same goes for other firmware when you stream the gcode, e.g. from octoprint.

There is even an RFC about encapsulation of serial into IP. I think the ESP3d firmware just puts the serial data into a TCP stream and hopes that TCP sorts missing segments out before the gcode buffer on the printer underruns.

I would not trust serial over WiFi for a multiple hours print.

manu7irl, 2020-07-09:
Nice conversation sorry to interrupt...
Can it be done maybe via encapsulated serial over tcp via ethernet port to
a switch then relayed to a klipper host.
Maybe can be done via pyserial library as in the example on their wiki.
The ethernet tcp would be good enough transmit in a short time frame.
Suggestions? How to do that?

yschroeder, 2020-07-09:
Ethernet in contrast to WiFi should be far more reliable. Of course if your network gets overloaded by other data, soft real-time klipper data will be delayed. I don't know what the requirements for jitter are.

I think klipper does an emergency shutdown if the deadline for the data is not met, even if it's just one packet that is late. Maybe @KevinOConnor can provide insights if klipper via Ethernet would be feasible.

Hairyloon, 2020-07-10:
> 
> Nice conversation sorry to interrupt... 

No problem. I think the opening question had been adequately answered, you're welcome to take the ponderings that it provoked in any direction that looks useful.

manu7irl, 2020-07-10:
Great thanks I took a look at this repo, that could be useful to integrate
to klipper, but I am not a python dev...
https://github.com/mosquito/serial2tcp
it uses pyserial on python 2 so as in klipper.
So it could be added to klipper to make klippy transceive tcp stream to
local serial,
if someone could direct me how to do that, I will try myself!

‫בתאריך יום ו׳, 10 ביולי 2020 ב-4:49 מאת ‪Hairyloon‬‏ <‪
notifications@github.com‬‏>:‬

closed 2020-07-25 by KevinOConnor