Klipper3d/klipper #3041: Current feedrate and flow modifier (M220 and M221)

MassimoVisona, 2020-07-02:
Hello,
I was trying to figure out how to obtain the current feedrate and flow rate previously setted via the M220 and M221 gcode commands. I think it would be similar to this topic.
https://github.com/KevinOConnor/klipper/issues/1705
Basically I'm searching a method to ask the printer the current feedrate and flow modifier.
Any idea? Thank you.

joris57, 2020-07-02:
You could write a macro:

`[gcode_macro xxxxx]
gcode: M118 Speed fact {printer.gcode.speed_factor}

MassimoVisona, 2020-07-03:
Hello,
thank you for the help. 
I'm getting unknown command on M118.

joris57, 2020-07-03:
Have a look at https://github.com/KevinOConnor/klipper/blob/master/docs/G-Codes.md . You will see that M118 writes to the host (terminal) and not to the display. It also requires the "respond" config section to be enabled. See the example in https://github.com/KevinOConnor/klipper/blob/master/config/example-extras.cfg .

MassimoVisona, 2020-07-05:
Thank you very much. It's working!

closed 2020-07-05 by MassimoVisona