Klipper3d/klipper #3043: Could not install Linux MCU

gsl12, 2020-07-03:
I tried to install linux mcu, but i failed. I followed the steps from the documentation. I also started from scratch.

First i installed klipper successfully. Then tried to "make flash" for linux mcu, but received an error.

Error: Failed to restart klipper_mcu.service: Unit klipper_mcu.service not found

What i'm doing wrong?

I use a RaspberryPi 3 with Raspberry Pi OS (32-bit) Lite Version:May 2020.

```
###### Launching Klipper host software...
[ ok ] Restarting klipper (via systemctl): klipper.service.
pi@tiny-m:~ $ cd ~/klipper/
pi@tiny-m:~/klipper $ sudo cp "./scripts/klipper-mcu-start.sh" /etc/init.d/klipp                                                                                                                                                             er_mcu
pi@tiny-m:~/klipper $ cd ~/klipper/
pi@tiny-m:~/klipper $ make menuconfig
scripts/kconfig/conf --olddefconfig /home/pi/klipper/src/Kconfig
#
# configuration written to /home/pi/klipper/.config
#
scripts/kconfig/mconf /home/pi/klipper/src/Kconfig
configuration written to /home/pi/klipper/.config

*** End of the configuration.
*** Execute 'make' to start the build or try 'make help'.

pi@tiny-m:~/klipper $ sudo service klipper stop
pi@tiny-m:~/klipper $ make flash
  Build Kconfig config file
#
# configuration written to /home/pi/klipper/.config
#
  Creating symbolic link out/board
  Compiling out/src/sched.o
  Compiling out/src/command.o
  Compiling out/src/basecmd.o
  Compiling out/src/debugcmds.o
  Compiling out/src/initial_pins.o
  Compiling out/src/gpiocmds.o
  Compiling out/src/stepper.o
  Compiling out/src/endstop.o
  Compiling out/src/adccmds.o
  Compiling out/src/spicmds.o
  Compiling out/src/thermocouple.o
  Compiling out/src/i2ccmds.o
  Compiling out/src/pwmcmds.o
  Compiling out/src/lcd_st7920.o
  Compiling out/src/lcd_hd44780.o
  Compiling out/src/buttons.o
  Compiling out/src/tmcuart.o
  Compiling out/src/spi_software.o
  Compiling out/src/neopixel.o
  Compiling out/src/linux/main.o
  Compiling out/src/linux/timer.o
  Compiling out/src/linux/console.o
  Compiling out/src/linux/watchdog.o
  Compiling out/src/linux/pca9685.o
  Compiling out/src/linux/spidev.o
  Compiling out/src/linux/analog.o
  Compiling out/src/linux/hard_pwm.o
  Compiling out/src/linux/i2c.o
  Compiling out/src/linux/gpio.o
  Compiling out/src/generic/crc16_ccitt.o
  Compiling out/src/generic/alloc.o
  Building out/compile_time_request.o
Version: v0.8.0-619-g7bc64700-20200703_091607-tiny-m
  Linking out/klipper.elf
  Flashing
Installing mirco-controller code to /usr/local/bin/
Attempting host MCU restart...
Failed to restart klipper_mcu.service: Unit klipper_mcu.service not found.
make: *** [src/linux/Makefile:13: flash] Fehler 5
´´´

gsl12, 2020-07-03:
[klippy.log](https://github.com/KevinOConnor/klipper/files/4868576/klippy.log)

gsl12, 2020-07-05:
OK solved the issue, while set permission to the start script "sudo chmod +x /etc/init.d/klipper_mcu". Also need to add my user to tty group with "sudo usermod -a -G tty pi".

closed 2020-07-05 by gsl12

Fabrizio3, 2020-09-19:
> 
> 
> OK solved the issue, while set permission to the start script "sudo chmod +x /etc/init.d/klipper_mcu". Also need to add my user to tty group with "sudo usermod -a -G tty pi".

can you give me a hand with that? im having the same exact issue it seems, doesn't wanna install, when running the command it only does apt-get update and stops there without continuing
best regards