Klipper3d/klipper #3045: Arduino aliases not supported on mcu 'atmega1284p'

wjg68, 2020-07-03:
Getting the following error message trying to connect. please help.
Arduino aliases not supported on mcu 'atmega1284p'

[klippy.log](https://github.com/KevinOConnor/klipper/files/4869911/klippy.log)

theopensourcerer, 2020-07-03:
The message tells you there is no support for Arduino pin names for that mcu. So you have to use the "native" hardware pin names:

https://github.com/KevinOConnor/klipper/blob/master/config/example.cfg#L11

The Creality CR-10 boards seem to use this MCU so maybe one of the CR-10 the example config files might help, e.g.:

https://github.com/KevinOConnor/klipper/blob/master/config/printer-creality-cr10-2017.cfg

wjg68, 2020-07-03:
thanks for the reply, I got the ender 3 w bltouch .cfg  file working. been working through some new errors but its connected to klipper. issues with the pid tuning. think i might be close. thanks againf for help.

KevinOConnor, 2020-07-10:
I'm going to close this as it looks like a solution was found for the original question.

-Kevin

closed 2020-07-10 by KevinOConnor