Klipper3d/klipper #3048: Printer not ready after unplugging

samnab, 2020-07-04:
[klippy.log](https://github.com/KevinOConnor/klipper/files/4871998/klippy.log)

Hi,

Novice here. So I had to move my printer and unplugged it from the Pi. After when the relocation I tried to print I continuously got "Klipper state: Not ready". I uninstalled klipper and reinstalled and I was not able to get the name of the USB port using "ls /dev/serial/by-id/*" not sure what to do. Log and cfg should be attached.
[printer.txt](https://github.com/KevinOConnor/klipper/files/4872004/printer.txt)

Any help would be appreciated.

ld3300, 2020-07-07:
Hi @samnab ,

is it clear that the MCU is powered? are you using the same USB cable?

David

KevinOConnor, 2020-07-10:
These types of issues are generally the result of a failed attempt to flash the micro-controller code.  Double check that procedure.  If you can't get any indication that there is a USB device attached to the machine then it may be a bad cable or problem on the printer board.

-Kevin

closed 2020-07-25 by KevinOConnor