Klipper3d/klipper #3049: Mesh probe while heating.

Hairyloon, 2020-07-04:
While waiting for my printer to run a bed mesh calibration, It occurred to me to wonder why it couldn't be running a probe while the bed is heating up.
There are a number of reasons, but none that I see dismiss the proposal as worthless.

The first that springs to mind is that the bed expands as it heats, but that expansion should be consistent: we can work with that.
It might be better run as an advisory system rather than calibration: perhaps just run a few points around the board to check they are where they are expected to be.

[klippy.log](https://github.com/KevinOConnor/klipper/files/4872094/klippy.log)

Hywelmartin, 2020-07-04:
M104
M140

doesn't they work?

Hairyloon, 2020-07-04:
Ah yes, so they do. I hadn't thought of that, but that does leave the question of heating.
I just tried it and the first print failed, which might be coincidence, or it may be that the heating has thrown it off.
So the question would be of whether it is worth the trouble of experimenting and coding to calculate the changes due to temperature.

Hywelmartin, 2020-07-04:
probably not.. the mesh is saved so you don't need to do it every time..

Hairyloon, 2020-07-04:
True, but I should probably do it more often than I do do, and it is the faff of waiting that is the main sticking point.
I may have a play later, to see how hard the numbers are going to be.

I think the coding is a bit beyond me, but possibly not a very big bit.

Arksine, 2020-07-04:
As you state, the bed will expand while heating, resulting a poor representation of the surface when the mesh is generated.  It isn't a good idea for the software to make assumptions about bed expansion, every printer will behave differently in this regard.  You not only would need to account for temperature, but also material, thickness, and where/how the bed is anchored to the frame.  Its also possible for the frame itself to expand.

Hairyloon, 2020-07-04:
We don't need to make assumptions, we can measure it. It may be different for every printer, but each printer should be consistent within itself and I'm pretty sure it would be linear.

Hywelmartin, 2020-07-04:
Its a little bit more complex than "at temperature X the thermal expansion is Y".  there is a time for equalization that has to be taken into the calculation, as the bending of the bed is the main expansion problem, and when it "pops".
The hotend expansion is easy to calculate...

DroneMang, 2020-07-04:
On my printer, FT-6, I have drastic changes to my bed if it is cold or heated up, about 1mm.  I have a 3rd leadscrew, so the part of the bed opposite the 3rd leadscrew drops as the bed heats.  My cfg is attached, if you want to compare the meshes.  The last mesh is test_5_17x8 and this is the best mesh so far.

Also, with BLTouch, I think it would be beneficial to have the bed heater turn off during the probe to help EMI.  But if I have this in my cfg:

[homing_heaters]
steppers: stepper_z
heaters: extruder, heater_bed

the temperature doesn't update on the temp screen in octoprint, and the heater is effectively turned off.  I have verified this by running M112 to check on the temps after the mesh probing has begun.

My current solution is to run 30 probes per point with the above code turned off so the bed heater is on the whole time, and I am using touch mode so it goes farily fast.

[klippy3.zip](https://github.com/KevinOConnor/klipper/files/4873047/klippy3.zip)

Hairyloon, 2020-07-06:
Is it possible to do a bed mesh run just for information, without it changing the settings?

ld3300, 2020-07-07:
Well it won't save it to the config unless you want it to.  If you want the bed mesh to clear or restore to a saved state use the instructions on the [g codes page](https://github.com/KevinOConnor/klipper/blob/4bdc11a8b3843b73079a2c8dfc8d8c0928c29662/docs/G-Codes.md#mesh-bed-leveling)

Hairyloon, 2020-07-07:
> 
> 
> Well it won't save it to the config unless you want it to...

No, but I presumed that it ran with the settings that it just measured. It's easy enough to reload the default settings though, I'd forgotten that was an option.

I may have a play with some measurements in due course.  I'll close this now though as it is quite likely one of those things that I don't get around to. I presume I can open it again later, or is that a breach of github etiquette.

closed 2020-07-07 by Hairyloon

ld3300, 2020-07-07:
It will run with the last mesh probing unless you do something to clear or load

I don't think it is a problem to reopen, you can also start a new ticket if the topic feels different enough.  

And you don't have to load a default mesh, you could do a probe, save it right then with a new name (which does require a klipper restart), then do a probe, print out the results then load the just saved mesh by name (which doesn't require a restart).  You can have as many saved meshes as you want.

Hairyloon, 2020-07-07:
> 
> 
> It will run with the last mesh probing unless you do something to clear or load
> 

That was what I understood it to do.

> And you don't have to load a default mesh, you could do a probe, save it right then with a new name (which does require a klipper restart), then do a probe, print out the results then load the just saved mesh by name (which doesn't require a restart). You can have as many saved meshes as you want.

It was my understanding that unless specified otherwise, "default" is the name that it saves the mesh under.  I'm a little puzzled as to what purpose there is in having multiple saved meshes: I can't see why you would want to do it... except perhaps for doing the kind of experiment I have in mind here.

ld3300, 2020-07-08:
You might want different meshes for different materials that require different bed temperatures.  A mesh for PLA and ABS might differ enough to be worth the change.  Some people also use different print surfaces on their machines for different materials.  If the placement is reliable then they may just load a different mesh on startup.

Hairyloon, 2020-07-09:
Ah yes, that makes sense. Thank you.