Klipper3d/klipper #3050: Unable to connect to SKR 1.4 Turbo

kondas, 2020-07-04:
Hello everybody. I'm trying to get Klipper working on my SKR 1.4 Turbo mainboard. My goal would be to use the rPi UART0 @ 250000 Baud, but I've tried via USB-serial and rPi UART0 without success. It looks like that after a successful flash, klippy logs are flooded with timeout messages regarding the serial connection

```Timeout on serial connect
Wait for identify_response
Traceback (most recent call last):
  File "/home/pi/klipper/klippy/serialhdl.py", line 68, in _get_identify_data
    params = self.send_with_response(msg, 'identify_response')
  File "/home/pi/klipper/klippy/serialhdl.py", line 186, in send_with_response
    return src.get_response(cmd, self.default_cmd_queue)
  File "/home/pi/klipper/klippy/serialhdl.py", line 242, in get_response
    self.serial.raw_send_wait_ack(cmd, minclock, minclock, cmd_queue)
  File "/home/pi/klipper/klippy/serialhdl.py", line 178, in raw_send_wait_ack
    raise error("Serial connection closed")
error: Serial connection closed
```

I've tried changing the usb device ID in klipper config, but doesn't help. Also changed baud from 250000 to 115200, without success. 
I've tried searching for issues with this board, but haven't found similar. Others reportedly using it both via USB - serial, and also rPi UART0. But i also see that their USB device descriptor is different.

Can somebody please help me what am i doing wrong? I'd love to try Klipper on my big delta.

USB Device info
```
Bus 001 Device 008: ID 1d50:6029 OpenMoko, Inc.
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass          239 Miscellaneous Device
  bDeviceSubClass         2
  bDeviceProtocol         1 Interface Association
  bMaxPacketSize0        64
  idVendor           0x1d50 OpenMoko, Inc.
  idProduct          0x6029
  bcdDevice            1.00
  iManufacturer           1 marlinfw.org
  iProduct                2 Marlin USB Device
  iSerial                 3 0D006023AF4A1C875D7C39AEF50020C7
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0062
    bNumInterfaces          3
    bConfigurationValue     1
    iConfiguration          0
    bmAttributes         0x80
      (Bus Powered)
    MaxPower              500mA
    Interface Association:
      bLength                 8
      bDescriptorType        11
      bFirstInterface         0
      bInterfaceCount         2
      bFunctionClass          2 Communications
      bFunctionSubClass       2 Abstract (modem)
      bFunctionProtocol       1 AT-commands (v.25ter)
      iFunction               0
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         2 Communications
      bInterfaceSubClass      2 Abstract (modem)
      bInterfaceProtocol      0
      iInterface              0
      CDC Header:
        bcdCDC               1.10
[... 263 lines cut ...]
    params = self.send_with_response(msg, 'identify_response')
  File "/home/pi/klipper/klippy/serialhdl.py", line 186, in send_with_response
    return src.get_response(cmd, self.default_cmd_queue)
  File "/home/pi/klipper/klippy/serialhdl.py", line 242, in get_response
    self.serial.raw_send_wait_ack(cmd, minclock, minclock, cmd_queue)
  File "/home/pi/klipper/klippy/serialhdl.py", line 178, in raw_send_wait_ack
    raise error("Serial connection closed")
...
more of this message
...
error: Serial connection closed
Timeout on serial connect
Wait for identify_response
Traceback (most recent call last):
  File "/home/pi/klipper/klippy/serialhdl.py", line 68, in _get_identify_data
    params = self.send_with_response(msg, 'identify_response')
  File "/home/pi/klipper/klippy/serialhdl.py", line 186, in send_with_response
    return src.get_response(cmd, self.default_cmd_queue)
  File "/home/pi/klipper/klippy/serialhdl.py", line 242, in get_response
    self.serial.raw_send_wait_ack(cmd, minclock, minclock, cmd_queue)
  File "/home/pi/klipper/klippy/serialhdl.py", line 178, in raw_send_wait_ack
    raise error("Serial connection closed")
error: Serial connection closed
Timeout on serial connect
```

kondas, 2020-07-04:
[Uploading klipper.log…]()

kondas, 2020-07-04:
[klipper.log](https://github.com/KevinOConnor/klipper/files/4872753/klipper.log)

BlackStump, 2020-07-04:
If you still have mention of marlin in your usb-id then you have not flashed the Klipper firmware to the skr1.4
ie from you config
>[mcu]
>serial = /dev/serial/by-id/usb-marlinfw.org_Marlin_USB_Device_0D006023AF4A1C875D7C39AEF50020C7-if00

kondas, 2020-07-04:
Hi, thanks for the comment. I also thought that, after seeing similar comments in other threads. 
After i build klipper.bin, i copy it to the microSD card as firmware.bin, and after the next reboot it gets renamed to firmware.cur. Since i have no explicit way to check if the firmware flash is successful, from this it looked like it is flashed correctly.

closed 2020-07-05 by kondas