Klipper3d/klipper #3051: Failed bed_mesh with bltouch

leZudiste, 2020-07-04:
[klippy.log](https://github.com/KevinOConnor/klipper/files/4872959/klippy.log)
Hello,
I cant use bed_mesh with blTouch.
The homming and SCREWS_TILT_CALCULATE  work perfectly just bed_mesh not working.

Thanks

DroneMang, 2020-07-04:
Here is my cfg. I have the mesh working.  Also a second file from someone else on the internet with 3 lead screws, the 5160 file.

[printer20.zip](https://github.com/KevinOConnor/klipper/files/4873866/printer20.zip)

KevinOConnor, 2020-07-10:
Alas, there is little we can do with such a vague description of a problem.  As far as we know, the code works as intended and will perform correctly when following the directions.

-Kevin

closed 2020-07-25 by KevinOConnor