Klipper3d/klipper #3052: default_parameter_EXTRUDER_TEMP overrides Slicer Temp

DroneMang, 2020-07-04:
Hi,

In my start script I define "default_parameter_EXTRUDER_TEMP", but it appears I have to change it for each print.  If I print at 0.1 layer height, I use a lower temp.  I found that the slicers temp would not update klippers set temp for the extruder once the print starts after the start script has run.  Can I just omit default_parameter_EXTRUDER_TEMP somehow?

Thanks,
DroneMang

[klippy3.zip](https://github.com/KevinOConnor/klipper/files/4873026/klippy3.zip)

DroneMang, 2020-07-04:
I have solved it.  Here is my start script.  S3D handles a lot of what is in the example start script.

Could a simplified start script be added to the documentation, and noted in the slicers guide?

[gcode_macro START_PRINT]
gcode:
    BED_MESH_PROFILE LOAD=test_5_17x8
    # Use absolute coordinates
    G90
    # Reset the G-Code Z offset (adjust Z offset if needed)
    SET_GCODE_OFFSET Z=+0.45
    # Home the printer
    G28

ld3300, 2020-07-07:
I believe the startup script is a really good starting place, but perhaps it needs an example of how to implement it in the slicer to function properly.  It is intended to pass the temperatures to the command and replace  or supplement the existing slicer g-code

KevinOConnor, 2020-07-25:
I'm unsure what is being reported here.  The parameters can be modified in the config, manually at the start of each print, or in the slicer - that choice is yours.  There is documentation on this in docs/Command_Templates.md.

-Kevin

DroneMang, 2020-07-25:
Okay. I think I've got this one figured out, thanks! This can be closed.

closed 2020-07-25 by KevinOConnor