Klipper3d/klipper #3053: Unable to print from sdcard sub directory.

changedsoul, 2020-07-04:
[klipper.log](https://github.com/KevinOConnor/klipper/files/4873052/klipper.log)

OK, so I use DWC2 for my interface and I dont know if its that causing this issue or not, but the log points to virtual_sdcard.py as the culprit, so thats why I am posting here in klipper.

When I try and print a file in a sub directory I get an error saying it cannot open the file. 
`DWC2:75281.96 - 10.0.0.44 - GET - /rr_gcode?gcode=M32%20%220%3A%2Fgcodes%2FCalibration%2Fsquare_tower.gcode%22
virtual_sdcard file open
Traceback (most recent call last):
  File "/home/pi/PrinterFarm/KlipperFarm/klipper/klippy/extras/virtual_sdcard.py", line 102, in cmd_M23
    try:
KeyError: 'gcodes/calibration/square_tower.gcode'
failed: M32Unable to open file`

I can print it from the root of the gcodes folder, so I dont know why its failing.
Does klipper not support printing from sub folders? Perhaps DWC2 is sending some garbage over to klipper? I dont really know how the two interact.

Hywelmartin, 2020-07-04:
You have to file that here https://github.com/Stephan3/dwc2-for-klipper

KevinOConnor, 2020-07-10:
I'm going to close this issue as this issue tracker is only for use when running the code from this repo.  (Dwc2 alters the internal state of Klipper in a way that we can not support.)

-Kevin

closed 2020-07-10 by KevinOConnor