Klipper3d/klipper #3055: Feature reguest -- Extruder Filament Force Sensor

p2004r, 2020-07-04:
A long time ago, in a 3d_print_forum, far away…

http://airtripper.com/1338/airtripper-extruder-filament-force-sensor-introduction/

When the extruder feeds the filament to the hot end, the extruder is effectively pushing against the filament causing the extruder to apply extra load on the load cell. Load cells have strain gauges attached that change in electrical resistance when under different loads. This resistance change provides small voltage levels that can be amplified and then read by an analogue to digital converter.

Basically, the purpose of doing these graphs is to find out what sort of information we can get out of the load cell that is used for the filament force sensor. We want to find out if the graphs can be used as a guide for better extruder set-ups.

This is a very cheap device, all parts for it cost now 2 - 3 US dollars. (HX711 + 20kg tensosensor)

By connecting this sensor to Klipper, the retract distance, the maximum printing speed and the optimal nozzle temperature are automatically adjusted. Filament plugs, nozzle clogging, filament breakage and damage to the fitings are recognized.

Hywelmartin, 2020-07-04:
#2977

KevinOConnor, 2020-07-06:
Sounds useful.  An interested developer would need to implement and submit it though.  If the target sensor just uses an analog pin to read the load then this could be implemented with just host Python code (see klippy/extras/hall_filament_width_sensor.py as an example).  If the sensor is using an hx711 than it looks like there would also need to be low-level C micro-controller code written for that chip's bit-banging interface.

Cheers,
-Kevin

Hywelmartin, 2020-07-06:
This would be perfect on a pellets feed machine.. as a pressure gauge..

p2004r, 2020-07-06:
> If the sensor is using an hx711 than it looks like there would also need to be low-level C micro-controller code written for that chip's bit-banging interface.

This load cell really uses HX711. This ADC has a maximum measurement frequency of 8 Hz and is probably not suitable for operation control with a slight delay.
In my opinion, the printer control software only requires a simple transfer a batch of the sensor values collected over a longer period of time (at least 1-3 seconds, but this interval can be set as a configurable parameter).
These accumulated values must be assigned to the " Status Printer" by the printer control board at the time of their measurement.

For example a single "attributed" measurement with HX711 from this batch:

- Time;
- Extruder Filament Force;
- Speed X: Y: Z: E.
- Print mode (line printing, rollback);
- Time from the start of the scheduled retraction;
- Time until the end of the scheduled retraction;

I hope this requires very simple low-level code in the printer controller.

All these batches of measurement tables are stored in Clipper and further analysed there. The easiest way to do this is to simply draw a continuous graph of pressure over time.

For the last 10 retractions, these rare filament measurements can be placed on the timeline relative to the common virtual point of the start of scheduled retraction. The result is the "average filament force on start of retract".

Similarly, a "average filament force on end of retract" can be obtained by placing samples relative to the "planned end of withdrawal".

When the printer "prints the line", you can link the measurements to the print speed. Or simply display the average filament feed force during line printing for the last 5 seconds, 15 seconds and 1 minute, as in the operating system performance monitors.

I am sure that after gaining experience and data it will be possible to adapt many print constants later automatically and precisely.

DroneMang, 2020-07-07:
This is interesting.  An alternate implementation I have thought of is to use Stallguard with the diag pin connected for TMC drivers on E, this could at least detect a blockage.  I'm not sure if you can get the current use live, if so, a pressure vs current relationship could be made to get the same data.

p2004r, 2020-07-07:
> This is interesting. An alternate implementation I have thought of is to use Stallguard with the diag pin connected for TMC drivers on E, this could at least detect a blockage. I'm not sure if you [...]

The filament is not hard enough, the extruder gears slipping on maximum strength relative to the filament surface. The exact length of the filament stretched by the extruder can give an optical encoder. Cheap and accurate filament real motion sensors are available. For example https://www.thingiverse.com/thing:2515750

yschroeder, 2020-07-09:
Stallguard cannot only be used to detect a stall, but also indicates overall motor load. The register value changes with motor load. Triggering the diag pin at a stall is only a bonus feature of this load measurement.

p2004r, 2020-07-10:
> Stallguard cannot only be used to detect a stall, but also indicates overall motor load. The register value changes with motor load. Triggering the diag pin at a stall is only a bonus feature of this load measurement.

https://www.trinamic.com/fileadmin/assets/Support/Appnotes/AN002-stallGuard2.pdf

This measurement method shows no micro effort during the end off retract (when the extruder is stopped, the measurement values are not clear). The second problem is the extruder gear. For example, a worm gear completely distorts the "force on the filament" display. Instead of "the effort put into the filament" we will see a complete mismatch with measurements at Retract.

I bought a set of load sensors and ADC HX711. The price for the load sensor and HX711 load cell was USD 2 upon delivery. In my case, the delivery unfortunately takes 30-40 days.

An inexpensive stepper motor microcontroller costs $ 21 and there is no ready-made load measurement function (unfortunately, I don't even suspect if he can after the firmware update). Also, my printer board has built-in drivers and I can't connect this external driver at all. Experimenting with him is too expensive for me. I don't do 3D printing for commercial purposes to compensate for the cost of experimenting with stepper motor controls and servo motors.

The only thing that I can still try financially is to install an encoder in the sensor that measures the actual movement of the filament and combines the measured values of two systems (strain gauge + encoder for the movement of the filament).

yschroeder, 2020-07-10:
Yes, I still think doing this with a proper load cell is still the better option.

By the way there is no need for a hx711. A load cell is basically a resistor. It has 4 wires to be able to measure the voltage drop at the resistor without measuring the voltage drop of the cables. You can build an analog amplifier to read the value with an ADC of the MCU. That way it will work with the current klipper MCU code.

p2004r, 2020-07-10:
>  You can build an analog amplifier to read the value with an ADC of the MCU. That way it will work with the current klipper MCU code.

Using the printer's integrated ADC controller itself would really use load sensor without changing the firmware.

The HX711 is such an amplifier with interference suppression filters that are integrated with a 24-bit ADC. 

As write to electronic forum, the load sensor has an extremely weak signal. The independent manufacture of such an amplifier is a difficult task. HX711 has 6-8 low order bits filled with noise. 

Perhaps you know an inexpensive amplifier that is available on the market?

When measuring the filament feed force, a maximum of 15,000 grams must be measured simultaneously (when printing with the maximum output of the hot end and the geared extruder) and at least 5 g force at the end of the retraction (to see how the filament breaks off from the alloy in the nozzle ). 

We need at least a 14-bit ADC.

yschroeder, 2020-07-10:
Without ever putting hands on this hardware, I guess the following would work:
https://m.aliexpress.com/item/33003166472.html

You can configure the out to be a 0-5V signal and set zero point and gain via potentiometers.

Sineos, 2020-07-11:
Are you aware, that there is a lot more information and schematics / sketch already available?

https://airtripper.com/1397/electronic-kitchen-scales-teardown-versus-load-cells/
https://airtripper.com/1473/airtripper-extruder-filament-force-sensor-design-3d-print/
https://airtripper.com/1626/arduino-load-cell-circuit-sketch-for-calibration-test/

yschroeder, 2020-07-11:
Oh, right. No need to reinvent the wheel. I didn't look through that website before, but it is described very extensively.

Especially the last page that describes the amplifier is worth a read.

p2004r, 2020-07-11:
> Without ever putting hands on this hardware, I guess the following would work:
> https://m.aliexpress.com/item/33003166472.html
> 
> You can configure the output to be a 0-5V signal and set zero point and gain via potentiometers.

Thank you for finding a relatively cheap measuring amplifier. The price on delivery to me is just over $ 15. It's still expensive (at least for me. Almost the cost of a stepper motor driver with no display option). However, if there are insurmountable difficulties integrating the filament performance sensor into the printer software, I will come back to using the amplifier.

I have already read this article on links. I had the impression that the filament feed force sensor may not have been widespread at that time due to the high total costs and the low measuring accuracy. Under ideal conditions, 10-bit ADC delivers a total of 1024 samples per measuring range. Modern extruders with gearwheels offer a maximum filament feed force of 15 kilograms. It is almost impossible to notice the subtle effects of the filament retract in this measurement range. It is also difficult to notice the influence of temperature on the filament feed.

(Of course, these are just my personal assumptions, based on the graphics from the original article about the filament power sensor. I can only get my own data in at least two months.)

Sineos, 2020-07-11:
> Modern extruders with gearwheels offer a maximum filament feed force of 15 kilograms

Doing such measurements is a long-time todo on my list. Personally I doubt that up to 15 kg is a realistic value in normal operation (although extruder marketing boasts with it).
At this force I would expect bowden and couplers to die long before.

p2004r, 2020-07-11:
> At this force I would expect bowden and couplers to die long before.

Bowden cannot withstand this effort, but many install a direct extruder. The direct extruder has a solid connector assembly with a cooler and can increase strength while the filament can withstand.

yschroeder, 2020-07-11:
How would you measure the force on a direct drive? That airtripper setup is for bowden, right?

p2004r, 2020-07-11:
> How would you measure the force on a direct drive? That airtripper setup is for bowden, right?

The extruder must be attached to the carriage through a load sensor (in a single point).

KevinOConnor, 2020-08-05:
Any further updates on this?

-Kevin

p2004r, 2020-08-06:
> Any further updates on this?

So far I have only received (just yesterday :)) the two hx711 debug boards I ordered. The order for load cells has not yet been delivered.

p2004r, 2020-09-02:
I am glad to inform you that today I received load cells (4 pieces, from 1 kg to 20 kg). I have also received arduino, optical stop-end, hx711.
So now I can start my experiments to measure the feed force of filament by the extruder into the hot end of the printer during the printing process. 
I would appreciate any advice and assistance from the participants in this discussion on this project to add a filament control system to the printer.

closed 2020-10-01 by github-actions[bot]