Klipper3d/klipper #3056: BLTouch wont trigger

wjg68, 2020-07-05:
[klippy.log](https://github.com/KevinOConnor/klipper/files/4874876/klippy.log)

[klippy.log](https://github.com/KevinOConnor/klipper/files/4874878/klippy.log)

having a couple issues. when i logon to octoprint first time it doesnt connect, times out. i have to restart octoprint or close it all together and relog in and then it connects. also, bltouch deploys but doesn't trigger when i push the probe up . 

thanks to all for the past help in getting me to this point. you guys rock.

wjg68, 2020-07-05:
think i have the issue with first boot fixed. did a m119 and x-triggered, y-open,z-open.

wjg68, 2020-07-05:
make that x=open,y-open,z-open. rebooted and did a m119 and all are open till i do a home then x is triggered while z is lowering to plate( turned off due to z not triggering). it probed z at the front left corner with the probe off the bed.

wjg68, 2020-07-05:
bltouch turns red when i push the probe up w finger but stepper keeps lowering the z.

ld3300, 2020-07-07:
have you been able to make it through the [BL touch guide?](https://github.com/KevinOConnor/klipper/blob/f0c394de8111ff7d498efd2e947d1d36788a9ad3/docs/BLTouch.md)

David

wjg68, 2020-07-14:
Finally able to get back to it, thanks for reply. tried bltouch guide, no triggered when i hold up on plunger. gonna take it apart and see what my wiring looks like. just had it all apart doing some mods and prob got something crossed.

KevinOConnor, 2020-08-05:
I'm going to close this as it looks like the conversation has concluded.  As far as we know, the bltouch code works as intended.

-Kevin

closed 2020-08-05 by KevinOConnor