Klipper3d/klipper #3057: delta_calibrate

goosst, 2020-07-05:
Hello,

Thanks for the great firmware.
I have a question about the delta_calibrate procedure:

```During the delta calibration process it may be necessary for the printer to probe below what would otherwise be considered the plane of the bed. It is typical to permit this during calibration by updating the config so that the printer’s minimum_z_position=-5. (Once calibration completes, one can remove this setting from the config.)```

Can the intention of the automatic delta_calibrate procedure be explained. Should the nozzle touch the bed during this procedure and klipper will automatically recognize this point on the bed?
I've tried to put the minimum_z_position at -5 and -10, but it seems to stop going lower in z when the inductive sensor on the head triggers. (i.e. there is no difference in the probed z-position independent to what I put this number. But the probing stops relatively high above the bed.)

Of course I've followed the DELTA_CALIBRATE METHOD=manual. But since I have a sensor on the head, I would like to understand if the automatic procedure can still be used in this case. Or if I should follow the manual procedure and make the sensor be useful to create the bed mesh.

Thanks and keep up the good work!

Sorry if this is not the right location for this type of question, but it doesn't seem like there is a forum or chatgroup?
[klippy.log](https://github.com/KevinOConnor/klipper/files/4875446/klippy.log)

Hywelmartin, 2020-07-05:
If you have a probe that is offset in x and y, I would recommend to do a DELTA_CALIBRATE METHOD=manual.. 
a slight effector tilt will screw up the calibration...

goosst, 2020-07-07:
thank you for the tip, I've compared manual and automatic method. 
The inconsistency between how the perceived bed shape is between the paper test and probe test is just day and night  ...

closed 2020-07-16 by goosst