Klipper3d/klipper #3060: square_tower.stl not a valid STL file

viperfan7, 2020-07-06:
Not actually a code issue, but it seems the square tower STL used by the pressure advance tuning guide isn't a valid STL file, it refuses to load in both cura and outright causes simplify3d to crash

BlackStump, 2020-07-06:
I just downloaded from [here](https://www.klipper3d.org/Pressure_Advance.html) and loaded it up in S3D without issue
[direct link](https://www.klipper3d.org/prints/square_tower.stl)

Hywelmartin, 2020-07-06:
or grab it from your Pie..
/home/pi/klipper/docs/prints/

Sineos, 2020-07-06:
If you try to download it directly from the github repo, you will receive a html file. Either use the options above or use https://gitcdn.link/ to grab it

closed 2020-07-13 by klipper-gitissuebot

viperfan7, 2020-07-18:
THis issue isn't with klipper itself, its with the repository

> It appears the Klipper log file was not provided.
> I'm closing this now, sorry. Feel free to attach the log to this ticket sometime in the future, or create a new issue once you can provide the log file.

This issue is not with klipper itself, its with the repository