Klipper3d/klipper #3061: Commit 4bdc11a8b3843b73079a2c8dfc8d8c0928c29662 left repo at a broken state.

Nandox7, 2020-07-06:
With above commit added klipper fails to start with the error bellow:

Starting Klippy...
Args: ['/home/pi/klipper/klippy/klippy.py', '/home/pi/printer.cfg', '-l', '/tmp/klippy.log']
Git version: 'v0.8.0-623-g3a57608'
CPU: 4 core ARMv7 Processor rev 4 (v7l)
Python: '2.7.9 (default, Sep 26 2018, 05:58:52) \n[GCC 4.9.2]'
Building C code module c_helper.so
Unable to build C code module (error=256)

Have to revert to 7bc64700827bf7cfe118bfdf8787c5817202db5e to be able to start Klipper again.

BlackStump, 2020-07-06:
seem to start fine for me at least
>Starting Klippy...
Args: ['/home/debian/klipper/klippy/klippy.py', '/home/debian/printer.cfg', '-l', '/tmp/klippy.log']
Git version: 'v0.8.0-623-g3a576086'
CPU: 1 core ARMv7 Processor rev 2 (v7l)
Python: '2.7.16 (default, Oct 10 2019, 22:02:15) \n[GCC 8.3.0]'
Building C code module c_helper.so
Start printer at Mon Jul  6 12:01:34 2020 (1594036895.0 336.1)
===== Config file =====

Nandox7, 2020-07-06:
Until I went one commit back it wouldn't work for me.
`git checkout 7bc64700827bf7cfe118bfdf8787c5817202db5e7bc64700827bf7cfe118bfdf8787c5817202db5e/`

always stuck at: Unable to build C code module (error=256)

BlackStump, 2020-07-06:
At a guess your os is out of date and/or you need to run the klipper install command ie install-octopi.sh

dmbutyugin, 2020-07-07:
@Nandox7 

Can you run
```
$ git checkout 4bdc11a8b3843b73079a2c8dfc8d8c0928c29662
$ gcc -c -O2 -Wall klippy/chelper/kin_shaper.c
```
and post the error here?

Nandox7, 2020-07-07:
Sure, see results bellow.

```
 $ git checkout 4bdc11a8b3843b73079a2c8dfc8d8c0928c29662
Previous HEAD position was 7bc6470... config: Added some sample glyphs for 128x6                                                            4 screens (#3028)
HEAD is now at 4bdc11a... input_shaper: Initial support of input shaping (#3032)

$  gcc -c -O2 -Wall klippy/chelper/kin_shaper.c
klippy/chelper/kin_shaper.c: In function ‘calc_position’:
klippy/chelper/kin_shaper.c:53:5: error: ‘for’ loop initial declarations are only allowed in C99 or C11 mode
     for (int i = 0; i < n; ++i)
     ^
klippy/chelper/kin_shaper.c:53:5: note: use option -std=c99, -std=gnu99, -std=c11 or -std=gnu11 to compile your code
```

dmbutyugin, 2020-07-07:
@Nandox7 
Aha, and which GCC version (and Raspbian)?
```
$ gcc -v
```

Nandox7, 2020-07-07:
Was just checking that out.
gcc version 4.9.2 (Raspbian 4.9.2-10+deb8u2) <<< Raspbian GNU/Linux 8 (jessie)

I was looking for that build command but had to print and haven't dig much into it.

dmbutyugin, 2020-07-07:
@Nandox7 thanks, I'll make a fix.

dmbutyugin, 2020-07-07:
@Nandox7 Though, to be fair, Jessie is a pretty old distro, and its official support supposedly ended this June.

Nandox7, 2020-07-07:
@dmbutyugin I know. Few reasons why I haven't upgraded yet, have quite a few customizations on top of it that will take some time to redo.

closed 2020-07-13 by klipper-gitissuebot