Klipper3d/klipper #3062: [Q] Tuning TMC Chopper Registers

Sineos, 2020-07-06:
I'm experimenting with the TMC driver registers. In my printer I have TMC5160 drivers.
According to the Excel spreadsheet https://www.trinamic.com/fileadmin/assets/Products/ICs_Documents/TMC5160_Calculations.xlsx and my stepper specification, I get following result:

[image]

Doing a `DUMP_TMC STEPPER=stepper_x` with the default (unmodified) parameters yields:
```
========== Write-only registers ==========
COOLCONF: 00000000
GLOBALSCALER: 0000004f GLOBALSCALER=79
IHOLD_IRUN: 00061f10 IHOLD=16 IRUN=31 IHOLDDELAY=6
PWMCONF: c40c001e PWM_OFS=30 pwm_autoscale=1 pwm_autograd=1 PWM_REG=4 PWM_LIM=12
TPOWERDOWN: 0000000a TPOWERDOWN=10
========== Queried registers ==========
GCONF: 00000000
CHOPCONF: 12410153 toff=3 hstrt=5 hend=2 tbl=2 tpfd=4 MRES=2(64usteps) intpol=1
GSTAT: 00000005 reset=1(reset) uv_cp=1(Undervoltage!)
DRV_STATUS: 81100048 SG_RESULT=72 CSACTUAL=16 stallGuard=1 stst=1
FACTORY_CONF: 0000000c FACTORY_CONF=12
IOIN: 30000050 DRV_ENN=1 SD_MODE=1 VERSION=0x30
LOST_STEPS: 00000000
MSCNT: 0000001e MSCNT=30
MSCURACT: 00f3002d CUR_A=45 CUR_B=243
OTP_READ: 0000000c OTP_FCLKTRIM=12
PWM_SCALE: 0000000f PWM_SCALE_SUM=15
PWM_AUTO: 0000001d PWM_OFS_AUTO=29
TSTEP: 000fffff TSTEP=1048575

```

Now starting with the values from the spreadsheet:
```
driver_TOFF: 3
driver_HSTRT: 0
driver_HEND: 0
```

Doing a `DUMP_TMC STEPPER=stepper_x` now:
```
========== Write-only registers ==========
COOLCONF: 00000000
GLOBALSCALER: 0000004f GLOBALSCALER=79
IHOLD_IRUN: 00061f10 IHOLD=16 IRUN=31 IHOLDDELAY=6
PWMCONF: c40c001e PWM_OFS=30 pwm_autoscale=1 pwm_autograd=1 PWM_REG=4 PWM_LIM=12
TPOWERDOWN: 0000000a TPOWERDOWN=10
========== Queried registers ==========
GCONF: 00000000
CHOPCONF: 12410003 toff=3 tbl=2 tpfd=4 MRES=2(64usteps) intpol=1
GSTAT: 00000005 reset=1(reset) uv_cp=1(Undervoltage!)
DRV_STATUS: 81100048 SG_RESULT=72 CSACTUAL=16 stallGuard=1 stst=1
FACTORY_CONF: 0000000c FACTORY_CONF=12
IOIN: 30000050 DRV_ENN=1 SD_MODE=1 VERSION=0x30
LOST_STEPS: 00000000
MSCNT: 0000001e MSCNT=30
MSCURACT: 00f3002d CUR_A=45 CUR_B=243
OTP_READ: 0000000c OTP_FCLKTRIM=12
PWM_SCALE: 0000000f PWM_SCALE_SUM=15
PWM_AUTO: 0000001d PWM_OFS_AUTO=29
TSTEP: 000fffff TSTEP=1048575
```

`hstrt` and `hend` are suddenly missing from the `DUMP_TMC STEPPER` output.

Is this intended behavior? Is 0 for `hstrt` and `hend` valid in Klipper?

[klippy.log](https://github.com/KevinOConnor/klipper/files/4880000/klippy.log)

FHeilmann, 2020-07-07:
DUMP_TMC only explicitly decodes anbd displays registers that are non-0 or have some special significance when they are 0 (such as interpolation). so your attempt to set HSTRT and HEND to 0 was a scuccess.

DroneMang, 2020-07-07:
I have an SKR Pro with 5160's and 0.9 steppers 24V, its an FT-6 printer so I have 2 steppers in parallel for X.  What have you found in working on this?  Are the default values in Klipper good?  I have some resonances occasionally I don't like.

Sineos, 2020-07-07:
I'm using 24V and 0.9° steppers as well.
I'm at a very early stage to experiment with this and these values did not make any noticeable difference at first glance. 

I wanted to do some resonance testing according to @dmbutyugin input shaping and maybe get an osziloscope to tap on the Rsense as per TMC's instructions. (Although I have near to none experience with such, so the result might be questionable)

TMC also reccomends some range of HSTRT and HEND for the driver to do some "self optimization". So with the registers set to 0 / 0, the effective range is 1 to -3. If this is good or bad, I don't know.

DroneMang, 2020-07-07:
What board are you running?  My clock speed may be different than yours.  I am running stock Klipper with px advance, I get a little blowing out of corners still, if I decrease more the walls don't bond well.

How do I run/test these new branches?  I have seen the posts about input shaping and s-curve, but the s-curve branch seems very old.  Is there any benefit to trying these new features yet?

Sineos, 2020-07-07:
@FHeilmann Thanks for clarifying. If you can share some experience with these parameters, it would be highly appreciated.

@DroneMang 
I think the new input shaping technique is excellent.
See here for more details:

1. https://github.com/KevinOConnor/klipper/pull/2030#issuecomment-637881098 and following posts
2. https://github.com/dmbutyugin/klipper/blob/scurve-shaping/docs/S-Curve.md#tuning up to "Selecting max accel"

The input shaping is meanwhile merged to Klipper master, so no need for branch switching except you want to try measuring your resonances with an accelerometer. Then follow link [1] and use https://github.com/dmbutyugin/klipper/tree/adxl345-spi

> What board are you running?
SKR 1.4 Turbo

DroneMang, 2020-07-07:
>    1. #2030 (comment)

I'm trying to set up a two-in-one hot end, from what I've gathered so far I
need to set up a T0 and T1 macro to handle the filament swap. Do you happen
to have a cfg for such a setup?

FHeilmann, 2020-07-07:
@Sineos sorry I haven't really played with the parameters as defaults have always worked fine for me. So I can't offer any insight unfortunately.

Sineos, 2020-07-08:
@DroneMang 
Sorry, I cannot help you there. Basically I'm a complete Klipper newbie and started with Klipper a couple of weeks ago because I found the resonance measuring tempting. Until then I have been long time using Marlin and was kind of "afraid" of Klipper. I wish I had switched earlier because I think it is a really great firmware and in many aspects superior to Marlin.

@FHeilmann 
Do you happen to know:
- How can I determine if the stepper currently is at a full step position? Is it visible from the driver registers?
- What is the fCLK value for the drivers in Klipper? 16 MHz? Depending on the Board?

FHeilmann, 2020-07-08:
@Sineos have a look at the TMC5160 datasheet page 47. You should be able to use the MSCNT register to determine at which microstep the driver is currently sitting.

The clocking depends on the board, since the TMC5160 can operate with or without an external clock source it depends on the board. I would assume that most boards don't bother clocking them externally, so the internal clock is used which runs at 12 Mhz and comes factory trimmed (see page 35 of the datasheet)

Hope this helps.

Sineos, 2020-07-08:
> have a look at the TMC5160 datasheet page 47. You should be able to use the MSCNT register to determine at which microstep the driver is currently sitting.

Many thanks. I noticed that the count goes over 256. Would it be correct to say that `MSCNT mod 4 = 0` means I have reached a full step position?

> The clocking depends on the board, since the TMC5160 can operate with or without an external clock source. I would assume that most boards don't bother clocking them externally, so the internal clock is used which runs at 12 Mhz and comes factory trimmed (see page 35 of the datasheet)

If it was determined by the MCU then Klipper would have to set it via firmware, right?

changedsoul, 2020-07-09:
This thread interests me. I was trying to bench test my TMC5160 with TMC's supplied Arduino code, but I could not get the motor to turn at all. I wanted to tune them off the printer so I could play with values so I would not have to  "FIRMWARE_RESTART every change, Home, Test, etc....."

I was having a problem where I got some strange noise when trying to rapid around 200mm/s. Strange thing, they were not stalling or missing steps. To eliminate any chassis resonance, I removed all my belts and tried them with no load. 

Well trying to move them with no load caused them to stall immediately. I do not know what this means. I can grip the motor shaft and it starts to spin. When I let go, it stalls with the normal sound like its trying to go to fast, as if the driver needs this little feedback to function properly. 
My XY is very lite and the real only resistance to the motors is the belt tension. So its my hopes, someone here knowing more about tuning these might know what I have to tweak to fix this issue?

TMC_DUMP right after home:  (CoreXY Printer)

> DUMP_TMC stepper_x
========== Write-only registers ==========
TPWMTHRS:   00000009 TPWMTHRS=9
COOLCONF:   00000000
GLOBALSCALER: 00000043 GLOBALSCALER=67
IHOLD_IRUN: 00061f13 IHOLD=19 IRUN=31 IHOLDDELAY=6
PWMCONF:    c40c001e PWM_OFS=30 pwm_autoscale=1 pwm_autograd=1 PWM_REG=4 PWM_LIM=12
TPOWERDOWN: 0000000a TPOWERDOWN=10
========== Queried registers ==========
GCONF:      00000004 en_pwm_mode=1
CHOPCONF:   15410153 toff=3 hstrt=5 hend=2 tbl=2 tpfd=4 MRES=5(8usteps) intpol=1
GSTAT:      00000005 reset=1(reset) uv_cp=1(Undervoltage!)
DRV_STATUS: 80134000 stealth=1 CSACTUAL=19 stst=1
FACTORY_CONF: 0000000c FACTORY_CONF=12
IOIN:       30000040 SD_MODE=1 VERSION=0x30
LOST_STEPS: 00000000
MSCNT:      000003f0 MSCNT=1008
MSCURACT:   00f601e8 CUR_A=-24 CUR_B=246
OTP_READ:   0000000c OTP_FCLKTRIM=12
PWM_SCALE:  00010030 PWM_SCALE_SUM=48 PWM_SCALE_AUTO=1
PWM_AUTO:   0000004c PWM_OFS_AUTO=76
TSTEP:      000fffff TSTEP=1048575

> DUMP_TMC stepper_y
========== Write-only registers ==========
TPWMTHRS:   00000009 TPWMTHRS=9
COOLCONF:   00000000
GLOBALSCALER: 00000043 GLOBALSCALER=67
IHOLD_IRUN: 00061f13 IHOLD=19 IRUN=31 IHOLDDELAY=6
PWMCONF:    c40c001e PWM_OFS=30 pwm_autoscale=1 pwm_autograd=1 PWM_REG=4 PWM_LIM=12
TPOWERDOWN: 0000000a TPOWERDOWN=10
========== Queried registers ==========
DWC2:52891.53 - 10.0.0.30 - GET - /rr_config
GCONF:      00000004 en_pwm_mode=1
CHOPCONF:   15410003 toff=3 tbl=2 tpfd=4 MRES=5(8usteps) intpol=1
GSTAT:      00000005 reset=1(reset) uv_cp=1(Undervoltage!)
DRV_STATUS: 80134000 stealth=1 CSACTUAL=19 stst=1
FACTORY_CONF: 0000000f FACTORY_CONF=15
IOIN:       30000040 SD_MODE=1 VERSION=0x30
LOST_STEPS: 00000000
MSCNT:      000000f0 MSCNT=240
MSCURACT:   001700f6 CUR_A=246 CUR_B=23
OTP_READ:   0000000f OTP_FCLKTRIM=15
PWM_SCALE:  00060033 PWM_SCALE_SUM=51 PWM_SCALE_AUTO=6
PWM_AUTO:   00000049 PWM_OFS_AUTO=73
TSTEP:      000fffff TSTEP=1048575

robthide37, 2020-07-10:
why are you factory clocks different on those 2 steppers?

changedsoul, 2020-07-10:
I have no clue. Which one shows the clocks?

robthide37, 2020-07-10:
i assume OTP_READ: 0000000c OTP_FCLKTRIM=12
OTP_READ: 0000000f OTP_FCLKTRIM=15
and FACTORY_CONF: 0000000f FACTORY_CONF=15 unless we use a otp programmer. if you look at the an001 spreadcycle doc on trinamic website their is alot more explanation of the settings. 
I ordered a 300 dollar scope hoping it is high enough resolution i believe it was 50 mhz 16 channel i saw some 100 dollar ones that went to 50 and 100khz but i am worried that they may not be best quality and not sure og the other specs. I figure if i will have much more info when they arrive and i have an assortment of steppers ldo and then steppers online and a few amzon/ali ones so i will try to compile as much data as i can since i have found nothing but gibberish when it comes to info on stepper configs

robthide37, 2020-07-10:
One thing i will say is before tuning i was at 960 mah and was 140 degrees now at 1.05 amp they are highest so far 118f and my pancake stepper at 800 mah was 120f this was on a 200mms tuning tower print

FHeilmann, 2020-07-10:
FCLKTRIM is as the name suggests a trim value which is programmed into the chip in the factory to make sure the internal clock runs at the expected rate (12mhz for the 5160)

changedsoul, 2020-07-11:
> One thing i will say is before tuning i was at 960 mah and was 140 degrees now at 1.05 amp they are highest so far 118f and my pancake stepper at 800 mah was 120f this was on a 200mms tuning tower print

I was also looking at getting a siglent or a rigdol scope to try and figure this stuff out. I have a single channel portable one that is sort of blah and don’t show anything useful. They are a bit steep in price for me though, so hoping you come up with some good data. 

I tried to set a driver up on the bench and drive it with the arduino but no matter what I did, I could not get the motor to turn. I gave up and just plugged them back into my printer.

robthide37, 2020-07-11:
Thankyou its very hard to come by good info. I assumed it was something set
at factory like a type of binning . Do you think their is any benefit to
switching steppers so both have the same trim for xy on a core xy and for
the z steppers if dual z?

changedsoul, 2020-07-25:
> Thankyou its very hard to come by good info. I assumed it was something set at factory like a type of binning . Do you think their is any benefit to switching steppers so both have the same trim for xy on a core xy and for the z steppers if dual z?
> […](#)

Agreed, info on this is very hard to come by. Have you made any progress?
I finally got my motors to turn on the bench. Happy day. 

Curious about your tuning you did so far....I tried using the values the spread sheet gave me and my motors got screaming hot. I can’t figure out why these run so much hotter than my TMC2100’s. 
Hot as in almost 70C where as before I could touch them and not get burned.

robthide37, 2020-07-25:
Did you make sure to lower the amperage and not run them at the max rated
rms? I had great results I actually am planning on try to see the diff with
the scope as far as spreadsheet vs defaukt timings. I also shot for 40khz
chopper timing and also 1.3 nano seconds blank time

changedsoul, 2020-07-25:
> Did you make sure to lower the amperage and not run them at the max rated rms? I had great results I actually am planning on try to see the diff with the scope as far as spreadsheet vs defaukt timings. I also shot for 40khz chopper timing and also 1.3 nano seconds blank time
> […](#)

Ya,  my current is set for .7. 
The Tmc2100 I don’t know what ran at, I guess I could put them in and physically measure the current draw and see what they are at. 
So far, it seems the tmc2100 is able to move my printer at 250mm/s where as the tmc5160 is stalling out at 175 and I can’t figure it out. 
I even increased the current to max and it still stalls so I am pretty sure it’s something else config wise causing it to mess up. 
What really confuses me is how the tmc5160 stalled out at a slow speed with zero belt tension, and while it was stalling, I grabbed the motor shaft and it started to spin. Like it was needing some resistance to operate. 
I don’t know what to make of that.

KevinOConnor, 2020-08-24:
I'm going to close this as it looks like the original question has been answered.

-Kevin

closed 2020-08-24 by KevinOConnor