Klipper3d/klipper #3067: Klipper shutdown while heating

kreaktiv-chris, 2020-07-08:
Hi,
i use Octoprint and my heatbed needs a long time to heat. 
In this case i do the follow steps:
Heating the bed to 80°C, heating the Nozzle to 175°C ( time -22 min )
waiting ... :)
Press start on the printjob. ( time -9:30 min )
The printjob try to heat the bed to 80° and the nozzle to 250°C ( time -9:30 min )
after maybe 20 secconds the printer shutdown while heating the nozzle. ( time -9:00 min)
after the shutdown i send M112 to klipper and restart the firmware to print the job again ( time -8 min )
This is not the first time this problem has arisen. What should i do?

Chris

[image]

[klipper.log](https://github.com/KevinOConnor/klipper/files/4890604/klipper.log)

jakep82, 2020-07-08:
I'm not sure I understand exactly what's being reported here, but if you're experiencing a shutdown because your extruder is too slow to heat, you can adjust the heater verification settings.  Try increasing the max_error setting.

https://github.com/KevinOConnor/klipper/blob/master/config/example-extras.cfg#L815

jakep82, 2020-07-08:
I'm also seeing some issues in your temperature reporting.  E0 is reporting 18C, E1 is reporting 10C, and your bed is reporting 23C.  You've configured non-standard pullup resistor values for your extruders, and they're different for each extruder.  Are you sure those values are correct?

yschroeder, 2020-07-09:
Could it also be the idle shutdown? Klipper turns off heaters if no print is started within some time after preheating, right?

Arksine, 2020-07-10:
Looking at the log file this doesn't appear to be related to the heater settings.  You can see at 855.0 seconds where the `extruder` target is set to 250.  At 861.0 its target is set to 0.   A gcode is being sent to Klipper that is resetting the target.  I can only guess that Octoprint is sending this gcode for some reason.

robthide37, 2020-07-13:
using an octoprint preheat plugin maybe ?

KevinOConnor, 2020-07-25:
I'm going to close this as it looks like the original question was answered (the log indicates klipper was commanded to turn off the heater).

-Kevin

closed 2020-07-25 by KevinOConnor