Klipper3d/klipper #3068: Is an enable pin needed for TMC2208 and TMC2209?

dewi-ny-je, 2020-07-08:
According to https://www.trinamic.com/fileadmin/assets/Products/ICs_Documents/TMC220x_TMC2224_datasheet_Rev1.09.pdf page 5 (the same applies to tmc2209) the control lines ENN, DIAG, INDEX, MS1, MS2 and the Vref pot setting can all be omitted when the driver is operated in UART mode 

I understand that Klipper uses that mode, since it can set various parameters via UART and (as far as I understood) the Vref trimmer is not relevant anymore (I haven't tested it yet, my printer is under heavy maintenance/rebuild).

Can the enable pin from the controller board be omitted too? This is not clear from the documentation.

If not, can the full UART mode be implemented so that one less pin is needed?

jakep82, 2020-07-08:
I can't answer your question about the enable pin, but you can't use "sensorless" homing on the 2209 without the diag pin.  The stallguard signal from the driver is sent through that pin, not through UART.

Gomer05, 2020-07-09:
I have tmc2208 running with the uart on a MKS Gen L for the last year and the below has worked form me:
Can't remember if I every tried it with out the enable_pin.

[stepper_x]
step_pin: ar54
dir_pin: !ar55
enable_pin: !ar38
step_distance: .0125
endstop_pin: ^ar3
position_endstop: 0
position_max: 235
homing_speed: 50

[tmc2208 stepper_x]
uart_pin: ar40
microsteps: 16
run_current: .6

dewi-ny-je, 2020-07-09:
> I can't answer your question about the enable pin, but you can't use "sensorless" homing on the 2209 without the diag pin. The stallguard signal from the driver is sent through that pin, not through UART.

True, but this meant that four lines are required by the chip, instead of five. Also, while sensorless homing requires a quick notification, the enable signal doesn't really need to be realtime so it's better suited to UART communication.

dewi-ny-je, 2020-07-09:
I checked the datasheet: CHOPCONF, page 29-30, bits 3,2,1,0: when 0000 driver disable.
Also, FREEWHEEL (0 for normal motor operation, 1-3 for motor disabled in various ways), TOFF (0 for driver off, 1-15 for motor enabled).
There are various options to disable the motor motion or even for the driver transistors via UART without needing a physical line.

KevinOConnor, 2020-07-10:
As indicated in the config/example.cfg file, the enable_pin parameter is optional, and so too is the wiring of it.

The Klipper tmc driver code has an additional feature that will enable/disable each driver individually via UART (or SPI) if the enable pin is shared or not specified.  Note though, if you're really designing your own board without an enable pin then I'd recommend using a global shared enable pins for all drivers - that way the micro-controller can rapidly disable all motors in the event of an emergency.

-Kevin

dewi-ny-je, 2020-07-11:
I checked again but I cannot find that information. Anyway, by searching "optional" or "shared" I found the passage in the source code.
I'll keep only a shared enable pin.

closed 2020-07-11 by dewi-ny-je