Klipper3d/klipper #3070: Delta not moving all axis at same speed or at the same time

TheBlackMini, 2020-07-09:
So I've issued G28 then DELTA_CALIBRATE and the printer moves Z (Tower C) really quickly but the others move slowly and it gets about 10mm from the bed and the whole thing stops and reports that the probe triggered early.

When it's in this position I then issue G28 again and Z tries to move to the endstop before the rest has had a chance to move 20mm.

Mainboard: BTT SKR Mini E3 1.2
Printer: Anycubic Kossel Linear Plus

Klippy.log [https://pastebin.com/b56MtjGZ](url)
Video [https://www.reddit.com/r/klippers/comments/hg1ayh/klipper_on_anycubic_kossel_linear_plus_issue_with/](url)

BlackStump, 2020-07-09:
check your config to make sure you have the same step distances on every tower
ie you have stepper a and b the same but not c
>step_distance = .0125
vs stepper_c
step_distance = .0025

TheBlackMini, 2020-07-09:
Ohhhh...dammmmn...

Thankyou! Stare at a problem long enough and everything looks OK! Updated the steps for C and volia all axis are moving correctly.

closed 2020-07-09 by TheBlackMini