Klipper3d/klipper #3071: Question: More than 3 axis?

Daniel-Kurth, 2020-07-09:
Hi, I'm planing a 3D printer project based around a delta printer which would require me to control up to 12 steppers via gcode:

3 steppers for the delta towers
2 Extruders
7 steppers additional axises
Is this possible with this firmware?

Daniel-Kurth, 2020-07-09:
I don't have an Issue which would require a log file.

Spacemarine2018, 2020-07-09:
You can control a large amount of manual steppers with the manual steppers command. I don't know of any firmware that is more flexible than klipper in that regard.

Daniel-Kurth, 2020-07-09:
> You can control a large amount of manual steppers with the manual steppers command. I don't know of any firmware that is more flexible than klipper in that regard.

Thanks for the quick reply!

closed 2020-07-09 by Daniel-Kurth