Klipper3d/klipper #3072: configuration Creality Ender 5 - incorrect for x and y home

njilk, 2020-07-09:
The Ender 5 has the x and y end stops not at 0 but at the max position, 235. 
Therefore the home positions are incorrect in the default https://github.com/KevinOConnor/klipper/blob/master/config/printer-creality-ender5-2019.cfg

The problem isn’t huge, you can rotate the object 180 degrees in the slicer, but it makes safe homing and bed mesh levelling using bed probing hardware a puzzle because of the inverted x end y position. 

For the changes to correct this:

[stepper_x]
step_pin: PD7
dir_pin: PC5 		        **- change to ! invert direction**
enable_pin: !PD6
step_distance: .012500
endstop_pin: ^PC2
position_endstop: 0 	**- change to 235**
position_max: 235
homing_speed: 30

[stepper_y]
step_pin: PC6
dir_pin: PC7  		       **- change to ! Invert direction**
enable_pin: !PD6
step_distance: .012500
endstop_pin: ^PC3
position_endstop: 0 	**- change to 235**
position_max: 235
homing_speed: 30

For completeness: 
Marlin settings for Ender 5: https://github.com/MarlinFirmware/Configurations/blob/release-2.0.5/config/examples/Creality/Ender-5/Configuration.h
[
// @section homing

// Specify here all the endstop connectors that are connected to any endstop or probe.
// Almost all printers will be using one per axis. Probes will use one or more of the
// extra connectors. Leave undefined any used for non-endstop and non-probe purposes.
//#define USE_XMIN_PLUG
//#define USE_YMIN_PLUG
#define USE_ZMIN_PLUG
**#define USE_XMAX_PLUG
#define USE_YMAX_PLUG**
//#define USE_ZMAX_PLUG

KevinOConnor, 2020-07-10:
@xelprep - any comments on this?

-Kevin

xelprep, 2020-07-12:
Feel free to modify it or revise in a PR. This is my first and only 3d printer so I just tweaked the Ender 3 config until it worked for me. I've always thought of the back right corner where the endstops are as being 0,0. Didn't realize that was counter to the actual design of the printer.

IIRC, that same corner was 0,0 when I was printing on Creality's built-in Marlin FW but I switched to Klipper shortly after obtaining the printer over a year ago so I might be wrong. I did have to invert the Y axis in Octoprint's Bed Level Visualizer to get an accurate view of the corner bed screws though so that might be a symptom of your described issue.

KevinOConnor, 2020-07-19:
Okay, thanks.  I made the requested change (commit 0f24406a).

-Kevin

cyrix2000, 2020-07-21:
It's all working well on my Ender 5.

njilk, 2020-07-22:
Great! Thanks for confirming.

closed 2020-07-25 by KevinOConnor