Klipper3d/klipper #3073: [FR] Implementing a "Fan Speed Factor Override Percentage".

JohnEdwa, 2020-07-09:
Recently slicers have been adding fan speed modifiers so that different features such as overhangs, bridges, infill and walls can have different amounts of cooling so you can prioritise adhesion vs visual quality. But sometimes the settings used aren't quite right for some model and I would like to tune them during a print, which is impossible with only M106 as that gets immediately overridden by the next change in Gcode. Therefore I'd like to suggest implementing a multiplier like there is for feed and flow which could be used to change the values during a print.

Two issues that I though up: 
* 0% and 100% need some special case so that they aren't normally changed, but there should be a way to do it.
* what if you want to only change one parameter, like your infill cooling, without touching anything else?

My suggestion would be implementing parameters MIN, MAX, VALUE that would allow commands like `SET_FAN_SPEED_MULTIPLIER FACTOR=125 MIN=25 MAX=80` which would increase fan speed by 25% for any M106 command that has the original value between 25 and 80% without touching anything outside of them, while a direct `VALUE=35` would only hit something that was set to be that fan speed in the slicer, like Infill for example - this could also be used for those 0% and 100% cases if setting MAX/MIN isn't quite right. There is the issue of M106 speeds being set by a PWM value of 0-255 while basically all slicers show the user a 0-100%, so Klipper should probably do the conversion as well.  

One complication is that this could/would cause Klipper to have multiple overlapping fan overrides though... Specific `VALUE` ones should override the ranges but I'm not sure how it should handle two overlapping ranges. Not allow them at all in the first place? Adding a `QUERY` that would print all the overrides as list and a `CLEAR` to clear all of them (or targeted ones with MAX/MIN/VALUE?)  them would probably also be needed.

Thoughts? 
Would this be useful or just way too complicated and niche? 
Or should it just be a simple `SET_FAN_MULTIPLIER 125` -> `M106 120*1.25` and let's call it a day?

(Happy Bot, Happy Life: [klippy.log](https://github.com/KevinOConnor/klipper/files/4898180/klippy.log))

ld3300, 2020-07-10:
hi @JohnEdwa 

You can already implement this any way you like.  You can replace M106 with a custom version and use parameters and variables on a permanent, per-print or even during live printing to set an override and using the G-code macro you could provide yourself ways to do a permanent or temporary override, or set the ranges.  [This section helps describe overriding g-code defaults](https://github.com/KevinOConnor/klipper/blob/4bdc11a8b3843b73079a2c8dfc8d8c0928c29662/config/example-extras.cfg#L425)

David

KevinOConnor, 2020-07-19:
As indicated, this can be done already by using macros and overriding the existing commands.

Also, please don't spoof the bot.  I'll clear the "incomplete issue" tag if the log is not needed.

-Kevin

KevinOConnor, 2020-08-05:
I'm going to close this as I think the original question was answered (it's possible today via macros).

-Kevin

closed 2020-08-05 by KevinOConnor