Klipper3d/klipper #3076: Homing G28 crashes effector on delta printer, sensorless homing

gonAlonso, 2020-07-10:
Every time I cancel a print on Octoprint, the effector crashes against the bed. I've created a g-code script to calibrate endstop phase (some G0 moves and G28 and then save the new config ) and it runs into the same problem
Any G28 g-code processed on a script is does not trigger a real home. Instead y calibrates the current position as home and goes on.
G28 codes manually inserted triggers the correct process.
I'm using TMC2130 sensorless home
Here I provide 3 different klippy.log cases where it runs into de same problem.
I've tuned driver_SGT to higher values, it bounces strongly when manually commanded G28 is issued, but does nothing on scripted G28
My custom phase calibrate script: [phase_calibrate.gcode.log](https://github.com/KevinOConnor/klipper/files/4904884/phase_calibrate.gcode.log)

jakep82, 2020-07-10:
Are you running Octoprint and DWC2 at the same time?  You won't get any official support if you're running DWC2 because it requires modifying code.

gonAlonso, 2020-07-10:
Sorry for the DWC config, but I'm not using it.
I tried to use it but wasn't able to make it work so I deleted klipper completely and git-cloned again.
I kept the same config file
This settings can be removed, thought

jakep82, 2020-07-10:
Remove that section from your config and submit a log without DWC2 running. 

Also note that the default Octoprint cancel script includes M84 which will disable your motors.  If disabling the motors causes your head to crash, you should edit the Octoprint cancel script and remove that line.

gonAlonso, 2020-07-11:
Edited **printer.cfg** and removed **DWC** config. Tried and still the same problem.
Removed Klipper using _klipper-uninstall_ and installed again. Kept the printer.cfg. Problem persists.
I added a G4 code previous to G28, maybe the wait forces a different behavior, but same issue.
Cancel script disables motors, but the effector crashes with the bed on any scripted G28 as on this:

> G90
> G28
> G0 F3600 X4 Y61.86 Z100
> G28 <-- HERE
> G0 F3600 X-3.7 Y61.66 Z121 <-- THIS CODE MAKES THE EFFECTOR TO CRASH ON THE BED

The first one is correctly executed but the second one just makes the motor to stop, seems like klipper resets its position to a homed state but no movement. Then the next G1 makes it go down and crash.
Typing this on the terminal do get a perfect result, but running a foo.gcode file with this g-code combination goes wrong

[klippy.log](https://github.com/KevinOConnor/klipper/files/4906555/klippy.log)

KevinOConnor, 2020-07-19:
It sounds like you need to tune "sensorless homing" - see: https://www.klipper3d.org/Sensorless_Homing.html

-Kevin

KevinOConnor, 2020-08-05:
I'm going to close this as it looks like the conversation has concluded.  As far as we know, the sensorless homing code works as intended.

-Kevin

closed 2020-08-05 by KevinOConnor

Afterschnitzel, 2021-08-23:
I agree with this issue on delta.
I am unable run G28 twice in one G-Code.
If there is more than one G28 in G-Code, it will not execute and Klipper will skip it. Just first G28 works.