Klipper3d/klipper #3077: SKR mini E3 V2.0 not working

jorgempy, 2020-07-11:
Hi!

I just received my SKR mini E3 V2.0 and i'm trying to install it.
did a git pull, make menuconfig, make clean, make, copied the bin to a microsd, it flashed. 
checked the microsd, fimware.cur exist now.
But no usb. No error, no nothing.

Started looking around and found #1884 .
checkout [ec3d865b](https://github.com/KevinOConnor/klipper/commit/ec3d865b517affd77678e5b1a45ef4691619726d) , did the same. nothing.
checkout [c83688b7](https://github.com/KevinOConnor/klipper/commit/c83688b7bc7fcbc1a413d5707d11a2d2b674d48d) (latest), did the same. nothing.

tried to check if i got a faulty board, with [BTT's firmware](https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/blob/master/firmware/V2.0/firmware.bin). booted just fine. usb working.
```
[ 1773.724760] usb 1-1.4: new full-speed USB device number 8 using dwc_otg
[ 1773.857832] usb 1-1.4: New USB device found, idVendor=1eaf, idProduct=0004
[ 1773.857846] usb 1-1.4: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[ 1773.857855] usb 1-1.4: Product: Maple
[ 1773.857863] usb 1-1.4: Manufacturer: LeafLabs
[ 1773.858922] cdc_acm 1-1.4:1.0: ttyACM0: USB ACM device
```

Any ideas of what to try next?
Thanks!

Attached are the configs of menuconfig.
[config_ec3d865b.txt](https://github.com/KevinOConnor/klipper/files/4906071/config_ec3d865b.txt)
[config_c83688b7.txt](https://github.com/KevinOConnor/klipper/files/4906073/config_c83688b7.txt)
[fake klippy.log](https://github.com/KevinOConnor/klipper/files/4906067/klippy.log)

BlackStump, 2020-07-11:
I have just powered up a SKR Mini E3 V2.0 without issue.
The config is [here](https://github.com/KevinOConnor/klipper/blob/master/config/generic-bigtreetech-skr-mini-e3-v2.0.cfg)
Please pay attention to the info at the top of the config!
[In particular to this line](https://github.com/KevinOConnor/klipper/blob/master/config/generic-bigtreetech-skr-mini-e3-v2.0.cfg#L5)

KevinOConnor, 2020-07-11:
Attaching a log is a requirement for using this github issue tracker.

closed 2020-07-11 by KevinOConnor

jorgempy, 2020-07-11:
Facelpalm!

Thanks @BlackStump!

For future reference (from config header):
```
# To use this config, the firmware should be compiled for the
# STM32F103 with a "28KiB bootloader". Also, select "Enable extra
# low-level configuration options" and configure "GPIO pins to set at
# micro-controller startup" to "!PA14".
```

TxBillbr, 2021-08-06:
There is something distinctly going south with the SKR Mini E3 v2.0. Not only have I had a problem with it, but have witnessed on a number of other folks printers. Following the instructions EXACTLY for some boards causes no USB registration while others work ok using the exact same instance of pi, cable/environment... I run Vorons but am a strong supporter of Klipper in the general community. These boards that will not communicate after flashing with klipper run just fine on Marlin firmware. This doesn't seem to be a board failure problem. I have three boards that fail to operate properly under klipper and one that works as expected.  Inspecting the boards doesn't reveal any differences in components or versions. I would be happy to send a failing board to any klipper dev to diagnose the issue. I suspect that BTT has changed some aspect of their USB implementation and handled that in their Marlin implementation. Please consider looking into this a little deeper as this is the recommended board for Voron V0 builders.  Thanks for all you have contributed. It is truly appreciated.

ChrisToxz, 2021-10-09:
Any updating regarding this issue? Same issue here with a some newer SKR Mini E2 V2.0 board. (No problems with an older E2V2.0 board)
Serial is not being detected after flashing with Klipper. Not with 'lsusb' or 'dmesg'. When flashing with Marlin, the serial port is being detected. Hereby I can confirm that the board is still working fine.

I have read many other post with same issue, but no solution at all.

Setting used:
STM32F103
28KiB Bootloader
8MHz crystal
Communication interface: USB
GPIO pins to set at micro-controller startup: "!PA14"

ChrisToxz, 2021-10-09:
Thanks for your quickly reply. This could be a potential fix to work around the USB, we will try and report back later.

Anyhow its seems to be clear that there are potential USB problems with new SKR E2 V2.0 boards. (Not sure how strictly BTT is regarding managing revisions, but would be not the first time that hardware is changed without clear notice).

It would be great if Klipper devs would clarify if they are aware of this issue and acknowledge it is an issue or if the cause can be blamed  to BTT them self.

TxBillbr, 2021-10-09:
@ChrisToxz You will note that @KevinOConnor closed this on July 11th, **2020** because of lack of logs. If you can figure out how to create a proper bug report, I am sure that he will get to it (in time.. lots on his plate). I didn't take the time to figure out how to instrument the board to capture klipper boot-up logs due lack of time. Without the USB port getting recognized in raspian, it takes extra steps to gather startup logs.

TxBillbr, 2021-10-09:
> I have no clue where to start to get a log from the BTT board(STM32F103) it self.
>  Also [ https://www.klipper3d.org/Contact.html#i-have-diagnosed-a-defect-in-the-klipper-software ]( https://www.klipper3d.org/Contact.html#i-have-diagnosed-a-defect-in-the-klipper-software ) (with the issue guide) is referring to the klippy.log which will not give useful information because Klipper is not connected at all.
> Is there any clear guide to get debug info from STM32F103 it self?

I came to the conclusion you needed an STM32 dev setup with debugger connections. I chose not to go down that path.

ChrisToxz, 2021-10-09:
Hmm. probably related to the following: https://twitter.com/BigTreeTech/status/1425478829867294722
It seems like there are board shipped with GD chips instead of ST chips, and they announced it after customer complained.

> Not sure how strictly BTT is regarding managing revisions, but would be not the first time that hardware is changed without clear notice

Wasn't wrong with this statement and also BTT is admitting this:
> We'd like to express our apology to you for the inconvenience due to our mistakes
https://docs.google.com/document/d/1lkGgzv1I1PG7LiQjgpBUTQ7e1_vC5ASGlN6ik3LCNSw/
BTW they promised now to exclusively use ST chips and you can request a refund or a replacement.

In their letter they also mention that GD chips works fine with just one line changed, but they are not mentioning what to change.

https://github.com/Klipper3d/klipper/issues/4535#issuecomment-939365577 confirmed that switching to STM fixed this issue.