Klipper3d/klipper #3078: I have ender 5 plus but have not been able to make work it give me the error State: Offline (Error: Internal error during connect: invalid syntax (adc_temperature.py, line 283)) i do have a slice engineering Thermistor: High Temperature

EMDM25, 2020-07-11:


EMDM25, 2020-07-11:
Here is the code that used to on abc_temperature.py

# Obtain temperature using linear interpolation of ADC values
#
# Copyright (C) 2016-2018  Kevin O'Connor <kevin@koconnor.net>
#
# This file may be distributed under the terms of the GNU GPLv3 license.
import logging, bisect

######################################################################
# Interface between MCU adc and heater temperature callbacks
######################################################################

SAMPLE_TIME = 0.001
SAMPLE_COUNT = 8
REPORT_TIME = 0.300
RANGE_CHECK_COUNT = 4

# Interface between ADC and heater temperature callbacks
class PrinterADCtoTemperature:
    def __init__(self, config, adc_convert):
        self.adc_convert = adc_convert
        ppins = config.get_printer().lookup_object('pins')
        self.mcu_adc = ppins.setup_pin('adc', config.get('sensor_pin'))
        self.mcu_adc.setup_adc_callback(REPORT_TIME, self.adc_callback)
        query_adc = config.get_printer().load_object(config, 'query_adc')
        query_adc.register_adc(config.get_name(), self.mcu_adc)
    def setup_callback(self, temperature_callback):
        self.temperature_callback = temperature_callback
    def get_report_time_delta(self):
        return REPORT_TIME
    def adc_callback(self, read_time, read_value):
        temp = self.adc_convert.calc_temp(read_value)
        self.temperature_callback(read_time + SAMPLE_COUNT * SAMPLE_TIME, temp)
    def setup_minmax(self, min_temp, max_temp):
        adc_range = [self.adc_convert.calc_adc(t) for t in [min_temp, max_temp]]
        self.mcu_adc.setup_minmax(SAMPLE_TIME, SAMPLE_COUNT,
                                  minval=min(adc_range), maxval=max(adc_range),
                                  range_check_count=RANGE_CHECK_COUNT)

######################################################################
# Linear interpolation
######################################################################

# Helper code to perform linear interpolation
class LinearInterpolate:
    def __init__(self, samples):
        self.keys = []
        self.slopes = []
        last_key = last_value = None
        for key, value in sorted(samples):
            if last_key is None:
                last_key = key
                last_value = value
                continue
            if key <= last_key:
                raise ValueError("duplicate value")
            gain = (value - last_value) / (key - last_key)
            offset = last_value - last_key * gain
            if self.slopes and self.slopes[-1] == (gain, offset):
                continue
            last_value = value
            last_key = key
            self.keys.append(key)
            self.slopes.append((gain, offset))
        if not self.keys:
            raise ValueError("need at least two samples")
        self.keys.append(9999999999999.)
        self.slopes.append(self.slopes[-1])
    def interpolate(self, key):
[... 224 lines cut ...]
    pheaters = config.get_printer().load_object(config, "heaters")
    for sensor_type, params in [("AD595", AD595),
                                ("AD8494", AD8494),
                                ("AD8495", AD8495),
                                ("AD8496", AD8496),
                                ("AD8497", AD8497),
                                ("PT100 INA826", PT100),
                                ("Sl100", Sl100),
                                ("CR100", CR100)]:
        func = (lambda config, params=params:
                PrinterADCtoTemperature(config, LinearVoltage(config, params)))
        pheaters.add_sensor_factory(sensor_type, func)
    for sensor_type, params in [("PT1000", PT1000)]:
        func = (lambda config, params=params:
                PrinterADCtoTemperature(config,
                                        LinearResistance(config, params)))
        pheaters.add_sensor_factory(sensor_type, func)

def load_config_prefix(config):
    if config.get("resistance1", None) is None:
        custom_sensor = CustomLinearVoltage(config)
    else:
        custom_sensor = CustomLinearResistance(config)
    pheaters = config.get_printer().load_object(config, "heaters")
    pheaters.add_sensor_factory(custom_sensor.name, custom_sensor.create)

EMDM25, 2020-07-11:
[klippy.log](https://github.com/KevinOConnor/klipper/files/4906287/klippy.log)

EMDM25, 2020-07-11:
[adc_temperature.log](https://github.com/KevinOConnor/klipper/files/4906288/adc_temperature.log)

ld3300, 2020-07-11:
You somehow have a dirty version of the code.  Something has been changed.  Try reinstalling a clean version of klipper and try again.  Make sure you don't follow the TWC2 instructions.

EMDM25, 2020-07-11:
What step do you recommend, I have done two times the installation I receive the same error I am not sure if Klipper has the RT Table for slice engineering Thermistor

ld3300, 2020-07-11:
try doing this
`git pull upstream master`
`git reset --hard`

That should force you to a clean install.  Going further with troubleshooting will be better if we can get your log to say the build you are on is a clean version.  Then, if there are still issues, we have a clean starting place.

David

EMDM25, 2020-07-11:
[klippy.log](https://github.com/KevinOConnor/klipper/files/4906319/klippy.log)
[adc_temperature.log](https://github.com/KevinOConnor/klipper/files/4906320/adc_temperature.log)

EMDM25, 2020-07-11:
This are the log

EMDM25, 2020-07-11:
Do you know how the screen should look like after install klipper on the ender 5 plus

ld3300, 2020-07-11:
Well, I've been out of it for a while, but if you are replacing the adc_temperature file you need to delete the 2 extra spaces before the bracket  on line 283.  That is what the klippy error is pointing to.

So open  klipper/klippy/extras/dc_temperature.log and make sure the closing bracket is all the way to the left (without the 2 preceding spaces)

-David

EMDM25, 2020-07-11:
Didn't work the same error.

ld3300, 2020-07-11:
Your last log also showed that the source has been modified.  Are you doing something after install to manipulate any of the files?

EMDM25, 2020-07-11:
The only file that I have manipulated is the  adc_temperature.log to add Thermistor.

ld3300, 2020-07-11:
it should be adc_temperature.py and try this one and let me know what it does:

[adc_temperature.zip](https://github.com/KevinOConnor/klipper/files/4906359/adc_temperature.zip)

EMDM25, 2020-07-11:
[klippy.log](https://github.com/KevinOConnor/klipper/files/4906369/klippy.log)

EMDM25, 2020-07-11:
the same

jakep82, 2020-07-11:
I see several typos in the file you're trying to edit, but that's not the correct way to add a thermistor.  You need to define the thermistor in your printer.cfg file.

https://github.com/KevinOConnor/klipper/blob/master/config/example-extras.cfg#L926

EMDM25, 2020-07-11:
Here is my printer.cfg file 

[printer.log](https://github.com/KevinOConnor/klipper/files/4906402/printer.log)

jakep82, 2020-07-11:
You need to define the thermistor in your config file using instructions at the link I provided and revert adc_temperature.py to it's original unmodified state.

KevinOConnor, 2020-07-11:
As indicated, it's not necessary to modify the code to add a new sensor.  That said, I went ahead and added a new default "SliceEngineering 450" definition (commit af981432).  If you upgrade to the latest code (see the FAQ for upgrade instructions) you can use that definition.

-Kevin

EMDM25, 2020-07-11:
Thanks -kevin

It did work at least the thermistor sensor correctly, I can connect and see temp, but now I get the error (Shutdown due to M112 command). I did a fresh installation of octoprint and Klipper.

Thanks so much for your help. here is the log file.

[klippy.log](https://github.com/KevinOConnor/klipper/files/4907542/klippy.log)

EMDM25, 2020-07-11:
[image]

jakep82, 2020-07-11:
You've missed a very important step in the installation.  Specifically this line: "In the Settings tab, navigate to the "Behavior" sub-tab and select the "Cancel any ongoing prints but stay connected to the printer" option. Click "Save".

https://github.com/KevinOConnor/klipper/blob/master/docs/Installation.md#configuring-octoprint-to-use-klipper

Also note that you can't move an axis until it's homed.

EMDM25, 2020-07-12:
You Where correct

KevinOConnor, 2020-07-25:
Okay, it sounds like the original issue was resolved.

-Kevin

closed 2020-07-25 by KevinOConnor