Klipper3d/klipper #3079: shared_heater/2in1 hotend -->Tool Change Script, + add to examples

DroneMang, 2020-07-11:
Hi,

I have a 2 in 1 hotend with a shared heater/nozzle.   I have it working with these macros with S3D handling the tool change with no scripts in S3D for the tool change.  

[gcode_macro T0]
gcode =
    ACTIVATE_EXTRUDER EXTRUDER=extruder

[gcode_macro T1]
gcode =
    ACTIVATE_EXTRUDER EXTRUDER=extruder1

The problem is that S3D pauses right on the print while conducting the tool change.  I need it to raise Z immediately and then do the tool change.  I want to have the T0, T1 macros defined in Klipper for the same reason as having the start script there so you don't have to reslice as much.  I have reviewed the multi-extruder and idex cfg's, but these are more complicated and not the same.  With this, I request that a sample-2in1-extruder be created and added to the example cfg folder.

Pseudocode for T0/T1:
(Determine state of extruders and which filament is loaded, would require saving the state at the end of the print somehow, this is not as important, perhaps just a SAVE_CONFIG prior to running the rest of the END_PRINT macro)
if statement ( if this is the first T0 (or T1) in the gcode, do nothing or just run ACTIVATE_EXTRUDER EXTRUDER=extruder1 # (or extruder) for good measure to make sure that extruder is active)
Raise Z
Retract 50 mm of current extruder
ACTIVATE_EXTRUDER EXTRUDER=extruder1 # (or extruder) Activate the next extruder 
Feed 50 mm of next extruder
Lower Z 

[gcode_macro T1] # This is missing the if statement to test if this is the first occurance of T0 or T1
gcode =
    G91; Relative Coordinates
    G1 Z+5 F480; Lift Z for Tool Change
    G1 E-50 F600
    ACTIVATE_EXTRUDER EXTRUDER=extruder1
    G1 E50 F600
    G1 Z-5 F480; Lower Z
    G90; Absolute Coordinates

1st Attempt at including the if statement inspired by the tz6-dual example cfg, is this how to do conditional statements in Klipper? I am using two if statements, could this be done with an else statement for the second block?

default_parameter_TOOL_CHANGE_COUNT: 1
[gcode_macro T1]
gcode =
    {% if TOOL_CHANGE_COUNT|int == 1 %}
    ACTIVATE_EXTRUDER EXTRUDER=extruder1
    default_parameter_TOOL_CHANGE_COUNT: 2
    {% endif %}
    {% if TOOL_CHANGE_COUNT|int == 2 %}
    G91; Relative Coordinates
    G1 Z+5 F480; Lift Z for Tool Change
    G1 E-50 F600
    ACTIVATE_EXTRUDER EXTRUDER=extruder1
    G1 E50 F600
    G1 Z-5 F480; Lower Z
    G90; Absolute Coordinates
    {% endif %}

Now I am unsure if the above macro will work.  I have attached my S3D gcode settings as a picture, these are needed for the cancel objects plugin to Octoprint.  I don't know if I should be in relative mode only as above, or if I need to zero the extruder and use absolute mode, as shown below.  Below is a script for S3D.  But here again, what happens when the first T0 or T1 occurs in the gcode?  I would already have the filaments positioned and ready to go for the print, so I don't need a filament swap there.

S3D tool change script (untested) inspired by http://forum.makergear.com/viewtopic.php?t=2535
G91; Relative Coordinates
G1 Z+5 F480; Lift Z for Tool Change
G90; Absolute Coordinates
T[old_tool]; Select Old Tool
G92 E0; Zero Extruder
G91; Relative Coordinates
G1 E-50 F600; Retract 50mm
G90; Absolute Coordinates
G92 E0; Zero Extruder
T[new_tool]; Select New tool
G92 E0; Zero Extruder
G91; Relative Coordinates
G1 E50 F600; Insert 50mm
G92 E0; Zero Extruder
G1 Z-5 F480; Return Z to resume printing
G90; Absolute Coordinates

Thanks!

Bogus log:
[klippy.log](https://github.com/KevinOConnor/klipper/files/4907332/klippy.log)

[image]

ld3300, 2020-07-12:
if it is helpful here is the link to my config file that manages the shared hot-end.  It has quite a few conditional statements and I use variables a lot.  I have some extra bells and whistles built in that you likely won't need.  I have a variable coming from another config file that sets the retract and restore differences.  Some of this can be optimized, but I have turned my attention elsewhere:  [extruders.cfg](https://github.com/ld3300/Klipper_Configs/blob/master/configs/extruders.cfg)

It is probably going to be a little cryptic because of my naming, but let me know if it is helpful or if you have any questions.

David

KevinOConnor, 2020-07-25:
Any further updates on this?  If someone has an improved example we can add it to a sample config in the config directory.

-Kevin

DroneMang, 2020-07-25:
I did find the above links very useful but those macros are pretty advanced
it would be nice to see a more intermediate step, basically the minimum
needed to have klipper handle the tool changes rather than the slicer.

KevinOConnor, 2020-08-18:
Okay, I'm going to close this as it looks like the conversation has concluded.  If someone wishes to contribute examples then please open a PR with those additions.

-Kevin

closed 2020-08-18 by KevinOConnor

ld3300, 2020-09-24:
Hi Kevin,
I am slowly coming back to life.  I should be able to make a simpler sample if that is still desired.

David

KevinOConnor, 2020-09-25:
Sounds good.

-Kevin

DroneMang, 2020-09-25:
I would greatly appreciate a simpler example! Thanks.

DroneMang

ld3300, 2020-09-25:
@DroneMang please let me know if #3368 is helpful, or if there are missing requirements or unclear explanations

Hi @KevinOConnor, I could use some feedback about if there is a more elegant way to do the variable comparisons, or if there is a method you think would be a more typical user application.  The variable comparisons are also quite long, so some advice on if the splitting into multiple lines is the right way to go and if I did it correctly.

Thanks again,

David