Klipper3d/klipper #3083: BL-Touch different probing speed

TheRunnerPi, 2020-07-12:
Hi all,

ich hope you can help me. 
I want to let my Printer home "Z"/BL-Touch probing with a fast speed.
After thats done, i want to do another home "Z"/BL-Touch probe with a slower speed, to get more accuracy.

Found this older request, but dont know what to do with that info:
https://github.com/KevinOConnor/klipper/issues/748

Maybe someone did something similar and can help me :)

Thanks,
Daniel

[klippy.log](https://github.com/KevinOConnor/klipper/files/4908783/klippy.log)

robthide37, 2020-07-13:
the answer is in the post you linked add speed:2="whatever you want " is what i would interpret. Put it in the probe section . By default my printer does a second slower touch which i believe may have been documented in the config extras iirc

TheRunnerPi, 2020-07-13:
> 
> 
> the answer is in the post you linked add speed:2="whatever you want " is what i would interpret. Put it in the probe section . By default my printer does a second slower touch which i believe may have been documented in the config extras iirc

sorry, still dont know how to handle that and i cant find anything in the  config extras about that..
https://github.com/KevinOConnor/klipper/blob/master/config/example-extras.cfg

ld3300, 2020-07-13:
Yeah, I'm not seeing that documented either.

jakep82, 2020-07-13:
You can add second_homing_speed to your stepper_z section to customize the speed of a second home.  By default the second_homing_speed is half the speed of homing_speed.  Make sure you also use a large enough homing_retract_dist so the pin won't hit the bed when it's deployed the second time.

https://github.com/KevinOConnor/klipper/blob/master/config/example.cfg#L62

TheRunnerPi, 2020-07-15:
> You can add second_homing_speed to your stepper_z section to customize the speed of a second home. By default the second_homing_speed is half the speed of homing_speed. Make sure you also use a [...]

Thanks Jakep82, it works now :)

closed 2020-07-15 by TheRunnerPi