Klipper3d/klipper #3085: No option 'min_x' in section: bed_mesh

goosst, 2020-07-13:
Hello,

had the following issue: 
[klippy.log](https://github.com/KevinOConnor/klipper/files/4914165/klippy.log)

- did a (manual) bed mesh calibration on a delta printer
- executed the commands: `BED_MESH_PROFILE SAVE=mirror_bed` and `SAVE_CONFIG`
- results in the following error:
```
Send: BED_MESH_PROFILE LOAD=mirror_bed
Recv: // No option 'min_x' in section: 'bed_mesh mirror_bed'
Recv: // Once the underlying issue is corrected, use the "RESTART"
Recv: // command to reload the config and restart the host software.
Recv: // Printer is halted
Recv: !! No option 'min_x' in section: 'bed_mesh mirror_bed'
Recv: ok
```

It doesn't sound like a legitimate error since it's a delta printer.

thanks!

goosst, 2020-07-13:
p.s. manually deleting the mirror_bed from the (autogenerated) configuration, resolved the issue as well.

Arksine, 2020-07-15:
I see what is happening here, I will push a fix in my next bed_mesh pull request. 

In the meantime you can work around this by saving the profile to a different name after calibration, before you enter SAVE_CONFIG.  It is the restart and load that is causing the issue.

KevinOConnor, 2020-07-25:
FYI, PR #3111 was merged.

-Kevin

closed 2020-08-05 by KevinOConnor