Klipper3d/klipper #3087: Multiple endstops or sensorless homing as a failsafe to prevent crashes.

spott, 2020-07-13:
I'm building a Voron V2.4, which uses TMC2209s for the (4) z-axis motors.  The printer has high accuracy mechanical end-stops which works great.  It also has TMC2209s, which allow sensor-less homing.  By measuring when a stepper motor has skipped a step, they can guess that the axis has been homed.

I would like to use the mechanical endstops for my homing procedure, but have the sensorless homing endstops be "failsafes"... if the printer notices a skipped step (or a few) when moving along an axis, it backs up and turns off.

Is there any way to do this with klipper right now?

If there isn't, where would I look to possibly add this?

mdealer, 2020-07-19:
Stumbled upon this while looking for the same thing. It's not even endstops, but rather triggers that execute a certain script when a pin changes. I can't find any info on triggers regarding Klipper... My printer has X min max and Y min max endstops, whereas max is a trigger to shut down the machine to prevent crashes and min is just re-home. Stall detection on all axes (X, Y, Z1-4) also just shut down the machine.

KevinOConnor, 2020-07-19:
Currently, homing only uses one endstop per stepper, and that endstop is only ever used during homing.  (There is no runtime hardware endstop detection.)

I'm sure it's possible to add additional features.  Start with the developer documentation: https://www.klipper3d.org/Overview.html#developer-documentation

The homing code is mainly in klippy/homing.py and the trinamic sensorless homing code is in klippy/extras/tmc.py.

-Kevin

spott, 2020-07-21:
Is there a reason there isn't any runtime hardware end-stop detection?  Is it performance? or just wasn't worth implementing elsewhere?

I guess the end-stops aren't implemented as interrupts but are instead polled during moves?

KevinOConnor, 2020-07-25:
To date, no one has taken the time to implement run-time hardware endstop detection.  There isn't really that much use in it on 3d printers as the builtin software endstop detection works well.

-Kevin

mdealer, 2020-07-31:
@KevinOConnor Could you elaborate more on what you mean by "hardware endstop" and "software endstop"? 

So, is it possible to configure a sensor otherwise used as an endstop as a plain trigger to, e.g. stop the machine entirely?

KevinOConnor, 2020-08-01:
> Could you elaborate more on what you mean by "hardware endstop" and "software endstop"?

By "hardware endstop detection", I mean active checks to see if the endstop switch changes state during a print.  By "software endstop detection" I mean checks during move planning to prevent moves that would command the toolhead into the printer frame.

> is it possible to configure a sensor otherwise used as an endstop as a plain trigger to, e.g. stop the machine entirely?

It is possible to define a gcode_button config section that invokes `{printer.gcode.action_emergency_stop()}`.  However, one can't define a single switch as both an endstop and as a gcode_button.

-Kevin

KevinOConnor, 2020-08-18:
I'm going to close this as it looks like the original question has been answered (multiple endstops are not currently supported, nor is runtime endstop detection).

-Kevin

closed 2020-08-18 by KevinOConnor