Klipper3d/klipper #3088: MCU 'zboard' shutdown

aegelsky, 2020-07-13:
Hi 

My big printer was upgraded in couple ways, so now I have SKR boards (1.3 on XYE and Pro on 6 Z's).
Also I want to get rid or 2000mm (2 meters) threaded rods - no matter CNC lead screw or just cheap threaded rods, they can't be produced straight enough if it's not as thick as my leg :) so I added a worm reductor to one of 6 Z's (with weird/uneven 1:13 ratio, but it fits and works).
Problem is that I can't tune it now using FORCE_MOVE STEPPER=stepper_z DISTANCE=10 VELOCITY=7
depending on VELOCITY it shutdowns somewhere near end of move of in the middle.
Also in one attempt it can work out, and in second it can shutdown MCU with the very same line of command.
Mostly it crashes after VELOCITY > 3, on 5-7 it's 99% that it will crash.
Updated klipper with git pull, then FIRMWARE_RESTART, same on latest.
Also sometimes it crashes XY MCU (no name) and other times Z's (zboard).
Attaching log file.
Any ideas what can it be?
[klippy.log](https://github.com/KevinOConnor/klipper/files/4915288/klippy.log)

KevinOConnor, 2020-07-14:
This is an unfortunate conflict between FORCE_MOVE and your configuration with very low Z acceleration.  Simplest solution is to always provide an `ACCEL=3` parameter to your FORCE_MOVE commands.  (Otherwise, FORCE_MOVE defaults to using instantaneous velocity changes which can trigger a safety check programmed into the micro-controller.)

-Kevin

aegelsky, 2020-07-14:
ok, thanks, guess we can close this.

closed 2020-07-14 by aegelsky