Klipper3d/klipper #3089: replace /tmp/printer with an actual serial port

smark-, 2020-07-13:
Is it possible to replace the standard Klipper virtual com port (/tmp/printer) with an actual serial port? I'd like to be able to send commands to the raspberry Pi from my computer

manu7irl, 2020-07-13:
You already can send commands.
From terminal via ssh.
echo "firmware_restart">/tmp/printer
For example...

KevinOConnor, 2020-07-19:
I would advise against sending g-codes to Klipper over a real serial port - line errors could corrupt the commands and result in drastic errors.

Also, this looks similar to the existing #1097.

-Kevin

mkuf, 2020-07-20:
Have a look at ser2net (https://linux.die.net/man/8/ser2net).
You should be able to publish klippys Serial device via ser2net and use socat on your local Computer to send commands.

This should work in theory but I haven't tested it nor recommend it for production use.

-Markus

closed 2020-08-05 by KevinOConnor