Klipper3d/klipper #3090: Safe Homing should ignore Z probe during XY movement

ianespana, 2020-07-14:
I'm trying out Klipper for the first time and ran into an interesting issue.

My printer uses a piezo disc as it's Z probe which has been very reliable (been using it for about 2 years with Marlin). However, when I moved to Klipper I encountered an issue when trying to home the Z axis. Every time I used safe homing the homing procedure would time out before even doing the first attempt at probing the bed.
Thanks to some folks on Discord we managed to track down the issue, and it turns out the piezo disk was triggering while the safe homing procedure was moving towards the specified XY coordinates, thus making Klipper think it had already reached the bed for the first time, and because of this the second probe timed out when it went beyond a specific height.

My suggestion to fix this is very simple: Ignore any Z endstop events during the traverse to XY specified in the safe homing config. It's not necessary to listen for such events during a movement unrelated to the Z axis, and it causes more issues than it should.
[klippy.log](https://github.com/KevinOConnor/klipper/files/4916070/klippy.log)

Hywelmartin, 2020-07-16:
A small pause before the  probeing takes place usually solves those kind of problems (there is this kind of function in marlin as well)

```
[probe]
pin = ar18
z_offset = 0
activate_gcode:
      G4 P500
```

KevinOConnor, 2020-07-19:
To be clear, Klipper does not check the Z endstop during XY homing.  As mentioned earlier though, it sounds like a previous move is causing the Z endstop to report as triggered at the start of the Z home.

-Kevin

KevinOConnor, 2020-08-05:
I'm going to close this as it looks like the original question was answered (add a delay before z homing).

-Kevin

closed 2020-08-05 by KevinOConnor