Klipper3d/klipper #3091: OLED Screen cut off by 2 pixels.

dorfman2, 2020-07-14:
### The Issue
Howdy! I've just built an Voron V0, and after asking the community at large, there's a small bug with respect to this [custom screen](https://github.com/VoronDesign/Voron-Hardware/tree/master/V0_Display).

[image]

In this image, the case is not blocking the edge of the screen, I have verified this issue with the case off (just failed to take a photo while it was off).

The consensus after asking in the Voron Discord seems to be the 1.3" screens claim to be SSD1306 but they're really SH1106. So the horizontal screen buffer is 132 even though the OLED is 128. The might be causing them to be off-center.

**Would there be the possibility of adding a pixel offset setting?**

### Current Workaround
It's been suggested and I've verified this current workaround.

In `klippy/extras/display/uc1701.py` look for the line 

> `pix_x = x * 8`

There should be 3 of them in the file (write_text, write_graphics, write_glyph). For each one, add this line after it:

> `pix_x += 2`

[image]

Thanks for reading!

[klippy.log](https://github.com/KevinOConnor/klipper/files/4916030/klippy.log)

john--, 2020-07-14:
These 1.3" OLEDs are SSD1306's but use a SH1106 controller. So the horizontal screen buffer is 132 even though the OLED itself is 128. This causes them to be off-center. It's not really an issue with Klipper itself, but it would be pretty nice to allow a horizontal offset to be configured for these strange displays.

More info on these displays here: https://forum.arduino.cc/index.php?topic=256374.0, particularly this comment:

> There is only one small difference between SSD1306 and SH1106: The SH1106 controller has an internal RAM of 132x64 pixel. The SSD1306 only has 128x64 pixel.
> 
> First problem for standard 128x64 OLEDs with SH1106 controller is: How is the 128x64 window mapped into the 132x64 RAM?
> It seems, that the 128x64 OLED is centered in most cases within the 132x64 area, that means pixel (2,0) in ram is pixel (0,0) on the display.

KevinOConnor, 2020-07-17:
Thanks for reporting.  It would really help if someone with the hardware could put together a github "pull request" with the necessary changes.   (These types of changes are really difficult to do without the hardware in hand.)

-Kevin

KevinOConnor, 2020-08-18:
I'm going to mark this as inactive as it doesn't look like a developer is currently working on this.  If there is interest, feel free to submit a PR with the changes.

-Kevin

closed 2020-08-18 by KevinOConnor