Klipper3d/klipper #3092: Multiple kill pins - possible?

nikkoaws, 2020-07-14:
Is there any way to define an extra kill input pin?

I normally use a 3d touch sensor for z homing, however some suck plastic under the nozzle area will result in a bed hit as the sensor is never triggered.

I would like to put in a physical micro switch lower down on the z-axis as a safety net as the frame is acrylic and the geared z-steppers can do quite some damage before they start to skip.

I could wire in parallel with the display kill switch, but was wondering if there is a way to do it by defining a second input pin?

Thanks!

nikkoaws, 2020-07-14:
Just to add some extra clarity, I did try a gcode_button entry, however the button handling is blocked while the bltouch homing is active.

ld3300, 2020-07-15:
Can you wire the switch to the raspberry pi and have it issue an emergency stop if triggered?

nikkoaws, 2020-07-16:
> Can you wire the switch to the raspberry pi and have it issue an emergency stop if triggered?

In my case I am using an old shuttle small form factor PC rather than a raspberry pi.

I wanted to put the printer back together, so I ended up soldering a wire to the ramps display adapter board so the limit switch is wired in parallel with the emergency stop button on the display.

I might close this ticket as I have the functionality I need.

Thanks.

closed 2020-07-16 by nikkoaws