Klipper3d/klipper #3093: Question: Which pins on a mcu for STEP signal?

Daniel-Kurth, 2020-07-14:
I have a question regarding the pin remapping which is done in the config file (thus no log file). 
My application requires 12 stepper drivers and because many boards don't support that natively I was wondering if I could remap additional pins (e.g. for LEDs or hobby servos) to drive the STEP and ENABLE signal to an external stepper driver. 

Here my question, does the STEP pin need to be a function pin (e.g. Timer) or can it be a. "normal" digital IO pin?

BlackStump, 2020-07-14:
https://github.com/KevinOConnor/klipper/blob/master/config/sample-multi-mcu.cfg

Daniel-Kurth, 2020-07-14:
I've already looked at several config files, but either I'm missing the key information I'm looking for or it's just not in them. 
To rephrase my initial question. Does a pin used for generating the STEP signal have to have a alternate function on the mcu like a timer?

Arksine, 2020-07-15:
A step pin can be any GPIO.

Daniel-Kurth, 2020-07-15:
Thank you for the clarification!

closed 2020-07-15 by Daniel-Kurth