Klipper3d/klipper #3094: Macro M600 after timeout

daredevil-ferreira, 2020-07-14:
[

[printer.txt](https://github.com/KevinOConnor/klipper/files/4919410/printer.txt)

When I M600, if I immediately change my filament, then execute de macro RESUME_MACRO, then resume Octoprint, it works.

If I leave my printer some time, temperature from bed & nozzle are set off automatically (I assume some security here when there is no move). 
In that case, when I try to use the RESUME_MACRO, I got the message to home first.

===========
Send: RESUME_MACRO
Recv: // Unknown command:"M117"
Recv: !! Must home axis first: 22.994 25.929 25.485 [250.850]
WARNING! Received an error from the printer's firmware, ignoring that as configured but you might want to investigate what happened here! Error: Must home axis first: 22.994 25.929 25.485 [250.850]
===========

daredevil-ferreira, 2020-07-14:
Is there a way also to maintain the bed temperature after pausing the print ?

Hywelmartin, 2020-07-15:
extend the idle timeout by incorporating SET_IDLE_TIMEOUT TIMEOUT=3600 into your M600 and reset the idle timeout in the resume macro with SET_IDLE_TIMEOUT TIMEOUT=600

daredevil-ferreira, 2020-07-16:
Great, it works perfectly.
Thanks

KevinOConnor, 2020-07-25:
I'm going to close this as it looks like the original issue was resolved.

-Kevin

closed 2020-07-25 by KevinOConnor