Klipper3d/klipper #3095: Conflict between gcc-pru and gcc-avr

goeland86, 2020-07-15:
Installation using the install-beaglebone.sh script doesn't account for gcc-pru and gcc-avr both trying to write the file `/usr/lib/libcc1.la`.

When installing on beaglebone, we do not need the AVR or ARM toolkits installed (in a single-MCU case) to the best of my understanding. And at present, the attempt to have both gcc-pru and gcc-avr installing fails the installation overall.

This error happens when using a Debian 10.3 base linux image retrofitted to use kernel 4.14 to keep the PRU API consistent with Klipper's expectations.

https://github.com/KevinOConnor/klipper/blob/180f44b8763159d55e1c9c9339849765b093cbb7/scripts/install-beaglebone.sh#L20

goeland86, 2020-07-15:
Just to clarify why this is an issue:
Beagleboard's currently supported images for Beaglebone black are based on Debian 10.3, which is where security updates are happening, and in the Replicape's particular case, python 3 versions that will support OctoPrint (i.e. Python 3 > 3.6) are only available in 10.3. If Replicape will keep being supported, it needs to run Klipper on Debian 10.3 - and I don't see any reason it cannot, except for this package conflict during installation at the moment.

goeland86, 2020-07-15:
Here is the klipper install log file, as done by [ReFactor](https://github.com/goeland86/ReFactor) using Ansible.
[klipper_install.log](https://github.com/KevinOConnor/klipper/files/4924324/klipper_install.log)

KevinOConnor, 2020-07-17:
This sounds like a bug in the debian gcc-avr package or the beaglebone gcc-pru package - they shouldn't be conflicting on this file.  We really do want to be able to install both the gcc-avr and gcc-pru packages because a user may be using multi-mcu or may be using the beaglebone as just a regular host machine.

Is it possible to report this to the beaglebone developers?

-Kevin

goeland86, 2020-07-17:
Yeah I'll submit it to the Debian and beaglebone teams. Just need to find
the respective bug report systems they use...

Just wanted to report it here in case anyone is running into this. The
Refactor beta image is removing the avr packages installation for now, with
a specific note that avr-based boards aren't supported for a multi MCU
package at this stage. Since it's a beta image, I'm ok with that, and
hopefully it'll be back in place for the actual release if upstream
provides a fix.

Also found the cause for the Klipper_pru service needing a manual restart
after booting. Will submit a PR for that though.

closed 2020-08-25 by goeland86