Klipper3d/klipper #3096: Klipper Firmware on ATMega1284p at 12Mhz

rrazgriz, 2020-07-16:
I have a nonstandard control board with an ATmega1284p running off of a 12Mhz crystal oscillator (host is an OrangePi Zero, will be running OctoPrint). During installation of Klipper, while configuring the firmware, I only have the option for 8, 16, or 20Mhz. in `~/klipper/src/avr/Kconfig`, the only clock frequencies listed under `CLOCK_FREQ` are those same three. 

Would 12Mhz work on an ATmega1284p with Klipper as-is, and if so, how would one configure this? If not, could support be added, and an option added during configuration? Thanks. 

[klippy.log](https://github.com/KevinOConnor/klipper/files/4928784/klippy.log) included for completeness; no printer.cfg for the moment as I haven't yet built it for the printer's board.

rrazgriz, 2020-07-16:
Solved: a dev friend looked at the source and determined that the speed could simply be added by modifying the avr settings in `src/avr/Kconfig` and adding an option for CLOCK_FREQ to be set to `12000000`, in the same pattern as 8, 16, and 20Mhz. They will send a pull req for this when convenient (I'm not very git-savvy).

closed 2020-07-16 by rrazgriz

harikattar, 2020-07-16:
For completeness, the patch but I'm not willing to dox myself over this, so no full-name account pull request.
```
Author: Harik <github@harik.welp.gs>
Date:   Thu Jul 16 01:52:51 2020 -0400

    Kconfig: Support 12mhz crystals on 1284p boards
    
    Some unusual variants of the 1284p design use 12mhz crystals. Because this
    is rare, gate the option behind LOW_LEVEL_OPTIONS.
    
    Tested-by: @rrazgriz
    Signed-off-by: Harik <github@harik.welp.gs>

diff --git a/src/avr/Kconfig b/src/avr/Kconfig
index e430b246..84979d31 100644
--- a/src/avr/Kconfig
+++ b/src/avr/Kconfig
@@ -65,6 +65,8 @@ config AVRDUDE_PROTOCOL
 
 choice
     prompt "Processor speed" if LOW_LEVEL_OPTIONS
+    config AVR_FREQ_12000000
+        bool "12Mhz"
     config AVR_FREQ_16000000
         bool "16Mhz"
     config AVR_FREQ_20000000
@@ -77,6 +79,7 @@ endchoice
 config CLOCK_FREQ
     int
     default 8000000 if AVR_FREQ_8000000
+    default 12000000 if AVR_FREQ_12000000
     default 20000000 if AVR_FREQ_20000000
     default 16000000
```